corna/me_cleaner #142: In the event the recent CPU bugs aren't accidental...

ilikenwf, 2018-01-05:
I'm not a hardware but software guy so I could be off base here, but...

What if Spectre and/or Meltdown were somehow inserted via National Security Letter (NSL) or a bribed Intel employee, and are being played off as bugs because of an NSL? 

Same with other manufacturers, but in our case, if this were true, wouldn't it stand to reason there's some kind of HAP bit to turn them off, or perhaps the HAP or AltMeDisable bit could somehow nerf them when the BUP process is run?

Just a thought, and curious to see if, once available, someone here can try the proof of concepts against an unpatched (kernel and bios/microcode updates both) me_cleaned machine to see if there's any difference at all.

seropp, 2018-01-05:
IMHO it's more of a case of naïveté / wishful thinking / "omg it's faster this way" (TM) than anything else. It's not something you can switch off, but something you can mitigate to a certain degree in the OS. It's more akin to dumb plumbing than to "smart" electrical switches.

ilikenwf, 2018-01-05:
There was that suspected LARP on  4chan by some "intel employee" indicating that there was a decade old bug like this in the hardware months ago, so the coincidences freak me out considerably regardless of the truth.

seropp, 2018-01-06:
Even so, it could range anywhere from:

"we gave you enough time to fix it, you just ignored it, so here it is for the world to see" 
to:
"the aliens we stole the microchip tech from have been using this to prepare for the invasion"

... and anything in between. 

For something more down-to-earth (note: JUNE):
https://arstechnica.com/gadgets/2018/01/intel-faces-class-action-lawsuits-regarding-meltdown-and-spectre/

ilikenwf, 2018-01-06:
I've kept my eye on the Talos, but I'm hoping for something, anything that can be trustworthy...so far no contenders exist, so it's now a question of "what's the lesser evil?" coupled with "mitigate the stuff we already own as much as possible and try to live with it (for now)..."

The only actual MIPS platform products I'm aware of are cheap android tablets from China. Because these exist I wonder if we couldn't get something custom made using existing components and open it up somehow? Expensive, but an option for sure.

seropp, 2018-01-06:
The way I see it, in 12-24 months, cpus will be fixed, desktop operating systems will be patched (a lot of them are already deployed, thanks Google!), coreboot/libreboot will have advanced their work to more recent boards (hopefully), alternative architectures will be faster, this is all actually great news, it's just another bump in the road to better things.

Recent kernel patches should already have mitigated the big issues on linux, turning it into a small performance problem more than anything else.

I'm not aware of any benefits MIPS brings over anything else. x86 is by itself regarded to have a lot of legacy baggage, I've never looked at it as 'bad' by design, same for ARM.

'Secure' is often in the eyes of the beholder. Sure, android phones that never update or PCs with old DDR memory and motherboards with bioses that never update or legacy operating systems are a huge problem. But buy a new pc in about a year from now or get a phone that's supported by actual updates (official or otherwise), and you're 98% set. 

ME had been regarded as unremovable for quite a while. I'm betting that there will be public tools to send it commands in the next 6 months. Maybe using a usb-to-usb cable on the same motherboard. We've never been more in control, but it's a process.

matt123b, 2018-01-10:
@ilikenwf There's definitely a possibility that the Intel ME is being used by the NSA/CIA or whatever to remotely access machines of high value targets, though I doubt it's used against the average person. I also think that the ME is mainly for enterprise use, just as Intel claims. However, the ME does provide and unnecessary and rather extreme security risk to the average Joe who has not use for it. And as far as the 4chan posts go, everything posted there should be considered an artistic work of fiction and nothing more. 

With Spectre and Meltdown, these were not intentional. The three letter agencies may have known about them and even leveraged them against specific high value targets, but I still have my doubts about that since they're using this hardware themselves in all their offices. This was caused by CPU makers like Intel taking a bunch of shortcuts on security to increase performance, and it actually worked really well, for a time at least. They've been doing it for 20+ years and now they're screwed, basically. 

Meltdown targets Intel's specific implementation of OOE (out of order execution). OOE is a cool trick where a program is split into pieces and all of them are processed separately, and then made available when needed. This basically works to more equally distribute the processing load across different parts of the CPU, so that the other parts are not wasted and doing nothing. So if we have parts 1 and 2 of the CPU and parts A and B of the program, with OOE you can execute part A on part 1 and part B on part 2, rather than putting both A and B through part 1 only, creating an unnecessary bottleneck by not utilizing the hardware. However, if part B is needed after part A, part A will still be made available first, but when B is needed it can be sent up immediately because it's already processed. This is the best way I can explain OOE. Anyways, this stuff is all stored in system memory somewhere, and in the old days an attacker could easily figure out where things were in memory because it would all get put in the same spot, or wherever there was free memory for the data. So then you get address space layout randomization which randomizes the process, making it like a bazillion times more difficult for an attacker to guess where stuff is located in memory. Well, Intel basically broke kernel ASLR, or KASLR, because this OOE flaw allows an attacker to read your memory because things aren't isolated right, so this has totally undermined this software protection. This is really bad but can be mostly patched over in the kernel by not allowing the kernel or programs certain access, but at the same time it will have a massive performance impact on some systems, depending on how the kernel is interacting with certain parts of the hardware. This is where the 30% thing comes into play.

Spectre is targeting speculative execution, which lots of CPUs other than Intel are using. SE is basically when the CPU can guess what a program will need to do next, and sort of load that part, and when that part is selected the other options are discarded and that path is used, but it all loads faster because of the preemptive loading that has taken place. SE has become a problem because you can use it to access parts of other programs, though not the kernel. This needs CPU microcode to fix but it's very difficult to use and not all that useful in the wild like Meltdown is. Basically BIOS updates or hardware replacement will fix this, and kernel patches can help a tiny bit. It's definitely not good but you can live with it and not have to worry about getting malware or anything as long as all software is kept up to date and sandboxed. It's really that simple. 

If you're getting patches from your OS vendor and BIOS updates from your motherboard manufacturer, you can install them and just deal with the slow downs. If you really want/need new hardware, AMD stuff is getting pretty good now, and the PSP is an ARM TrustZone implementation, which to me on the surface at least seems better than the ME, due to it being more standardized hardware for SecureBoot related functionality and not advertised as having networking capabilities IIRC. But then there are also a bunch of cheap ARM SBCs and the more expensive server tier stuff like the Talos II. I've already ordered a new AMD ThinkPad A275 for my general computing needs, and I'm still debating the Talos II as a desktop.

Atavic, 2018-01-11:
[NSA Didn't Know of Meltdown, Spectre, Trump Cyber Czar Says](https://github.com/sudocurse/reading-list/issues/3248)

>The National Security Agency didn't know about the Meltdown or Spectre flaws, White House cybersecurity coordinator Rob Joyce said at the International Conference on Cyber Security at Fordham University Law School here today (Jan. 11).

Do you beLIEve?

minecrawler, 2018-01-12:
> high value targets, though I doubt it's used against the average person

I remember a similar discussion about why everyone should encrypt their website, even though there is no special content. Good thing that is now slowly enforced by browser vendors~ Because! If only high-profile content is encrypted, it is easy to guess from a data stream if the data is high-profile or not. That makes things a lot easier for black-hats. Same for ME or people who try to actively defend themselves against security bugs and shortcuts (like Meltdown and Spectre). If only high-profile targets disable their ME, then it is easy to find out who those high-profile targets are - even if they are not on the list - and develop new attacks against them. Sometimes, they are criminals, but very often, they are just reporters or victims of unfair prosecution in difficult environments. You really should be happy that you have the freedom to decide that you don't even care about the small stuff.

Also, with growing automation, I find that argument more and more naïve. The NSA can access the computers of everyone, automatically, without a single human doing anything, 24/7, and do whatever they want with the data they extract. They could setup an AI to search your computer for certain files, contacts, phone numbers, etc. and then take action in case something is found. That might start with terrorist activity searches, but easily lead to you getting an invitation to your local police station because of a crack you downloaded, because you want to play an old game without connecting to Windows Live. No matter the reason and arguments, I have a right to privacy, and that alone should be enough to block the NSA and any other agency, organization, group or individual from my computer.

I think, the best option we have is more open source. I am not even talking about free or libre. I don't care if a company wants money and puts a special contract on their source or blueprints. It's sad, but at least I can take a look at important parts. That allows professionals to decide if there are risks and then those risks should be fixed asap. Ideally, software should be compiled on _my_ machine and hardware should be built by _my_ 3D electronics printer. But unfortunately, only a few people are able to do the former and the latter has not been invented, yet. So let's keep our distrust up and do the best we can in our situation.

seropp, 2018-01-12:
> The NSA can access the computers of everyone, automatically, without a single human doing anything, 24/7, and do whatever they want with the data they extract.
Citation needed.

https://en.wikipedia.org/wiki/Reductio_ad_absurdum

start with a false premise, anything becomes real.

matt123b, 2018-01-12:
@minecrawler While this certainly could be true, I have not heard of the ME/AMT "features" being used by any government to remotely connect to home PCs without the end user doing anything to set it up. After all, if you trust your router, you can grab a trusted machine, like a Libreboot laptop, and sit in betweena dn monitor your network traffic. People have done this, and the government trolls know people will do this, so it would be silly to have an always active connection to CIA/NSA/FBI servers. I believe that the ME is an extremely low level security risk, acting as a potential entry point for malicious modifications to your boot firmware, which includes the BIOS/EFI, some low level drivers, and CPU microcode. The threat comes not just from the state, but also from hackers out in the wild that could use the thing to hijack the most sensitive and low level parts of your PC. According to some past leaks, the NSA is supposedly developing lots and lots of malware that infects the boot firmware on everything from iPhones to enterprise servers, but on Intel machines, how the hell do they consistently deploy this malware without depending on OS or device specific exploits? Well, you use the ME! The ME processes CPU microcode, and often BIOS updates as well. It has full access to said boot firmware, so anyone with full access to the ME can just install their own software. This is the most likely scenario, as it can be used against individual targets with consistent results. There are drivers for the ME in Linux and Windows, as it's officially advertised as an enterprise management feature. It's actually incredibly useful, just not for home users that don't need it and shouldn't have it. 

If I were one of the turds working for the three letter agencies and I needed a way to backdoor people using Intel chips, I would just use the ME to deploy malware into the boot firmware by running programs on a machine that I infect with easier to find OS exploits, or by bundling the malicious program with something else the user installs with admin rights, like a video game or a PDF viewer or something like that. It's just the common sense approach.

minecrawler, 2018-01-12:
> Citation needed.

citation not needed, it is a fact. Is you computer connected to the internet? Yes? Then it can be accessed, else you would not be able to have stuff like Skype phone calls or surf the net. And if, but only by the slight chance, that the NSA has computers, it does not matter when you start up your computer. It can be registered and handled, once connected, at once, and software can start working on getting your data and profile, no human intervention necessary.

> I have not heard of the [..] being used

Why would you ever hear about that? While it certainly is like an Area51 mystery (no one heard about the stuff they did there, right?), it concerns me personally, so I tend to be pessimistic and think of the worst case scenario. It might be reality, because who knows the whole truth?

> grab a trusted machine [..] and sit in between and monitor your network traffic

sure can do, however, just because you cannot find that kind of traffic does not mean that it does not exist or that it cannot happen. Since I already started with the extraterrestrial comparison: if you look at the sky, will you see a spaceship? Probably not. Does that mean that humans are the only intelligent life form? According to your argument: yes, because there is obviously no spaceship in the sky.

Also, if I wanted to use ME, I would not do it the obvious way, but use internal local sockets and proxy my traffic over another application. Update services, telemetry services or assistance systems (Cortana...), for example. Usually, if you want to protect yourself, you would use a hardware firewall, anyway, so you want a way to bypass that. When on the computer, you don't need admin/root rights to open a socket connection to a local address, but by doing so, you would be able to access ME, resulting in the highest rights, probably even without (extensive enough) logging.

> extremely low level security risk

is what you say, and follow up with a few ways it can be misused and even state that you would use it, if you were employed by one of those agencies, which makes it a lot higher risk than it should be, especially for people who don't need it, like home users. It has very central access to my personal data...

seropp, 2018-01-12:
I just wanted to

Well, I mean...

You know

oh, nevermind

matt123b, 2018-01-12:
@minecrawler You're comparing apples to oranges here. Embedded coprocessors in Intel chipsets have absolutely nothing to do with aliens, and it's a false equivalency anyways. As far as the networking thing goes, the Intel ME is not magic. It can't just teleport packets through anther dimension. It has to be networked to at least your router to get to the internet, and if you trust the router, you can trust the connections you're seeing, and thus far, nobody has seen the ME making an active connection with anything when it isn't configured to do so. And by a low level risk, I meant low level as in ring -3 in the machine, not that it's harmless (quite the opposite is true). It operates below the rest of the hardware. In fact, the ME starts before the BIOS and is always running as long as the machine has power. When you want to start the rest of the hardware, the ME firmware from the BIOS ROM gets loaded runs first, and then hands off the higher level boot processes to the BIOS/EFI, which hands it to the software on the first bootable disk, like the bootloader/kernel. 

Not everything is a government conspiracy, and at a certain point you need take a more realistic approach to analyzing this piece of hardware. It's an embedded ARC or x86 CPU with really high privileges and its own operating system. It is not a spooky ghost that lives in your computer, and it is not something to be afraid of. Neither Intel engineers or US government employees are perfect either, so don't elevate them to a God-like status. It's just ridiculous. We're a bunch of hobbyist security researchers on GitHub that are modifying and testing software. That's all this is and all it ever will be. 

Conspiracy theories are fun and interesting to entertain as long as there's some amount of truth to them, but you shouldn't take them so seriously. It just leads to excessive and unnecessary fear mongering.