corna/me_cleaner #143: Make output images play nicer with latest flashrom 1.0 features

ArthurHeymans, 2018-01-05:
Hi

With the latest flashrom release, 1.0, it is possible to flash regions on the flash using the --ifd flag. (this reads the layout from the ifd which is currently on flash)
e.g. If I just want to read or write the bios region:

flashrom -p internal -r/-w bios_region.rom --ifd --image bios

Reading will generate a file that has the other regions padded with 0xff.
One can also add the --noverify-all flag or -N to have flashrom only verify the region it operates on to speed things up

With this feature there is no need anymore to modify an existing full image with a cleaned me. One can simply flash the cleaned me with '--image me' in flashrom. This requires however that the this cleaned me is padded to the size of the flash. Some holds when one wants to flash a descriptor.

sidenote: this also removes the need to add these blobs in the coreboot build system as flashrom handles them gracefully.

So my question is whether it would be possible to add a flag like '--padded' when outputting a modified descriptor or me, which flashrom can then easily use with the --ifd and layout support? (or do that by default and add a flag --unpadded, which IMO is better as I don't like the coreboot build system handling such blobs, but that is personal opinion ofc)

ArthurHeymans, 2018-01-06:
OTOH https://review.coreboot.org/#/c/flashrom/+/23021/ will probably remove the need for padding...

corna, 2018-01-10:
I'll keep an eye on that change, thanks.