corna/me_cleaner #144: HP ZBook 15 by any chance internally? Nothing works...

nagyrobi, 2018-01-05:
$ ./ifdtool -d L70_0137.bin 
File L70_0137.bin is 8388608 bytes
Unknown descriptor version: 7

$./me_cleaner.py -O L70_013p.bin L70_0137.bin 
Unknown image

skochinsky, 2018-01-07:
It seems this update does not contain the full flash image, but only the BIOS region.  You will need to use an external flasher to dump the flash chip and clean that image before flashing it back.

nagyrobi, 2018-01-07:
Thank you. That's not going to happen any time soon, but I intend to look into this too.

nagyrobi, 2018-01-14:
01.38 Rev.A BIOS updated here too:
- Updates the Intel Microcontroller unit (MCU) to version 0x96.
- Provides improved security of UEFI code and variables. HP strongly recommends transitioning promptly to this updated BIOS version which supersedes all previous releases.

NOTE: Due to security changes, after this BIOS update is installed, previous versions cannot be reinstalled.

nagyrobi, 2018-01-14:
@skochinsky, your words made me think maybe HP releases ME image separately. And indeed, if you look at the download page I linked previously, there's an item called "Intel Full Management Engine (ME) Firmware Component". If you decompress the .exe, you'll find a subdirectory "ME" with a file "PCH_8M_MB_UPD_Production.bin" inside.

Running ME analyzer against it, it shows:
-------[ ME Analyzer v1.38.1 r109 ]-------

File:     PCH_8M_MB_UPD_Production.bin (1/1)

Family:   ME
Version:  6.2.61.3535
Release:  Production
Type:     Update
SKU:      5MB MB
Date:     2017-04-09
Platform: Ibex Peak
Latest:   Yes

Unfortunately ifdtool doesn't recognize it:
# ./ifdtool -d PCH_8M_MB_UPD_Production.bin 
File PCH_8M_MB_UPD_Production.bin is 4219424 bytes
No Flash Descriptor found in this image

neither me_cleaner:
./me_cleaner.py -c PCH_8M_MB_UPD_Production.bin 
Unknown image

But it should be damn simple, because this is only the ME image separately. BIOS is not touched! Should be possible to clean this separate file directly... any idea?

nagyrobi, 2018-01-14:
@skochinsky, your words made me think maybe HP releases ME image separately. And indeed, if you look at the download page I linked previously, there's an item called "Intel Full Management Engine (ME) Firmware Component". If you decompress the .exe, you'll find a subdirectory "ME" with a file "ME9.1_5M_Production.bin" inside.

Running ME analyzer against it, it shows:
-------[ ME Analyzer v1.38.1 r109 ]-------

File:     ME9.1_5M_Production.bin (1/1)

Family:   ME
Version:  9.1.42.3002
Release:  Production
Type:     Region, Stock
SKU:      5MB
SVN:      1
VCN:      15
PV:       Yes
Date:     2017-10-24
Size:     0x8D2000
Platform: LPT/WPT
Latest:   Yes

ifdtool doesn't recognize it:
./ifdtool -d ME9.1_5M_Production.bin 
File ME9.1_5M_Production.bin is 9248768 bytes
No Flash Descriptor found in this image

me_cleaner does:
./me_cleaner.py -c ME9.1_5M_Production.bin 
ME/TXE image detected
Found FPT header at 0x10
Found 28 partition(s)
Found FTPR header: FTPR partition spans from 0x160000 to 0x210000
ME/TXE firmware version 9.1.42.3002
Checking the FTPR RSA signature... VALID

Patching it succeedes:

./me_cleaner.py -O ME9.1_5M_Production_patch.bin ME9.1_5M_Production.bin 
ME/TXE image detected
Found FPT header at 0x10
Found 28 partition(s)
Found FTPR header: FTPR partition spans from 0x160000 to 0x210000
ME/TXE firmware version 9.1.42.3002
Reading partitions list...
 PSVN (0x00000bc0 - 0x000000c00, 0x00000040 total bytes): removed
 FOVD (0x00000c00 - 0x000001000, 0x00000400 total bytes): removed
 MDES (0x00001000 - 0x000002000, 0x00001000 total bytes): removed
 FCRS (0x00002000 - 0x000003000, 0x00001000 total bytes): removed
 EFFS (0x00003000 - 0x0000df000, 0x000dc000 total bytes): removed
 BIAL (NVRAM partition, no data, 0x0000add3 total bytes): nothing to remove
 BIEL (NVRAM partition, no data, 0x00003522 total bytes): nothing to remove
 BIIS (NVRAM partition, no data, 0x00036000 total bytes): nothing to remove
 NVCL (NVRAM partition, no data, 0x000069c9 total bytes): nothing to remove
 NVCM (NVRAM partition, no data, 0x0000439b total bytes): nothing to remove
 NVCP (NVRAM partition, no data, 0x0000a3fe total bytes): nothing to remove
 NVHM (NVRAM partition, no data, 0x00000058 total bytes): nothing to remove
 NVJC (NVRAM partition, no data, 0x00003da0 total bytes): nothing to remove
 NVKR (NVRAM partition, no data, 0x00005fb4 total bytes): nothing to remove
 NVNF (NVRAM partition, no data, 0x0000175f total bytes): nothing to remove
 NVOS (NVRAM partition, no data, 0x0003a34d total bytes): nothing to remove
 NVSH (NVRAM partition, no data, 0x000022c0 total bytes): nothing to remove
 NVSM (NVRAM partition, no data, 0x00001de8 total bytes): nothing to remove
 NVTD (NVRAM partition, no data, 0x00001feb total bytes): nothing to remove
 NVUK (NVRAM partition, no data, 0x00008940 total bytes): nothing to remove
 PLDM (NVRAM partition, no data, 0x000043c5 total bytes): nothing to remove
 TMNN (NVRAM partition, no data, 0x000001a6 total bytes): nothing to remove
 GLUT (0x000df000 - 0x0000e8000, 0x00009000 total bytes): removed
 LOCL (0x000e8000 - 0x0000ec000, 0x00004000 total bytes): removed
 WCOD (0x000ec000 - 0x000160000, 0x00074000 total bytes): removed
 FTPR (0x00160000 - 0x000210000, 0x000b0000 total bytes): NOT removed
 NFTP (0x00210000 - 0x00048a000, 0x0027a000 total bytes): removed
[... 42 lines cut ...]
             [-GENERIC] [-HALTRCFG]

-H|?                  Displays help screen.
-VER                  Displays version information.
-EXP                  Displays example usage of this tool.
-VERBOSE<file>        Display the debug information of the tool.
-F<file>              File used for updating the FW.
-Y                    Automatically answer Yes to prompts.
-SAVE<file>           Save the current FW to an update image.
-FWVER<file>          Display the FW Version of current FW or update image.
-PARTID<Partition ID> Provide specific Partition ID to perform partial update.
-ALLOWSV              Allows same version firmware updates.
-FORCERESET           Automatically Reboots system after update (if needed).
-OEMID<UUID>          OEM ID needed to perform firmware update.
-PASS<pass>           MeBX password. Optional with the '-f' option.
-GENERIC              Perform the update through MEI without credentials.
-HALTRCFG             Halts remote configuration.

Unfortunately the kill switch cannot be applied either because:

./me_cleaner.py -s -O ME9.1_5M_Production_switch.bin ME9.1_5M_Production.bin 
ME/TXE image detected
-d, -D, -M, -S and -s require a full dump

Is there any way to skip checking in FWUpdLcl64.exe? Or a prepared version of FWUpdLcl64.exe whick skips check of the image file?

skochinsky, 2018-01-14:
The firmware update is done by the running ME itself, `FWUpdLcl64` basically just sends the image over.

nagyrobi, 2018-01-14:
I see. I get "Checking the FTPR RSA signature... VALID" - don't know why it rejects it...