corna/me_cleaner #145: Does internal oem bios flashing require Intel ME Region Write Access and Intel ME Region Read Access to be enabled?

laserstrike, 2018-01-07:
Me cleaner failed for me using oem bios flashing, but then I read the instructions to internal coreboot flashing and it says that these two settings need to be enabled for it work.

Is the same thing required for oem bios flashing?

corna, 2018-01-07:
Which settings?

If you mean the R/W access to the ME region:
IIRC during the power on all the regions can be read/written freely, then the BIOS locks them and they can't be written anymore until the next boot. In this way the BIOS can freely write the updates and then lock the regions.

If you have the ME region locked you can only flash:
 * through the OEM BIOS (but it seems that it's not possible in your case, are the firmware updates signed?)
 * externally

laserstrike, 2018-01-07:
OK thanks. Yea I've tried flashing the modified file through OEM BIOS

closed 2018-01-07 by laserstrike