corna/me_cleaner #146: MSI H81M-E33: need to reflash BIOS each time I want to change BIOS settings

victorn1, 2018-01-08:
Motherboard: MSI H81M-E33
Processor: i5-4690 (Haswell)
BIOS: OEM 6.7

me_cleaner was applied without -s or -S keys. Flashing was done with M-Flash built-in BIOS utility.
After applying initial BIOS settings, Windows worked fine without a glitch. But when I tried to enter BIOS setup again, the message appeared: 

All setting were reset to default values.
Press F1 to Run SETUP
Press F2 to Continue

After that I was not able to change BIOS settings, always got this message when booting. I tried to restore original OEM ME with no success. M-Flash silently ignores ME updating. And DOS flash utility says:
Error: BIOS does not support ME Entire Firmware update

What I have now. System works fine. But if I want to change BIOS settings, I have to first flash BIOS (without ME) and only then enter BIOS setup. After booting OS I am not able to enter BIOS setup flawlessly, always get the message above, and the BIOS settings are reset to defaults at each reboot if I try to enter BIOS setup.

My questions: 
1. Is there a way to enter BIOS setup and change BIOS settings after I boot Windows?
2. How can I reflash ME to the original OEM without a special flashing device, since I do not have any?

corna, 2018-01-10:
1. I don't know. With BIOS you can change the CMOS memory from the OS (but the layout is board-specific, so you have to reverse-engineer it), but I don't know if (and how) UEFI exposes its variables to the OS.
2. I'm afraid you can't, you need an external flasher. Note that they're quite common (Raspberry Pi, Beaglebone, ...) and quite cheap (like the [CH341A](https://www.ebay.com/itm/CH341A-Series-24-EEPROM-BIOS-Writer-25-SPI-Routing-LCD-Flash-USB-Programmer-NEW/172143017320?hash=item281485fd68:g:JOkAAOSwAuZX45aR), 2.50$ on eBay).

victorn1, 2018-01-11:
Thank you for your answer. I am still exploring the problem. It may be buggy M-Flash (MSI BIOS built-in utility) that caused the problem. I will try to use Intel flash programming tool from
https://www.win-raid.com/t596f39-Intel-Management-Engine-Drivers-Firmware-amp-System-Tools.html

victorn1, 2018-01-14:
The problem was in buggy M-Flash. With FPT (intel flash programming tool) everything is Ok. Now there are no problems with patched ME.

victorn1, 2018-01-15:
Unfortunately, the problem has reappeared after two days. I have to revert to unmodified OEM BIOS.

5685C4A059D5, 2018-07-19:
It is a rather strange behavior. I got this issue on the B85M-G43 too.
MSI uses AMI Aptio BIOS but this issue seems exclusive to MSI.