corna/me_cleaner #147: me_cleaner on Lenovo Y70-70

ivanB1975, 2018-01-08:
I wrote this tutorial (https://github.com/ivanB1975/y70_mecleaner/wiki/howto) on how to apply the me_cleaner to a Lenovo Y70-70.
Would it be possible to add it to the links of this page "https://github.com/corna/me_cleaner/wiki/How-to-apply-me_cleaner" ?
Thanks

corna, 2018-01-10:
Thanks for sharing your tutorial. Just some notes:

 * Can you split the tutorial in two parts, one for me_cleaner and one for the extra modifications? You can add a link to the second part in the first one if you want.
 * There's some broken formatting at the top, can you fix it?
 * The _intelmetool_ Boot Guard patch has been merged, the manual patching is not required anymore.

(I totally forgot to reply your email, sorry)

ivanB1975, 2018-01-15:
Sorry I didn't receive any notification for your answer.
Sure I can split it easily and I will check also the other 2 points.
No problem for the email :)

ivanB1975, 2018-01-15:
Ok I applied all the modification to the document. It should be ok. Let me know if you need other modifications.

corna, 2018-01-16:
Added, thanks for the contribution.
Note that, starting from Debian/Raspbian Stretch, `flashrom` is [available in the repos](https://packages.debian.org/search?keywords=flashrom&searchon=names&suite=all&section=all&sourceid=mozilla-search), so you gan just `apt-get` it (instead of building it from source).

closed 2018-01-16 by corna

ivanB1975, 2018-01-16:
Thanks I will add this point to the wiki