corna/me_cleaner #149: Not working on Dell Vostro V131 Intel(R) Core(TM) i5-2430M CPU

laserstrike, 2018-01-10:
*Bad news, you have a `HM67 Express Chipset Family LPC Controller` so you have ME hardware on board and you can't control or disable it, continuing...

MEI** found: [8086:1c3a] 6 Series/C200 Series Chipset Family MEI Controller #1

ME Status   : 0x1e000245
ME Status 2 : 0x66000006

ME: FW Partition Table      : OK
ME: Bringup Loader Failure  : NO
ME: Firmware Init Complete  : YES
ME: Manufacturing Mode      : NO
ME: Boot Options Present    : NO
ME: Update In Progress      : NO
ME: Current Working State   : Normal
ME: Current Operation State : M0 with UMA
ME: Current Operation Mode  : Normal
ME: Error Code              : No Error
ME: Progress Phase          : Host Communication
ME: Power Management Event  : Pseudo-global reset
ME: Progress Phase State    : Host communication established

Through the OEM BIOS method I tried both -s and -S but ME still turns up normal. It's weird, too, because when flashing, it's as if I'm flashing the normal BIOS, because I get no errors except a brief message to disable the beeps and high fans with FN-X, which I think only happens because of the method it was flashed.

I ran me_cleaner.py -c on both the original and modified files to see if the script was even really working, and what I got is a difference on how many partitions where found. 11 on the original firmware and 1 on the modified. So I'm _guessing_ the script is working, although it does say on both, "Full Image Detected." I'm going to have to postpone trying again until I get the gear to try to do it externally, but I just wanted to leave this here in case someone with this same system managed to figured it out. Thanks. 

BIOS for this system http://www.dell.com/support/home/us/en/04/Drivers/DriversDetails?driverId=RM8PG

- as bin file (original_firmware) https://my.mixtape.moe/cjeknj.bin
- result (modified_firmware; had to rename to modified.rom to flash) https://my.mixtape.moe/ssqhxu.bin

corna, 2018-01-10:
It seems to me that your flashing tool is not flashing the ME region, maybe because it's detecting that the ME version is the same as the one installed.

You should definitely try with an external programmer.

laserstrike, 2018-01-10:
Kinda scary. But I'll try it when I can. Thanks for all your work with this tool!

corna, 2018-01-10:
It's not, trust me ;)
You're welcome!

closed 2018-02-20 by laserstrike

laserstrike, 2018-02-27:
I tried it, but it didn't work. All attempts to reflash stock bios also failed; I think my motherboard is fried. Just going to get an x230 or somethin.

Found Winbond flash chip "W25Q32.V" (4096 kB, SPI) on ch341a_spi.
(blank@blank):~/Downloads/Bios$ flashrom -p ch341a_spi -c W25Q32.V -w modified.bin
flashrom v0.9.9-r1954 on Linux 4.13.0-32-generic (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Calibrating delay loop... OK.
Found Winbond flash chip "W25Q32.V" (4096 kB, SPI) on ch341a_spi.
Reading old flash chip contents... done.
Erasing and writing flash chip... FAILED at 0x0025ec01! Expected=0xff, Found=0xfe, failed byte count from 0x0025e000-0x0025efff: 0x1c
ERASE FAILED!
Reading current flash chip contents... done. Looking for another erase function.
FAILED at 0x000ae801! Expected=0xff, Found=0xfe, failed byte count from 0x000a8000-0x000affff: 0x7
ERASE FAILED!
Reading current flash chip contents... done. Looking for another erase function.
FAILED at 0x00053201! Expected=0xff, Found=0xfc, failed byte count from 0x00050000-0x0005ffff: 0x4b
ERASE FAILED!
Reading current flash chip contents... done. Looking for another erase function.
FAILED at 0x00037700! Expected=0xff, Found=0x00, failed byte count from 0x00000000-0x003fffff: 0x228
ERASE FAILED!
Reading current flash chip contents... done. Looking for another erase function.
FAILED at 0x000f7900! Expected=0xff, Found=0x7f, failed byte count from 0x00000000-0x003fffff: 0x284
ERASE FAILED!
Looking for another erase function.
No usable erase functions left.
FAILED!
Uh oh. Erase/write failed. Checking if anything has changed.
Reading current flash chip contents... done.
Apparently at least some data has changed.
Your flash chip is in an unknown state.

FAILED at 0x00000200!

skochinsky, 2018-02-28:
Please use flashrom support (e.g. IRC or mailing list) to figure out why flashing failed. It does not seem to be an issue with me_cleaner.

corna, 2018-02-28:
I agree, you should use the flashrom support. Make sure that both the write protect and the HOLD pin are pulled up.