corna/me_cleaner #150: Can I use me_cleaner on ME only FW updates?

archfan, 2018-01-10:
Hi,

I just asked Asrock for an updated BIOS due to the critical Intel ME vulnerability "SA-00086".

Unfortunately I didn't get a full patched BIOS with the latest ME firmware but just the latest ME firmware instead.

https://www.asrock.com/microsite/2017IntelFirmware/

I did apply the ME cleaner here for testing. Do you think it's possible to flash the cleaned FW.BIN with the official Intel FWUpdLcl  tooll? Has anybody tried this?

Results below:

> python me_cleaner.py -c ../FW.bin
> ME/TXE image detected
> Found FPT header at 0x10
> Found 13 partition(s)
> Found FTPR header: FTPR partition spans from 0x1000 to 0x130000
> Found FTPR manifest at 0x1448
> ME/TXE firmware version 11.8.50.3425
> Checking the FTPR RSA signature... VALID

> 
> python me_cleaner.py -O modified_FW.bin FW.bin                     
> ME/TXE image detected
> Found FPT header at 0x10
> Found 13 partition(s)
> Found FTPR header: FTPR partition spans from 0x1000 to 0x130000
> Found FTPR manifest at 0x1448
> ME/TXE firmware version 11.8.50.3425
> Reading partitions list...
>  FTPR (0x00001000 - 0x000130000, 0x0012f000 total bytes): NOT removed
>  FTUP (0x00272000 - 0x000600000, 0x0038e000 total bytes): removed
>  DLMP (      no data here      , 0x00000000 total bytes): nothing to remove
>  PSVN (0x00000e00 - 0x000001000, 0x00000200 total bytes): removed
>  IVBP (0x0026e000 - 0x000272000, 0x00004000 total bytes): removed
>  MFS  (0x00130000 - 0x00026e000, 0x0013e000 total bytes): removed
>  NFTP (0x00272000 - 0x00057d000, 0x0030b000 total bytes): removed
>  ROMB (      no data here      , 0x00000000 total bytes): nothing to remove
>  WCOD (0x0057d000 - 0x0005fd000, 0x00080000 total bytes): removed
>  LOCL (0x005fd000 - 0x000600000, 0x00003000 total bytes): removed
>  FLOG (0x00600000 - 0x000601000, 0x00001000 total bytes): removed
>  UTOK (0x00601000 - 0x000603000, 0x00002000 total bytes): removed
>  ISHC (0x00603000 - 0x000634000, 0x00031000 total bytes): removed
> Removing partition entries in FPT...
> Removing EFFS presence flag...
> Correcting checksum (0xa5)...
> Reading FTPR modules list...
>  FTPR.man     (uncompressed, 0x001448 - 0x002180): NOT removed, partition manif.
>  rbe.met      (uncompressed, 0x002180 - 0x002216): NOT removed, module metadata
>  kernel.met   (uncompressed, 0x002216 - 0x0022a4): NOT removed, module metadata
>  syslib.met   (uncompressed, 0x0022a4 - 0x002308): NOT removed, module metadata
>  bup.met      (uncompressed, 0x002308 - 0x0028b2): NOT removed, module metadata
>  pm.met       (uncompressed, 0x0028b2 - 0x002960): NOT removed, module metadata
>  vfs.met      (uncompressed, 0x002960 - 0x0033f8): NOT removed, module metadata
>  evtdisp.met  (uncompressed, 0x0033f8 - 0x003586): NOT removed, module metadata
>  loadmgr.met  (uncompressed, 0x003586 - 0x0036ae): NOT removed, module metadata
>  busdrv.met   (uncompressed, 0x0036ae - 0x003a56): NOT removed, module metadata
>  gpio.met     (uncompressed, 0x003a56 - 0x003ba0): NOT removed, module metadata
>  prtc.met     (uncompressed, 0x003ba0 - 0x003d50): NOT removed, module metadata
>  policy.met   (uncompressed, 0x003d50 - 0x003f12): NOT removed, module metadata
>  crypto.met   (uncompressed, 0x003f12 - 0x00409c): NOT removed, module metadata
>  heci.met     (uncompressed, 0x00409c - 0x004268): NOT removed, module metadata
>  storage.met  (uncompressed, 0x004268 - 0x004564): NOT removed, module metadata
>  pmdrv.met    (uncompressed, 0x004564 - 0x004688): NOT removed, module metadata
>  maestro.met  (uncompressed, 0x004688 - 0x004772): NOT removed, module metadata
>  fpf.met      (uncompressed, 0x004772 - 0x00487e): NOT removed, module metadata
>  hci.met      (uncompressed, 0x00487e - 0x004980): NOT removed, module metadata
>  fwupdate.met (uncompressed, 0x004980 - 0x004a88): NOT removed, module metadata
>  ptt.met      (uncompressed, 0x004a88 - 0x004b94): NOT removed, module metadata
>  touch_fw.met (uncompressed, 0x004b94 - 0x004cc0): NOT removed, module metadata
>  rbe          (Huffman     , 0x004cc0 - 0x0073c0): NOT removed, essential
>  kernel       (Huffman     , 0x0073c0 - 0x017380): NOT removed, essential
>  syslib       (Huffman     , 0x017380 - 0x02a400): NOT removed, essential
>  bup          (Huffman     , 0x02a400 - 0x054a00): NOT removed, essential
>  pm           (Huffman     , 0x054a00 - 0x058400): removed
>  vfs          (Huffman     , 0x058400 - 0x065a00): removed
>  evtdisp      (Huffman     , 0x065a00 - 0x068480): removed
>  loadmgr      (Huffman     , 0x068480 - 0x06d100): removed
>  busdrv       (Huffman     , 0x06d100 - 0x070d40): removed
>  gpio         (Huffman     , 0x070d40 - 0x072bc0): removed
>  prtc         (Huffman     , 0x072bc0 - 0x073ec0): removed
>  policy       (Huffman     , 0x073ec0 - 0x07ed80): removed
>  crypto       (Huffman     , 0x07ed80 - 0x0999c0): removed
>  heci         (LZMA/uncomp., 0x0999c0 - 0x09d8c0): removed
>  storage      (Huffman     , 0x09d8c0 - 0x0a3e40): removed
>  pmdrv        (Huffman     , 0x0a3e40 - 0x0a5bc0): removed
>  maestro      (Huffman     , 0x0a5bc0 - 0x0aaa00): removed
>  fpf          (Huffman     , 0x0aaa00 - 0x0ad340): removed
>  hci          (LZMA/uncomp., 0x0ad340 - 0x0adc00): removed
>  fwupdate     (LZMA/uncomp., 0x0adc00 - 0x0b2940): removed
>  ptt          (LZMA/uncomp., 0x0b2940 - 0x0c8000): removed
>  touch_fw     (LZMA/uncomp., 0x0c8000 - 0x130000): removed
> The ME minimum size should be 364544 bytes (0x59000 bytes)
> Checking the FTPR RSA signature... VALID
> Done! Good luck!

corna, 2018-01-10:
I don't think so, see [here](https://github.com/corna/me_cleaner/issues/61#issuecomment-327871037).

archfan, 2018-01-11:
Interesting. Thanks!

Do you think there is a way to replace the ME blob of the bios file with an updated one? I have no external flasher at hand and this would be my preferred way of flashing.

archfan, 2018-01-11:
I found this neat tool "UEFITool". Tried to replace the ME image of the bios file but unfortunately the ME file has to be of the exact same size. The newer ME blob is a bit larger than the old blob.

reconstructIntelImage: reconstructed body size 95C000h (9814016) is smaller then original 1000000h (16777216) 
reconstructIntelImage: reconstructed body size 1005000h (16797696) is bigger then original 1000000h (16777216) 

So I should probably truncate first and then add enough padding to get the blob to the desired size.

archfan, 2018-01-11:
Just flashed my modified image. It worked flawlessly!

closed 2018-01-11 by archfan

corna, 2018-01-11:
Great!
However you should have used `me_cleaner.py -r -t` to create a minimum-size ME image, and then add padding to it, instead of truncating it ;)