corna/me_cleaner #152: "Unknown image" for Lenovo T420 ME7.1 updater

victorn1, 2018-01-14:
I would like to patch ME for Lenovo T420. 

There is ME firmware update https://pcsupport.lenovo.com/ru/ru/products/laptops-and-netbooks/thinkpad-t-series-laptops/thinkpad-t420/downloads/ds014884 on Lenovo site. After the first run this program unpacks its contents to C:\DRIVERS\WIN\ME 

There is ME bin file ME7_5M_UPD_Production.bin  However this file is not recognized by me_cleaner.py with "Unknown image" error message. 

How can I unpack/repack this file to patch it? Would it be possible to flash the patched file with this FWUpdLcl.exe applet?

skochinsky, 2018-01-14:
you can't use the update binary; you need to update the ME first then use mme_cleaner on the full flash image

victorn1, 2018-01-15:
Then why, for ME images from https://www.win-raid.com/t596f39-Intel-Management-Engine-Drivers-Firmware-amp-System-Tools.html can I use me_cleaner? These images are correctly recognized  by me_cleaner.

skochinsky, 2018-01-15:
Some of those images are full ME region images and can be processed by me_cleaner, however you'll have to flash them directly to the flash chip as FWUpdLcl update will fail due to missing NFTP

victorn1, 2018-01-16:
Can I flash these patched images with "fpt.exe -me"?

skochinsky, 2018-01-16:
> Can I flash these patched images with "fpt.exe -me"?

Yes, it should work if you have write access to the ME region.

corna, 2018-01-16:
Unfortunately Lenovo laptops usually don't have write access to the me region

tlaurion, 2020-11-09:
> Then why, for ME images from https://www.win-raid.com/t596f39-Intel-Management-Engine-Drivers-Firmware-amp-System-Tools.html can I use me_cleaner? These images are correctly recognized by me_cleaner.

Personally tested the 3 images for 7.x ME from [above link]( https://www.win-raid.com/t596f39-Intel-Management-Engine-Drivers-Firmware-amp-System-Tools.html):
```
user@xx20-tests:~/heads$ ~/me_cleaner/me_cleaner.py -r -t -O ~/heads/blobs/xx20/me.bin ~/QubesIncoming/disp5838/7.0.1.1205_SLM_PRD_UPD.bin
Unknown image
user@xx20-tests:~/heads$ ~/me_cleaner/me_cleaner.py -r -t -O ~/heads/blobs/xx20/me.bin ~/QubesIncoming/disp5838/7.1.80.1214_1.5MB_PRD_UPD.bin 
Unknown image
user@xx20-tests:~/heads$ ~/me_cleaner/me_cleaner.py -r -t -O ~/heads/blobs/xx20/me.bin ~/QubesIncoming/disp5838/7.1.92.3273_5MB_PRD_UPD.bin
Unknown image
```

Thrilleratplay, 2020-11-27:
For anyone interested, I created a script that uses a lot of me_cleaner's code with the ME 7.1 update file from Lenovo to create the equivalent of what me_cleaner would produce if minified.  [It can be found here](https://github.com/Thrilleratplay/me7_update_parser).