corna/me_cleaner #153: (Possibly) bricked mainboard

c0d3z3r0, 2018-01-16:
Hi,

looks like I've bricked my mainboard while experimenting with me_cleaner...
I use a FT232H and a SOIC clip for external flashing.
Before applying me_cleaner I made a backup of course (and checked I twice!).

After one non-working flash procedure I needed to restore the original image.
When flashing was done I forgot to remove the SOIC clip and plugged in the power cord.

Now the PC does not boot anymore... no fans spinning, nothing.
Any ideas how to fix that?
Thank you so much!

corna, 2018-01-16:
Things that could be dead:
 * SPI memory
 * 3.3V regulator on the MB
 * SPI data lines of the PCH

Can you check:
 * if you are still able to read the SPI with the programmer
 * (if you have a multimeter) if the 3.3V pin of the SPI chip is powered when the board is powered on
 * (if you have an oscilloscope) if the four SPI lines toggle during the power on

c0d3z3r0, 2018-01-16:
SPI flash seems to be OK, I can read/erase/write/verify without any problems.
The chip is powered correctly when the power cord is plugged in.
3.3V on pins 1,2,3,7,8.

(by the way: thanks for the great talk @34C3!)

corna, 2018-01-16:
You're welcome :)

The you should check if the lines are correctly toggled with an oscilloscope/logic analyzer:
 * CS (1) should go down during read/write
 * CLK (6) should oscillate at some MHz (from 17 to 50 MHz IIRC, depending on the generation)
 * MISO (2) should be ok (as you can read/write the memory with the external programmer)
 * MOSI (5) shouldn't be stuck at GND or 3.3

c0d3z3r0, 2018-01-16:
As I don't have an LA I tried to get at least some information with FT232H in bitbang mode...
The only thing I can tell this way is that the pins are toggling

c0d3z3r0, 2018-01-16:
I have to correct this... MOSI seems to be stuck at 3.3V

c0d3z3r0, 2018-01-16:
Works again after removing bios battery for >10 min o.O

closed 2018-01-16 by c0d3z3r0

corna, 2018-01-16:
Oh right, that should have been the first suggestion ;)