corna/me_cleaner #157: x260 working with 25 sec boot delay

c0d3z3r0, 2018-01-19:
x260 working but boot 25 sec boot delay
latest UEFI/Bios R02ET63W (1.36)

me_cleaner parameters: -M /tmp/me -S -d -r -t

skochinsky, 2018-01-19:
All ME firmwares are the same for all OEMs. Only Apple gets somewhat personalized ("slim") builds AFAIK.

corna, 2018-01-19:
Generally you should update the ME firmware before using me_cleaner.

Single modules can't be replaced (the list of the hashes is signed), you can only replace whole partitions. However it doesn't make much sense, as using me_cleaner on the new image leads to the same result (only the ftpr partition is kept).

I don't know the format of the ME images, but I doubt it's unsigned.

c0d3z3r0, 2018-01-19:
@skochinsky Looks like there _is_ a difference between some images.
I compared Lenovo x260 Corporate image and Lenovo P320 Corporate image, both at v11.8.50.3425.
Maybe this is because of different CPUs (i5 vs Xeon).

However, the format of the ME update image and dumped image is exactly the same. The only thing I don't know yet is if there is any OEM data (I don't think so even when there seem to be differences).
That means one could replace the ME image in the dump before applying me_cleaner so there is no need to go through the uncomfortable "need-to-install-windows-for-me-update process" on a Linux machine.

I'll try that...

c0d3z3r0, 2018-01-19:
Oops. Replaced ME image with UEFItool, applied me_cleaner and flashed the image.
x260 turns on shortly as normal for 3 secs, then turns off completely and cannot be enabled anymore.
Removing all batteries did not help. Flashing back the backup image did not work also.

Any ideas? o.o

corna, 2018-01-20:
Have you already tried to restore the original dump?

corna, 2018-01-20:
> Looks like there is a difference between some images.

Remember that ME has an internal file system (`MFS`/`EFFS`), so even two dumps of the same machine can be different. You should check which partitions are not identical.

c0d3z3r0, 2018-01-20:
Yep, already restored the backup but no sound, no light, no nothing...

c0d3z3r0, 2018-01-20:
What is really strange is that it turned "on" for some seconds and now seems completely dead... A bad image should not cause any hardware damage so this must be a software problem anyhow.

corna, 2018-01-20:
Have you already removed the RTC battery for ~10 minutes to clear the CMOS?

c0d3z3r0, 2018-01-20:
Yes, all batteries (main, rtc) were disconnected over night

c0d3z3r0, 2018-01-21:
The only thing working is the 3 times led blinking when attaching the ac... hmm could this problem be EC (embedded controller) related?

c0d3z3r0, 2018-01-21:
I opened my x260 again. When pressing the power button the fan twitches shortly. This can be repeated after some seconds waitinh...

c0d3z3r0, 2018-01-21:
x260 working again after repeatedly flashing the original image and having all batteries (again) detach for some hours. I countinue my tests with ME update via image

corna, 2018-01-21:
Well, that's weird...
At least it came back

c0d3z3r0, 2018-01-22:
Looks like ME cannot be updated by simply replacing the image... I'll try to compare them and find the differences that prevents that

c0d3z3r0, 2018-01-22:
I now was able to update ME by simply replacing the image with ifdtool.
The system boots fine but something seems to be wrong. The UEFI/BIOS does not show any version number for ME anymore and there is a boot delay just like me_cleaner would have been applied.
Updating ME this way does not seem to work correctly.

I found a way for updating ME without a full installation of windows: https://www.flamingspork.com/blog/2017/11/22/updating-windows-management-engine-firmware-on-a-lenovo-without-a-windows-install/
Just don't use the x86 iso as described but the x64 one. All the wine stuff can be skipped by using innoextract, cabextract and wimtools on Linux.

Maremagnum, 2018-01-23:
@c0d3z3r0  how did you flash the x260? I was just trying to flash one yesterday but after disassembling the mainboard and looking under the black tape that covers it.. i found the rom chips on board are smd (one infineon and another winbond), so there is no way to tap/connect the pins in them easily without desoldering :/

I feel for this model they actively tried for people to not touch the bios/me, since in the x270 the chips are dip8 and way easier to find.. suspicious..

c0d3z3r0, 2018-01-23:
@Maremagnum yep, I ordered a SOIC clip from china and waited weeks for it. When I finally received it and opened my x260 I was like o.O WTF!?

AFAIK x260 is the only one with WSON chip on the bottom side of the mainboard... all others - older and newer ones - have SOIC on the top.

I soldered 8 wires to the chip...

[image]

c0d3z3r0, 2018-01-23:
@corna 
Intel ME v11.0.0.1194 works, v11.8.50.3425 does not work with me_cleaner.
Only `-s` works, `-S` + normal module removal does not work. Any ideas?

c0d3z3r0, 2018-01-31:
@skochinsky Do you have an idea what FTUP and touch_fw do?

skochinsky, 2018-01-31:
Not 100% sure but `touch_fw `sounds like "touchscreen controller firmware". Not sure why it could be required for boot, but apparently it is on your system... 

FTUP seems to be a backup copy of NFTP+WCOD and contains  some non-critical modules like amt, cls, dal, ish, nfc, wifi firmware and so on.

c0d3z3r0, 2018-01-31:
Hmm, interesting. If this is correct, `touch_fw` should not be required at all on my system since I don't have a touch screen...

The thing I do not understand is why FTUP is needed on newer versions but not on e.g. 11.7.4.3314.

resetcoder, 2018-05-31:
Hello,

I flashed anothet x260 and I'd just share some information and as a question.

First of all: Cleaning and flashing was successful for the first time, everything worked well, just the ~30s startup delay is what I'm having, too. I've just noticed it seems like the bios is working hard on something in the background and/or the system clock is very very low during startup, because in the bios all the screen displaying and the keyboard sensing is laggy (like on an old XT computer, or worse), but during the boot it gets fast and normal, the laptop is having good performance.

Other thing is: few months after the flashing I've got the same problem, first the laptop didn't turn on, had only the 3 times power button blinking, after disassembling and reassembling it worked again. Few weeks later it didn't turn on at all, only the blinking, nothing else. Disassembling didn't help, but I thought there must be a contact failure somewhere, so after checking all the connectors I decided to resolder the flash, and voila, it turned on and works as before, so if the symptoms are this, it worth a check of the flash solderings.

Question: could anyone find a solution for the ~30s startup delay?

c0d3z3r0, 2018-05-31:
Ah, interesting.. I didn‘t check the soldering of the flash itself.. Is it WSON or SOIC-8 for you?

I did not found a solution for the delay, yet.
What ME Version do you have?

resetcoder, 2018-05-31:
> Ah, interesting.. I didn‘t check the soldering of the flash itself.. Is it WSON or SOIC-8 for you?
>
> I did not found a solution for the delay, yet.
> What ME Version do you have?

It has WSON packaging. Reflowing the solderings completely fixed it.