corna/me_cleaner #159: g73sw laptop AMI bios Unknown image

Morghul, 2018-01-21:
https://www.asus.com/Laptops/ROG-G73SW/HelpDesk_BIOS/
AMI bios v4 tools can dump/flash the .bin and configure some settings in the bios normally hidden, lots of ME stuff related, but me_cleaner says "unknown image" on both the download from the site, and the dumped bin using AFU, UEFI tools can read the update/bin too so its valid i think, can you add support for this?

skochinsky, 2018-01-21:
This seems to contasin only the BIOS region, not the full flash image. You should update your bios to it, then use an external flasher to read the full image for `me_cleaner`.

Morghul, 2018-01-21:
so when the changelog for the bios file says "3.Update 1.5MB ME Firmware to 7.0.4.1197." it means they're applying a patch of some sort and not flashing a full me or something?

skochinsky, 2018-01-21:
it probably contains the firmware inside the BIOS and will flash it on first reboot.

Morghul, 2018-01-21:
thank you for looking, i will have to grab a clip