corna/me_cleaner #161: Intel Sensor Hub After HAP and/or Cleaning

ilikenwf, 2018-01-28:
While cleaning works just fine, as does HAP for 11.8.50.3399 of the ME, the real issue is that on any version this seems to cripple the Intel Sensor Hub, making accelerometers and such not work, even under Linux, as I don't think there's a direct access to these devices without the hub.

I'm actually somewhat surprised that HAP doesn't allow the sensor hub firmware to load and start up; one would think government types would want to retain these functions despite using HAP devices. I wonder if a workaround could be found, perhaps a way to dump and then force load the ISH firmware from the Linux kernel?

Either way, just a heads up. Since 11.8.50.3399 supposedly mitigates all current known CVE's I guess I'll leave it enabled, begrudgingly, because I want accelerometer/sensor support.

I wonder if it would be worth it for me to contact the Intel employees who wrote the kernel driver to see if they have any ideas...

skochinsky, 2018-01-28:
The ME firmware contains modules with names like `ish_bup` or `ish_srv` so it seems it's responsible for at least some of the ISH functionality and you need to have it running if you need sensors. I guess it should be possible to modify `me_cleaner` to whitelist these modules so they're not removed and see what happens...

corna, 2018-01-28:
I'm going to review and merge #160 in the next days, then we can freely keep or not specific partitions or modules and verify it.

You can use @c0d3z3r0's tree in the meanwhile.

ilikenwf, 2018-01-31:
So I tried leaving all modules in, and upon boot the ISH kernel module under Linux still fails/times out, so it looks like I am stuck, at least presently, using it with the ME enabled.

On the upside, this model doesn't include AMT.

Here are the partitions and modules inside my 11.8 image...any suggestions?

```
 FTPR (0x00001000 - 0x0000a8000, 0x000a7000 total bytes): NOT removed
 FTUP (0x00110000 - 0x0001bc000, 0x000ac000 total bytes): removed
 DLMP (0x000a6000 - 0x0000a9000, 0x00003000 total bytes): removed
 PSVN (0x00000e00 - 0x000001000, 0x00000200 total bytes): removed
 IVBP (0x0010c000 - 0x000110000, 0x00004000 total bytes): removed
 MFS  (0x000a8000 - 0x00010c000, 0x00064000 total bytes): removed
 NFTP (0x00110000 - 0x0001bc000, 0x000ac000 total bytes): removed
 ROMB (      no data here      , 0x00000000 total bytes): nothing to remove
 FLOG (0x001bc000 - 0x0001bd000, 0x00001000 total bytes): removed
 UTOK (0x001bd000 - 0x0001bf000, 0x00002000 total bytes): removed
 ISHC (0x001bf000 - 0x0001ff000, 0x00040000 total bytes): removed

Reading FTPR modules list...
 FTPR.man     (uncompressed, 0x003000 - 0x003bd0): NOT removed, partition manif.
 rbe.met      (uncompressed, 0x003bd0 - 0x003c66): NOT removed, module metadata
 kernel.met   (uncompressed, 0x003c66 - 0x003cf4): NOT removed, module metadata
 syslib.met   (uncompressed, 0x003cf4 - 0x003d58): NOT removed, module metadata
 bup.met      (uncompressed, 0x003d58 - 0x004302): NOT removed, module metadata
 pm.met       (uncompressed, 0x004302 - 0x0043b0): NOT removed, module metadata
 vfs.met      (uncompressed, 0x0043b0 - 0x004d10): NOT removed, module metadata
 evtdisp.met  (uncompressed, 0x004d10 - 0x004e9e): NOT removed, module metadata
 loadmgr.met  (uncompressed, 0x004e9e - 0x004fc6): NOT removed, module metadata
 busdrv.met   (uncompressed, 0x004fc6 - 0x00534a): NOT removed, module metadata
 gpio.met     (uncompressed, 0x00534a - 0x005494): NOT removed, module metadata
 prtc.met     (uncompressed, 0x005494 - 0x005644): NOT removed, module metadata
 policy.met   (uncompressed, 0x005644 - 0x005806): NOT removed, module metadata
 crypto.met   (uncompressed, 0x005806 - 0x005990): NOT removed, module metadata
 heci.met     (uncompressed, 0x005990 - 0x005b5c): NOT removed, module metadata
 storage.met  (uncompressed, 0x005b5c - 0x005e58): NOT removed, module metadata
 pmdrv.met    (uncompressed, 0x005e58 - 0x005f7c): NOT removed, module metadata
 maestro.met  (uncompressed, 0x005f7c - 0x006066): NOT removed, module metadata
 fpf.met      (uncompressed, 0x006066 - 0x006172): NOT removed, module metadata
 hci.met      (uncompressed, 0x006172 - 0x006274): NOT removed, module metadata
 fwupdate.met (uncompressed, 0x006274 - 0x00637c): NOT removed, module metadata
 ptt.met      (uncompressed, 0x00637c - 0x006488): NOT removed, module metadata
 touch_fw.met (uncompressed, 0x006488 - 0x0065c0): NOT removed, module metadata
 rbe          (Huffman     , 0x0065c0 - 0x009300): NOT removed, essential or whitelisted
 kernel       (Huffman     , 0x009300 - 0x0192c0): NOT removed, essential or whitelisted
 syslib       (Huffman     , 0x0192c0 - 0x02b180): NOT removed, essential or whitelisted
 bup          (Huffman     , 0x02b180 - 0x054e00): NOT removed, essential or whitelisted
 pm           (LZMA/uncomp., 0x054e00 - 0x057400): removed
 vfs          (LZMA/uncomp., 0x057400 - 0x05f380): removed
 evtdisp      (LZMA/uncomp., 0x05f380 - 0x060d80): removed
 loadmgr      (LZMA/uncomp., 0x060d80 - 0x063b80): removed
 busdrv       (LZMA/uncomp., 0x063b80 - 0x065440): removed
 gpio         (LZMA/uncomp., 0x065440 - 0x066580): removed
 prtc         (LZMA/uncomp., 0x066580 - 0x067140): removed
 policy       (LZMA/uncomp., 0x067140 - 0x06bd00): removed
 crypto       (LZMA/uncomp., 0x06bd00 - 0x079900): removed
 heci         (LZMA/uncomp., 0x079900 - 0x07d800): removed
 storage      (LZMA/uncomp., 0x07d800 - 0x081e40): removed
 pmdrv        (LZMA/uncomp., 0x081e40 - 0x083000): removed
 maestro      (LZMA/uncomp., 0x083000 - 0x084dc0): removed
 fpf          (LZMA/uncomp., 0x084dc0 - 0x0867c0): removed
 hci          (LZMA/uncomp., 0x0867c0 - 0x087080): removed
 fwupdate     (LZMA/uncomp., 0x087080 - 0x08be00): removed
 ptt          (LZMA/uncomp., 0x08be00 - 0x0a1440): removed
 touch_fw     (LZMA/uncomp., 0x0a1440 - 0x0a8000): removed
```

ilikenwf, 2018-01-31:
I've tried keeping the modules and the relevant partitions but unfortunately, the ISH doesn't come up.

skochinsky, 2018-01-31:
maybe try not removing other partitions (NFTP, FTUP, IVBP, MFS). You'll  have to do some experiments to find the minimal set of partitions/modules you need to keep.

c0d3z3r0, 2018-01-31:
@ilikenwf Try to keep FTUP partition and ish_* modules

ilikenwf, 2018-02-01:
@c0d3z3r0 

./me_cleaner.py -k -w ISHC,FTUP

Still gives me "ishtp-ish: timed out waiting for FW initiated reset

ilikenwf, 2018-02-01:
Even this doesn't work:

./me_cleaner.py -b MFS -k

ilikenwf, 2018-02-01:
After opening my bios image in the flash image tool (win-raid forums has the latest one), I can at least rest assured that my machine doesn't support AMT or networking from the ME subsystem, so I guess it's only purpose in being here is to be a huge attack surface should more exploits be found, and to manage power/temp/sensor hub operations.

That said the FIT tool can extract the ISH firmware binary, so I'm curious if we could strip the ME and modify the kernel module to load up the ISH binary on it's own without having to rely on ME to do anything...doubtful, but still an interesting thought?

c0d3z3r0, 2018-02-01:
I don't think there is a way to load the firmware from linux.
I wanted to have a look at ISH on my systems but neither Lenovo x260 nor Dell E5550 seem to have ISH.

ilikenwf, 2018-02-01:
I think some cheaper chromebooks and laptops out there may have ISH, it's Skylake and newer that have it as an option.

I'm curious if Google used ISH or not in the Pixelbook, since they use coreboot on their chromebooks. If they used it, did they disable the ME or keep it for the hub?

KL1RL, 2019-03-24:
Confirmed issue is present on Lenovo X1 Yoga Gen3.  Attempted soft clean by setting HAP bit only.  Me version didn't show in BIOS and intelmetool couldn't find me device afterwards.  Kernel continued to report:

`Mar 24 09:19:50 censored kernel: intel_ish_ipc 0000:00:13.0: [ishtp-ish]: Timed out waiting for FW-initiated reset`
`Mar 24 09:19:50 censored kernel: intel_ish_ipc 0000:00:13.0: ISH: hw start failed.`

myfreeweb, 2019-10-05:
@ilikenwf (hopefully this is still relevant, months later :D) Google does not seem to use ISH at all — all the sensors are connected to the Chrome EC which runs user-controllable FOSS firmware. They did not disable the ME, but they use the smaller 2mb image. Disabling works fine with `-S -w MFS` (#300)

ilikenwf, 2019-10-06:
Yes, indeed.

I recently got the docs for my EC so hopefully I can figure something out to control the fans natively, then strip the ME portion.

ihsakashi, 2020-05-21:
I have ux360uak with an unlocked bios. I've disabled ME from there and run into the same issue. Is there anything I can do to help?