corna/me_cleaner #162: ME partitions reappearing after flashing cleaned BIOS?

khromov, 2018-01-29:
I have an Asus P7P55D motherboard. I've ran the me_cleaner on the BIOS file and it successfully removed the ME partitions. I then flash the BIOS (using the Asus Updater application for Windows or the "EZ Flash" functionality built into-the BIOS which reads the upgrade file from an attached USB stick). 

The Asus Update utility has a functionality to read the BIOS and save it back to a file. After doing this, I find that that ME partitions reappear and the image can again be cleaned using me_cleaner. 

The possibilities are:

1.) me_cleaner cleans the BIOS, but both EZ Flash and the Asus Update utility put the ME partitions back in during the flashing process.
2.) The ASUS Updated software puts in 
3.) The me_cleaner software is giving incorrect results. 
4.) The partitions are supposed to reappear, and this is expected behaviour.

I find all of the first three options somewhat unlikely, so I'm looking for advice. 

BIOS file:
http://dlcdnet.asus.com/pub/ASUS/mb/LGA1156/P7P55D/P7P55D-ASUS-2101.zip

First run on clean BIOS:
```
Full image detected
The ME/TXE region goes from 0x10000 to 0x30000
Found FPT header at 0x10010
Found 5 partition(s)
Found FTPR header: FTPR partition spans from 0xf000 to 0x10000
ME/TXE firmware version 6.0.20.1185
Public key match: Intel ME, firmware versions 6.x.x.x
Reading partitions list...
 FAD0 (0x00000400 - 0x000001000, 0x00000c00 total bytes): removed
 NFTP (0x00001000 - 0x00000f000, 0x0000e000 total bytes): removed
 FTPR (0x0000f000 - 0x000010000, 0x00001000 total bytes): NOT removed
 FAD1 (0x00010000 - 0x000010c00, 0x00000c00 total bytes): removed
 IGRT (      no data here      , 0xffffffff total bytes): nothing to remove
Removing partition entries in FPT...
Removing EFFS presence flag...
Correcting checksum (0x6a)...
Reading FTPR modules list...
 ROMP             (Huffman, fragmented data, ~1 KiB   ): NOT removed, essential
The ME minimum size should be 81920 bytes (0x14000 bytes)
The ME region can be reduced up to:
 00010000:00023fff me
Checking the FTPR RSA signature... VALID
Done! Good luck!
```

Second run on the same BIOS image:

```
Full image detected
The ME/TXE region goes from 0x10000 to 0x30000
Found FPT header at 0x10010
Found 1 partition(s)
Found FTPR header: FTPR partition spans from 0xf000 to 0x10000
ME/TXE firmware version 6.0.20.1185
Public key match: Intel ME, firmware versions 6.x.x.x
Reading partitions list...
 FTPR (0x0000f000 - 0x000010000, 0x00001000 total bytes): NOT removed
Removing partition entries in FPT...
Removing EFFS presence flag...
Correcting checksum (0x6a)...
Reading FTPR modules list...
 ROMP             (Huffman, fragmented data, ~1 KiB   ): NOT removed, essential
The ME minimum size should be 81920 bytes (0x14000 bytes)
The ME region can be reduced up to:
 00010000:00023fff me
Checking the FTPR RSA signature... VALID
Done! Good luck!
```

After flashing the BIOS (Via BIOS - EZ Flasher or AsusUpdate utility)  and then reading it again:

``` 
Full image detected
The ME/TXE region goes from 0x10000 to 0x30000
Found FPT header at 0x10010
Found 5 partition(s)
Found FTPR header: FTPR partition spans from 0xf000 to 0x10000
ME/TXE firmware version 6.0.20.1185
Public key match: Intel ME, firmware versions 6.x.x.x
Reading partitions list...
 FAD0 (0x00000400 - 0x000001000, 0x00000c00 total bytes): removed
 NFTP (0x00001000 - 0x00000f000, 0x0000e000 total bytes): removed
 FTPR (0x0000f000 - 0x000010000, 0x00001000 total bytes): NOT removed
 FAD1 (0x00010000 - 0x000010c00, 0x00000c00 total bytes): removed
 IGRT (      no data here      , 0xffffffff total bytes): nothing to remove
Removing partition entries in FPT...
Removing EFFS presence flag...
Correcting checksum (0x6a)...
Reading FTPR modules list...
 ROMP             (Huffman, fragmented data, ~1 KiB   ): NOT removed, essential
The ME minimum size should be 81920 bytes (0x14000 bytes)
The ME region can be reduced up to:
 00010000:00023fff me
Checking the FTPR RSA signature... VALID
Done! Good luck!
```

corna, 2018-01-29:
> 1.) me_cleaner cleans the BIOS, but both EZ Flash and the Asus Update utility put the ME partitions back in during the flashing process.

More like "EZ Flash detects an update image with a corrupted ME region and refuses to flash it (without telling you, it has already happened to other users)". But also "EZ flash doesn't flash a new ME image if the ME version is the same".

> 2.) The ASUS Updater software puts the ME partitions in when dumping the BIOS from the flash to a file.

Unlikely. However it's possible that the ASUS BIOS has a backup firmware that gets flashed back when something's wrong (like a corrupted ME image).

> 3.) The me_cleaner software is giving incorrect results.

Unlikely, _me_cleaner_ is pretty mature now.

> 4.) The partitions are supposed to reappear, and this is expected behaviour.

Nope.

You can try with an external flasher, it'll probably fix your issues: I see that your motherboard has a socketed DIP-8 BIOS chip, so it's really a piece of cake with a [CH341A](https://www.ebay.com/itm/CH341A-Series-24-EEPROM-Writer-25-SPI-routing-BIOS-LCD-Flash-USB-Programmer-D/112049160886?hash=item1a16a682b6:g:XUUAAOSwke9aTaXf).

khromov, 2018-01-29:
@corna Thank you for the feedback. It does sound likely that the Asus Updater wouldn't flash the ME portion of the BIOS when no update is available!

Could you recommend a flasher that includes a clip-on tool to avoid having to desolder the BIOS?

Thanks!

corna, 2018-01-30:
From [here](http://www.ixbt.com/mainboard/asus/p7p55d-le/board.jpg) it seems to me that the chip is socketed, all you have to do is to pull it (carefully, don't bend the pins) and insert it in the ZIF socket of the CH341A, no desoldering required.

platomav, 2018-01-30:
@corna me_cleaner is indeed bugged as ME 2-6 did not have a size at the last $FPT entry so it acts like there are no data to remove. In this case, Ignition Runtime Partition (IGRT) is still there after cleanup. This is not the reason behind khromov issue but a bug nonetheless.

By the way, systems with ME 6 Ignition firmware do not need me_cleaner. Just add 0xFF at the entire Engine region as explained [here](https://www.win-raid.com/t2443f39-Guide-Disable-amp-Remove-Intel-Ibex-Peak-ME-Ignition-firmware.html). Maybe me_cleaner can do that automatically when ME 6 Ignition IBX/CCK is detected.

corna, 2018-01-30:
> me_cleaner is indeed bugged as ME 2-6 did not have a size at the last $FPT entry so it acts like there are no data to remove. In this case, Ignition Runtime Partition (IGRT) is still there after cleanup. This is not the reason behind khromov issue but a bug nonetheless.

I'll fix it, thank you.

> By the way, systems with ME 6 Ignition firmware do not need me_cleaner. Just add 0xFF at the entire Engine region as explained here. Maybe me_cleaner can do that automatically when ME 6 Ignition IBX/CCK is detected.

Yep, I'll do it as well. How can I detect if an image is  Ignition or not (without using the size)?

platomav, 2018-01-30:
> Yep, I'll do it as well. How can I detect if an image is Ignition or not (without using the size)?

Check if variant = ME, Major = 6 and FTPR --> $SKU = 00000000 which signifies Ignition. The $SKU field for ME 2-6 consists of Tag [0x4] + Size in dwords [0x4] + SKU [0x4].

corna, 2018-01-30:
> $SKU = 00000000

That was the missing part, thanks. However I don't see the offset of the $SKU section in the partition header, any idea how to find it without looking for its tag?

corna, 2018-01-31:
@platomav Done in `dev`, thanks for the suggestions

platomav, 2018-01-31:
@corna Nice. A small fix is needed as it fills one extra byte at the end of the region.

corna, 2018-02-01:
You're right, fixed

khromov, 2018-02-02:
Thanks for the help @corna. I'll update this issue once I've tried flashing using the hardware flasher and report back.

xinterix, 2018-08-01:
I have Asus Mark 1 z270 and I have successfully cleaned up ME.bin from the recent vulnerability patch (MEUpdateTool_UI_20171103_TP.zip). IT all works, no partitions reappearing. Extract the zip, and in sub-folder titled FW is the ME.bin. Use me_cleaner on that original file(only works without the -S -s options) and after copy the modified.bin back the FW and rename it to its original ME.bin. To flash it just run the MEUpdateTool.exe from the rootfolder of the extracted zip. It will sucessfully flash. make sure to unplug when reboot and battery, reconfigure bios.

xinterix, 2018-08-01:
I still can't get any tools do a full image dump which allow the -S option  which require full image. So this does not change HAP, correct??? Does anyone have a method to get full dump with this board??