corna/me_cleaner #163: Soon the owner of Dell Precision M4800 - will it work?

peterwilli, 2018-01-30:
Hi there,

I'm getting a Dell Precision M4800 with Core i7 (www.dell.com/en-us/work/shop/dell-laptops-and-notebooks/precision-m4800/spd/precision-m4800-workstation).

I was wondering, did anyone try me_cleaner for this laptop and did you have any success?
If no one has tried, could you point me in the right direction, so I can try myself? (I'll post the results here :))

It should arrive by friday.

Apocryp, 2020-07-29:
@peterwilli  How did it go? Did you manage to install me_cleaner on M4800? @garytinn reported success on M4600. 

https://github.com/corna/me_cleaner/issues/3#issuecomment-610903045

peterwilli, 2020-08-06:
> @peterwilli How did it go? Did you manage to install me_cleaner on M4800? @garytinn reported success on M4600.
> 
> [#3 (comment)](https://github.com/corna/me_cleaner/issues/3#issuecomment-610903045)

Ok uhm. This is really awkward, but when I got it, I found out it was already riddened of ME. I have no idea if the previous owner did that themselves, or that their IT department did it. But somehow, it was already gone. I therefore didn't want to mess with it naymore.

closed 2020-08-06 by peterwilli