corna/me_cleaner #166: Thinkpad T530: Setting the HAP/AltMeDisable bit makes the BIOS settings inaccessible

Housies, 2018-02-09:
I have a Thinkpad W530, but I replaced the motherboard with one from a T530, so for all intents and purposes, it's a T530. It has an i7-3720QM. When trying to read the chip with a Raspberry Pi v1 flashrom worked intermittently, giving different results and sometimes not seeing the chip at all. The only way I could get it to work was to flash with the computer turned on and the 3.3v from the Pi connected.

With the -S argument, it boots, intelmetool says ME is disabled, and it stays on beyond half an hour, but the BIOS settings screen is inaccessible. If I press the button to access the settings, it hangs on a black screen with a cursor. It seems to remember the settings I set before flashing me_cleaner.

With no arguments, it works as before except that the BIOS settings work normally.

matt123b, 2018-02-26:
I've also had BIOS problems on the Lenovo xx30 series boards after flashing. For example, stripping the IME firmware and keeping the stock BIOS on a ThinkPad X230 Core i5 3320M causes frequent boot hang issues. I wasn't able to determine an exact cause but I think that it's down to a buggy UEFI implementation, since manufacturers often take shortcuts and stray from the standards on these things. So to fix this, try totally restoring the IME and stock BIOS firmware, and then find a disk with Windows 7 or later and run the Lenovo BIOS update utility on it, and make sure that the BIOS is at the newest version. This fixed it for me on the X230 when I went from version 2.68 to 2.70, and then started the me_cleaner process from scratch. Another option you might have is to use Coreboot, though I don't know the status of Coreboot support for your board (if there's any at all). If all else fails, buy a new flasher clip. Be sure to clean the GPIO pins on the RasPi and also clean the pin connectors on the inside of the clip and the ones on the ThinkPad's ROM chip. Use a q-tip and some alcohol pads and be gentle with it. There could be dust and dirt in there that's messing with your connections.

These are experimental chipset firmware mods and there are no absolutes with me_cleaner, so this is all trial and error. Start by ruling out all other possibilities before deciding that the board is incompatible. Good luck.

Housies, 2018-02-26:
I tried restoring the stock BIOS and reapplying me_cleaner several times. I'm already using version 2.70. Once I got it working the Pi recognised the chip and consistently dumped images with the same hashes, so I doubt it's the connection. I assume it doesn't work with the computer turned off because the Pi version 1 outputs less current from the GPIO pins than later versions.

7XFugvwR, 2020-09-21:
Old Topic but I encoutered the same behaviour with my X230.

With "-S"-flag working but no BIOS-menu anymore, without "-S"-flag BIOS is accessible again, laptop is still working, me is not functional.