corna/me_cleaner #167: me_cleaner has problems on Lenovo ThinkPad X201

haxalicious, 2018-02-11:
Running me_cleaner with the -S flag seems to not boot, however it could be that it just takes much, much longer to boot. Using the -s flag does seem to work, however with a few side effects:

1. Turbo Boost does not work. It is still listed as supported and active in cpupower, but it does not actually work. I do not know whether this is a problem with the EC firmware, the BIOS or ME itself, but I have not been able to get it working no matter what I do.

2. The system takes much longer to boot up. This is not really a problem though and it seems to have something to do with disabling ME.

3. The fan is not controlled by the EC (Embedded Controller). When the system is booted, it will stay off no matter what the temperature is. This can be worked around on Linux using a program such as fancontrol and setting the manual control flag on the thinkpad_acpi driver to control the fan through software. However, this is not an ideal solution because sometimes inconsistent as it stops working whenever the order of the sensors changes, which seems to happen at random. I suspect this is a problem with the EC firmware or possibly the BIOS.

I have not tried running me_cleaner without either the -s or -S flag set.

Stitch626, 2018-05-13:
I can confirm the issues. I'm using ThinkPadFanControl on Windows 7. The software doesn't display any temperature while CPUID HwMonitor displays at least the core temperature. And yes, Turboboost seems to stop working too. I've tried the -S flag, I'll try another ones and post my results here.

Stitch626, 2018-05-18:
UPDATE: 
It seems that Turboboost still works on Windows, but the thermal throttling and the high-temp shutdown are disabled. I can use this by now because I was able to reconfigure TPFanControl on Windows to take care of my temps :)

But there is a screenshot:

[image]

Stitch626, 2018-08-12:
I had the thermal compbound already replaced, and the heatsink is the all-copper one ^^
Turbo boost worked oob but I may have the same power plan as you ("Battery Optimized").
And complaining about my OS as a bigger security hole than the hardware is a bit rude, no?!