corna/me_cleaner #169: ME Disable Bits MCHSTRP0[0], ICHSTRP0[0], MCHSTRP0[7]

c0d3z3r0, 2018-02-14:
**ICHSTRP0[0]** (==PCHSTRP0[0]) is the so called "ME_Disable Bit" used in me_cleaner.
There is an additional note in section 22.2.5.1 of the _Intel ICH9 EDS_ document [1] saying not only **ICHSTRP0[0]** has to be set but also **MCHSTRP0[0]** (see section 22.2.6.1).
@corna @h0t @ptsecurity @skochinsky Did you know that bit? Shouldn't it be set by me_cleaner, too?

The _ICH9 Specification Update_ [2] even mentions a third bit in a LAN PHY workaround, **MCHSTRP0[7]**:
"MCHSTRP0 bit 7 in the SPI flash descriptor **can be set to disable all other ME FW
based features**, while keeping the ME FW workaround enabled."

MCHSTRP0 controls ME and TPM features. Since **MCHSTRP0[31:3]** is reserved (*cough cough*), there may be many more bits controlling ME.

[1] [Intel ICH9 EDS](http://read.pudn.com/downloads212/ebook/999452/%E6%96%B0%E5%BB%BA%E6%96%87%E4%BB%B6%E5%A4%B9/ICH9_EDS_Rev2_0.pdf)

[2] [Intel ICH9 Specification Update](https://www.intel.com/content/dam/www/public/us/en/documents/specification-updates/io-controller-hub-9-family-specification-update.pdf)

skochinsky, 2018-02-14:
I did not know about `MCHSTRP0`, thanks for the links!

"ME FW workaround" sounds like a synonym of "Silicon Workaround Capability" (SWC) mentioned in some other ME materials. AFAIK the internal name is "PMC patch" and is performed by the BUP module (grep for `$PMC_PAT`).

c0d3z3r0, 2018-02-14:
> "ME FW workaround" sounds like a synonym of "Silicon Workaround Capability" (SWC) mentioned in some other ME materials. AFAIK the internal name is "PMC patch" and is performed by the BUP module (grep for $PMC_PAT).
Yes, I think so, too, but the interesting part would be what ME functions can be controlled there.

(I wouldn't be surprised if we find something similiar for BootGuard... haha)

c0d3z3r0, 2018-02-15:
Hmm, Intel FIT only sets PCHSTRP0[0] when selecting "Yes" for the "Reserved" (ME Disable) value...

c0d3z3r0, 2018-02-15:
Some more stuff in appendix A in this document:
http://www.corus.pro/pilotes/CorusX/X37/XP/ME/SPI%20Programming%20Guide.pdf

skochinsky, 2018-02-15:
I think `PCHSTRP0` is the combination of `ICHSTRP0` and `MCHSTRP0` (PCH used to be two separate chips).

c0d3z3r0, 2018-02-15:
No, I'm not sure about this. I think PCHSTRP0 is the new ICHSTRP0. Intel Series 7 called it PCHSTRP already, but Series 9 and 10 call it again ICHSTRP... However, in FIT there are for example the two settings "PCH   `Platform Protection->Intel PTT Supported` and `ME Kernel->Reserved`. When diffing the ifdtool results the first setting modifies `MCHSTRP0` and the second `PCHSTRP0`. I don't know if ifdtool works correctly.

Additionally, Intel 100 Series and C230 Series Datasheet Vol. 1 and 2 [1,2] mention all three, MCH, PCH and ICH.

c0d3z3r0, 2018-02-15:
After some more reading in Intel datasheets etc. I think what I said is correct.
With Chipset series 5 Lynnfield (P55 was the first one) Intel fully integrated MCH (northbridge) into the CPU package. That's maybe why MCH Straps are called CPU Straps. ICH became PCH then. so `ICHSTRP` should be the same as `PCHSTRP` but MCHSTRP didn't change.
The newer revision, Series 9, for which we have an EDS Datasheet (see first post) has `PCHSTRP` and `MCHSTRP`. Since, for my case, Series 100/C230 (specifically C236 for me), is Series 9's direct successor, there should not be much difference but at least FIT doesn't touch `MCHSTRP` for disabling ME as mentioned above.

skochinsky, 2018-02-15:
ICH9 is **not** from the Intel 9 Series but much older Bearlake chipset from 2007 (yes, it's confusing). It used ME 3.0.

https://en.wikipedia.org/wiki/Intel_P35
https://en.wikipedia.org/wiki/I/O_Controller_Hub#ICH9

c0d3z3r0, 2018-02-15:
Ouch. Indeed, you're right ... very confusing.

c0d3z3r0, 2018-02-16:
MEInfo does not report any difference between setting `PCHSTRP0[0]` vs. `PCHSTRP0[0]` and `PCHSTRP0[0]`. So, at least for newer chipsets MCH can be ignored. Closing this.

closed 2018-02-16 by c0d3z3r0