corna/me_cleaner #170: Help with output on Haswell 4670K

baxax, 2018-02-18:
I have GIGABYTE GA-H87M-D3H with CPU 4670K. I applied me_cleaner by Internal flashing with OEM firmware with -S and got this output:
MEI found: [8086:8c3a] 8 Series/C220 Series Chipset Family MEI Controller #1

ME Status   : 0x1e020191
ME Status 2 : 0x124d2142

ME: FW Partition Table      : OK
ME: Bringup Loader Failure  : NO
ME: Firmware Init Complete  : NO
ME: Manufacturing Mode      : YES
ME: Boot Options Present    : NO
ME: Update In Progress      : NO
ME: Current Working State   : Initializing
ME: Current Operation State : Bring up
ME: Current Operation Mode  : Debug
ME: Error Code              : No Error
ME: Progress Phase          : BUP Phase
ME: Power Management Event  : Clean global reset
ME: Progress Phase State    : 0x4d

ME: Extend SHA-256: 8cc01d4211e46fb486270080e6bbc4a11446e0dcd196d9784b884aaa983319cb

ME: failed to become ready
ME: failed to become ready
ME: GET FW VERSION message failed
ME: failed to become ready
ME: failed to become ready
ME: GET FWCAPS message failed

When I apply with argument -s everything is the same except this line:
ME: Power Management Event     : Pseudo Global Reset
Can someone "translate" this ouptut to me, is it ok?

baxax, 2018-04-01:
nobody?!

corna, 2018-04-01:
Sede [here](https://github.com/corna/me_cleaner/wiki/Get-the-status-of-Intel-ME). Seems OK to me