corna/me_cleaner #171: Check Intel SPS Status

Yannik, 2018-02-25:
I'm trying to use me_cleaner on an Asus P10S-I with XEON E3-1240v6 (Kaby-Lake). I successfully extracted the firmware using spi. The output when applying me_cleaner also looks good: 
```
$ ./me_cleaner.py --soft-disable original-asus-p10s.rom --output soft-disable-asus-p10s.rom
Full image detected
The ME/TXE region goes from 0x1000 to 0x800000
Found FPT header at 0x1010
Found 15 partition(s)
Found FTPR header: FTPR partition spans from 0xa000 to 0x6a000
Found FTPR manifest at 0xa0e8
ME/TXE firmware version 4.1.4.54
Public key match: Intel SPS, firmware versions 4.x.x.x
Reading partitions list...
 FTPR (0x0000a000 - 0x00006a000, 0x00060000 total bytes): NOT removed
 FTUP (0x000f9000 - 0x0002f9000, 0x00200000 total bytes): removed
 DLMP (      no data here      , 0x00000000 total bytes): nothing to remove
 PSVN (      no data here      , 0x00000000 total bytes): nothing to remove
 IVBP (      no data here      , 0x00000000 total bytes): nothing to remove
 MFS  (0x0008a000 - 0x0000ee000, 0x00064000 total bytes): removed
 NFTP (      no data here      , 0x00000000 total bytes): nothing to remove
 ROMB (      no data here      , 0x00000000 total bytes): nothing to remove
 FPTB (0x00001000 - 0x00000a000, 0x00009000 total bytes): removed
 MFSB (0x0006a000 - 0x00008a000, 0x00020000 total bytes): removed
 IVB1 (0x000ee000 - 0x0000f2000, 0x00004000 total bytes): removed
 IVB2 (0x000f2000 - 0x0000f6000, 0x00004000 total bytes): removed
 FLOG (0x000f6000 - 0x0000f7000, 0x00001000 total bytes): removed
 UTOK (0x000f7000 - 0x0000f9000, 0x00002000 total bytes): removed
 OPR1 (0x000f9000 - 0x0002f9000, 0x00200000 total bytes): removed
Removing partition entries in FPT...
Removing EFFS presence flag...
Correcting checksum (0x51)...
Reading FTPR modules list...
 FTPR.man     (uncompressed, 0x00a0e8 - 0x00a58c): NOT removed, partition manif.
 rbe.met      (uncompressed, 0x00a58c - 0x00a622): NOT removed, module metadata
 kernel.met   (uncompressed, 0x00a622 - 0x00a6b0): NOT removed, module metadata
 syslib.met   (uncompressed, 0x00a6b0 - 0x00a714): NOT removed, module metadata
 bup_rcv.met  (uncompressed, 0x00a714 - 0x00abc0): NOT removed, module metadata
 rbe          (LZMA/uncomp., 0x00abc0 - 0x012bc0): NOT removed, essential
 kernel       (Huffman     , 0x012bc0 - 0x022940): NOT removed, essential
 syslib       (Huffman     , 0x022940 - 0x0382c0): NOT removed, essential
 bup_rcv      (Huffman     , 0x0382c0 - 0x06a000): NOT removed, essential
The ME minimum size should be 454656 bytes (0x6f000 bytes)
The ME region can be reduced up to:
 00001000:0006ffff me
Setting the HAP bit in PCHSTRP0 to disable Intel ME...
Checking the FTPR RSA signature... VALID
Done! Good luck!
```

However, `intelmetool` always exits with the following error (even with original ROM):
```
Could not get RCBA address
Error reading RCBA
```

Also, dmesg reports `[    1.589870] mei_me 0000:00:16.0: Device doesn't have valid ME Interface`

It _seems_ like disabling SPS (Intel ME for Server Platforms) worked, but is there any way to check in linux?

Yannik, 2018-02-25:
Using the [Intel-SA-00086 Detection Tool](https://www.intel.com/content/www/us/en/support/articles/000025619/software.html) (download [here](https://downloadcenter.intel.com/download/27150?v=t)), I was able to get the following information:
Original ROM:
```
$ SA00086_Linux/common/spsInfoLinux64

Intel(R) spsInfo Version: 4.2.74.9
Copyright(C) 2005 - 2017, Intel Corporation. All rights reserved.

FW Status Register 1: 0x000F0345
  CurrentState (3:0):                     Normal (5)
  ManufacturingMode (4):                  Disabled (0)
  FlashPartition (5):                     Valid (0)
  OperationalState (8:6):                 M0 with no UMA (5)
  InitComplete (9):                       Complete (1)
  BUPLoadState (10):                      Success (0)
  FwUpdateInProgress (11):                No (0)
  ErrorCode (15:12):                      No Error (0)
  ModeOfOperation (19:16):                Server Platform Services (15)
  MeResetCount (23:20):                   0
  FlashDescriptorVerificationStatus (24): Verification failed (0)
  OEMDefinedCPUDebugPolicyStatus (25):    CPU Debug capability enabled (0)
  FIASKULimitViolationStatus (26):        SKU Limit not Violated (0)

FW Status Register 2: 0x89116806
  ICC programmed successfully (1):        Yes (1)
  ICC: valid data read from SPI (2):      Yes (1)
  Restricted Mode (3):                    Disabled (0)
  Chipset Hard Fused (5):                 False (0)
  MfsFailure (6):                         No Mfs failure (0)
  WarmReset (7):                          No warm reset request (0)
  EndOfPOST (11):                         Received (1)
  TargetImageBoot (12):                   Success (0)
  Heartbeat (15:13):                      3
  ExtendedStatusData (23:16):             11h
  PM Event (27:24):                       Non-power cycle reset (9h)
  Phase (31:28):                          MAESTRO (8)

Server Platform Service firmware is detected on the system.
SPS Image FW version: 4.1.4.54 (Recovery), 4.1.4.54 (Operational)
Feature list:
  HECI Interface Version:                   1.2
  Node Manager:                             disabled (0)
  PECI Proxy:                               enabled (1)
  Reset Suppression (Pre-Go-S1):            disabled (0)
  PMBus Proxy over HECI:                    enabled (1)
  MCTP Proxy:                               disabled (0)
  Power Thermal Utility Support:            disabled (0)
  PCH Thermal Sensor Init:                  FW supported (1)
  MCTP Infrastructure:                      disabled (0)
  Turbo State Limiting:                     FW supported (1)
```

After me_cleaner with soft-disable option:
```
$ SA00086_Linux/common/spsInfoLinux64

Intel(R) spsInfo Version: 4.2.74.9 
Copyright(C) 2005 - 2017, Intel Corporation. All rights reserved.

FW Status Register 1: 0x000F0382
  CurrentState (3:0):                     Recovery (2)
  ManufacturingMode (4):                  Disabled (0)
  FlashPartition (5):                     Valid (0)
  OperationalState (8:6):                 Bring Up (6)
  InitComplete (9):                       Complete (1)
  BUPLoadState (10):                      Success (0)
  FwUpdateInProgress (11):                No (0)
  ErrorCode (15:12):                      No Error (0)
  ModeOfOperation (19:16):                Server Platform Services (15)
  MeResetCount (23:20):                   0
  FlashDescriptorVerificationStatus (24): Verification failed (0)
  OEMDefinedCPUDebugPolicyStatus (25):    CPU Debug capability enabled (0)
  FIASKULimitViolationStatus (26):        SKU Limit not Violated (0)

FW Status Register 2: 0x364E1404
  ICC programmed successfully (1):        No (0)
  ICC: valid data read from SPI (2):      Yes (1)
  Restricted Mode (3):                    Disabled (0)
  Chipset Hard Fused (5):                 False (0)
  MfsFailure (6):                         No Mfs failure (0)
  WarmReset (7):                          No warm reset request (0)
  RecoveryReason (10:8):                  FW internal error (4)
  EndOfPOST (11):                         Not Received (0)
  TargetImageBoot (12):                   Failure (1)
  Heartbeat (15:13):                      0
  ExtendedStatusData (23:16):             4Eh
  PM Event (27:24):                       Pseudo-global reset (6h)
  Phase (31:28):                          BUP (3)

Server Platform Service firmware is detected on the system.
ERROR while version reading...
```

Seems like a full success to me!

@corna Maybe this should be included in the wiki and SPS 4.x marked as officially supported?

corna, 2018-02-25:
I'm working on SPS in these weeks, I think there may be some uncommon cases (SPS 1.x-3.x) which might need some modifications in me_cleaner.

It probably worked, but can you post here the output of MEInfo (there's a UEFI version if you don't want to use Windows, non-free software unfortunately...) to make sure that everything worked? How did you flash back the modified image?

Yannik, 2018-02-25:
@corna I flashed the ROM using SPI from Raspberry Pi 3. This Mainboard uses a DIP-8 so it was very easy to do.

Can you point me to a (trusted) binary of the MEInfo EFI? I don't have a ms windows license so I won't be able to check on windows. I'm currently also exploring the different results of soft-disable/soft-disable-only/hard-disable-only and will post updates here when I'm done.

corna, 2018-02-25:
--soft-disable should be the better choice.

MEInfo is version-specific, I don't know exactly which version should be used with SPS. @platomav?

skochinsky, 2018-02-25:
for SPS firmwares you need to use `spsInfo`, also version-specific (1.x, 2.x, 3.x, 4.x).

Yannik, 2018-02-25:
@skochinsky Can you point me to a (trusted) download of spsInfo? Are we still talking about a UEFI binary here?

skochinsky, 2018-02-25:
In your previous posts you pasted output of `spsInfoLinux64` so it seems you already have it, or am I missing something?

Yannik, 2018-02-25:
@skochinsky After I posted the output from `spsInfoLinux64`, @corna asked me to post additional info from some `MEInfo` UEFI module, so that's what I was looking for in this case.

corna, 2018-02-25:
My bad, I didn't realize that Intel released a Linux version of spsinfo and that you posted the output of it ;)

Yes, it should be disabled (`  OperationalState (8:6):                 Bring Up (6)`)

corna, 2018-02-25:
Sorry for the mess

Yannik, 2018-02-25:
@corna No problem. I linked the intel-provided download link to the linux version of spsInfo I used in my second post.

corna, 2018-02-25:
Now I see it ;)
Thanks for the info, please report the success also in #3

closed 2018-02-25 by corna