corna/me_cleaner #172: Disabling Intel ME remote management Via Metasploit

ravenise, 2018-02-26:
Once a user has access to the intel ME internals, they have access to modify intel ME's settings; then theoretically disabling these back door features should be as easy as using Metasploit "auxiliary/scanner/http/intel_amt_digest_bypass" and then disabling remote features.  

https://rapid7.com/db/modules/auxiliary/scanner/http/intel_amt_digest_bypass

matt123b, 2018-02-26:
One of the issues with the IME is that it has its own little operating system that's running in there whenever the PC has power supplied to it. It's an always-on hardware/firmware hypervisor, and Windows and Linux have drivers that can be used to interface with the IME from the OS level. With me_cleaner all of the required firmware for the IME to work is stripped out, except for the FPT and BUP, so a partition map and one module for hardware init and some power management. The IME also has a kernel module and others that are responsible for processing updates and managing other modules. These other modules, like AMT, aren't responsible for processing updates to the IME firmware. 

If there was a dangerous program running on a Linux machine for example, I could uninstall it. But there's still an entire OS under that, and if I don't trust that OS, how do I know that this can't be used to reinstall the bad program? This basically means that I can never connect the machine to the outside web, which totally defeats the purpose of removing the malicious program anyways. The IME is similar here because you're just shutting down/removing some of the lesser software while the core OS components remain. 

Don't get me wrong here, this would be better than nothing at all, but still not as through as me_cleaner. Using an external flasher to gut the firmware will always be a better option because there's no chance of the IME reinstalling its own modules without a kernel and other essential parts used to install them. Really for all intents and purposes, me_cleaner is disabling the IME.

ravenise, 2018-02-26:
Thank you for your prompt response my brother. That's great and I have no problem with that, I agree that me_cleaner is the best option.  Thing is me_cleaner exists, the suggestion I am offering doesn't so far as I am aware. If AMT and all the remote features were disabled VIA an exploit then ostensibly there would be no way for an attacker to re-activate those remote mechanisms unless the firmware was re-flashed again. However, it would appear firmware updates can be disabled as well, as evident by the settings in my descriptor. 

"local firmware update enabled "false"
"bios reflash capable" false
"ME flash protection override enabled" false

(I have two different setting layouts in my descriptor, "Boulder Creek -EL_ICH10" and "McCreary -EL_ICH10D." I'm not sure which is active or if both are active at the same time.)

So using the method I suggest, the only possible mechanism for exploitation would be if an exploit was previously embedded inside a compromised Intel ME (or firmware hub), that is capable of changing the settings back, and or establishing a connection through the O/S somehow.  If the computer board is fresh, never connected to the internet, and you did this immediately, bingo, no chance for an exploit living inside there. 

The biggest issue with me_cleaner is it requires people flash their firmware with external devices with the risk of bricking their hardware. I so far I haven't been able to get a handle on what version of Intel ME is on my 10 year old board.  Possibly because I re-flashed it with an Intel firmware update some years ago;  Nothing from Intel would detect it.  MeInfoWin.exe says "PMXUtil: Error during PMX Call: sseiddrvedll32e.dll!IDRVInstallDriver[}: Couldn't Open Service Manager err(5). In dos I managed to dump the descriptor, which doesn't seem to provide a version number, looks like it may have been stripped. "Flash Image" -> "ME Region" says "major version 0", "minor version 0," "hotfix version 0," "build version 0." fpt.exe stated, "reading region information from flash descriptor" --- Flash Image Information, Signature: INVALID! No more information to be displayed." But at least I can see the configuration of ME.  

How many people can or are willing to flash their boards? I would only do this if I knew it was 100%  safe. But that is a lot of messing around. Using an actual open source exploit from a trustworthy source to disable the most dangerous components of ME, its authentication mechanisms and remote access and reflashability, I believe is one of the best options.  The descriptor can be dumped later to detect any modifications and changes.

Take a look at my descriptor, tell me what you think is going on in here. The only tool that worked for me was Intel ME 5.0; 
[MX25L80005.zip](https://github.com/corna/me_cleaner/files/1760014/MX25L80005.zip)

skochinsky, 2018-02-27:
@ravenise This dump contains only the legacy BIOS (AMIBIOS8) and has no descriptor. If this is full content of your flash, it seems your board is using flash in descriptor-less mode  (it was possible until  ICH10), in which the ME firmware is not present in the flash so the ME is only running the Boot ROM code. So you don't need to clean it because it's already absent :)

ravenise, 2018-02-28:
@skochinsky I am using an Asus P5Q Pro; It has a ICH10R chipset; The board also has a thing called "Express Gate" which is a linux kernel / distro that allows one to boot onto the net right at post.

ravenise, 2018-02-28:
@skochinsky So this is the same BIOS I am accessing during post? Are yous saying me Intel ME on my ICH10R poses no risk at all?

ravenise, 2018-02-28:
I'm curious how my hackers managed to disable my computer from connecting from linux, including live distros; for a while anyway.

ravenise, 2018-02-28:
So a very strange thing occurred the other day while I was logged into windows. After disabling a few entries in the registry governing hardware "redirect" (aka remote control and access to the hardware of my PC) Registry entries grew from ~461000, to ~540000. A massive hidden layer of my registry was uncovered that I had never seen or knew about before; I should have made a backup of this, because I can no longer access it anymore; 1/5'th of my registry vanished again. All the entries I had previously searched for and changed, are no longer visible in my registry. For example.

[HKEY_LOCAL_MACHINE\SOFTWARE\Wow6432Node\Policies\Microsoft\Windows\DeviceRedirect\Restrictions\]
"AllowRedirect"=dword:00000000

[HKEY_LOCAL_MACHINE\SOFTWARE\Policies\Microsoft\Windows\DeviceRedirect\Restrictions]
"AllowRedirect"=dword:00000000

I had changed these from 0, to 1. After which, some of them were persistent on reboot; they would switch back to 1 again. It wasn't much time before I was no longer able to find them anymore; I had not deleted them.

It is as though some remote admin was redirecting a huge chunk of my registry / O/S and hardware. I was shocked to see these hidden registry keys unlocked; In the process of disabling the redirect settings, while searching for "DeviceRedirect" double or more more redirect keys became available to me on the next boot. (Along with 1/5 of my registry which I had never seen before)You can find these keys by searching for "DeviceRedirect" and or "allowredirect".

These six objects also disappeared.

[HKEY_USERS\S-1-5-21-2240875929-447784991-1091042645-1000\Software\Microsoft\Windows\CurrentVersion\Group Policy Objects\{007A0536-350C-47ED-9868-DF5C42F80CEA}Machine\Software\Policies\Microsoft\Windows\DeviceRedirect\Restrictions]
"AllowRedirect"=dword:00000000

[HKEY_USERS\S-1-5-21-2240875929-447784991-1091042645-1000\Software\Microsoft\Windows\CurrentVersion\Group Policy Objects\{2228C782-A61F-4964-BF35-039C64C5762A}Machine\Software\Policies\Microsoft\Windows\DeviceRedirect\Restrictions]
"AllowRedirect"=dword:00000000

[HKEY_USERS\S-1-5-21-2240875929-447784991-1091042645-1000\Software\Microsoft\Windows\CurrentVersion\Group Policy Objects\{971B4056-B53A-4932-85C0-1C67D70BFD18}Machine\Software\Policies\Microsoft\Windows\DeviceRedirect\Restrictions]
"AllowRedirect"=dword:00000000

[HKEY_USERS\S-1-5-21-2240875929-447784991-1091042645-1000\Software\Microsoft\Windows\CurrentVersion\Group Policy Objects\{AF3CE3F0-520B-435A-A45C-312C2374AB79}Machine\Software\Policies\Microsoft\Windows\DeviceRedirect\Restrictions]
"AllowRedirect"=dword:00000000

[HKEY_USERS\S-1-5-21-2240875929-447784991-1091042645-1000\Software\Microsoft\Windows\CurrentVersion\Group Policy Objects\{DA63EBEF-85CB-4B8A-AC11-43F92721699F}Machine\Software\Policies\Microsoft\Windows\DeviceRedirect\Restrictions]
"AllowRedirect"=dword:00000000

[HKEY_USERS\S-1-5-21-2240875929-447784991-1091042645-1000\Software\Microsoft\Windows\CurrentVersion\Group Policy Objects\{E615A4B9-1522-4ED3-82A0-03A4B4BD5848}Machine\Software\Policies\Microsoft\Windows\DeviceRedirect\Restrictions]

I uncovered all of this while finding possible shady group policy entries for Chrome browser.
One of the steps I took that may have uncovered my hidden registry entries was followed here: 
https://www.bleepingcomputer.com/vir...-administrator

I ran RD /S /Q "%WinDir%\System32\GroupPolicyUsers"
RD /S /Q "%WinDir%\System32\GroupPolicy"
gpupdate /force 

A chrome extension (GhostVPN) had access to system settings and had injected a group policy to allow access to system Proxy settings;

Registry Group Policy Objects \SOFTWARE\Microsoft\Windows\CurrentVersion\Group Policy Objects\{FC974C88-F4E9-4026-B8E4-839875341946}Machine\Software\Policies\Google\Chrome\ExtensionInstallWhitelist\1 = nhippelchacimnkamngddemhkifekini

ravenise, 2018-02-28:
Will anything have changed if I try dumping after having removed the ram from slot 0?

ravenise, 2018-02-28:
@skochinsky   Will anything have changed if I try dumping after having removed the ram from slot 0?

ravenise, 2018-02-28:
I'm curious how my hackers managed to disable my computer from connecting to the internet from linux, including live distros; for a while anyway. Though they were also changing the mac address on the router and the passwords at one point as well.

skochinsky, 2018-02-28:
Since your flash is missing the ME firmware, presence or absence of RAM in slot 0 has no effect on ME: no code will run except the most basic hardware init in the boot ROM.

Your PC issues have nothing to do with ME, please use superuser.com or similar to fix them.

ravenise, 2018-03-01:
[image] Ok so as it is depicted in this image, it says iAMT, and ME supported. That is all.

ravenise, 2018-03-01:
I tried importing bios the me software dump, and the original bios i had recently flashed; it was after market; [kets bios ](http://www.overclock.net/forum/167-intel-bios/818744-kets-p5q-modded-bios-files.html) I had added some xeon micro codes; So you are basically saying this board shipped with intel ME already disabled? I thought a hard flash was mandatory.

ravenise, 2018-03-01:
I am reading via libreboot quote 

> "Before version 6.0 (that is, on systems from 2008/2009 and earlier), the ME can be disabled by setting a couple of values in the SPI flash memory. The ME firmware can then be removed entirely from the flash memory space."

So it appears Ket may have actually done this himself with his bios?

skochinsky, 2018-03-01:
On ICH10 (NB: *not* the Intel 10 series!), the ME firmware is optional, so it's up to OEM whether to provide it or not; it seems on your board the OEM is not using the ME from the beginning (probably to save on flash chip cost).

skochinsky, 2018-03-01:
P.S. from the [ICH10 datasheet](https://www.intel.com/content/dam/doc/datasheet/io-controller-hub-10-family-datasheet.pdf):

> 5.23.1.1 Non-Descriptor Mode
> Non-descriptor mode is similar to the flash functionality of ICH7. In this mode, SPI
> Flash can only be used for BIOS. Direct read and writes are not supported. BIOS has
> read/write access only through register accesses. Through those register accesses
> BIOS can read and write to the entire flash without security checking. There is also no
> support for the integrated Gigabit Ethernet, Intel Management Engine, chipset soft
> straps, as well multiple SPI Flash components.

The value at the start of flash (offset 0) determines mode:

> FLVALSIG - Flash Valid Signature Register
> Memory Address: FDBAR + 000h
> Flash Valid Signature. This field identifies the Flash Descriptor sector as valid. If the
> contents at this location contain 0FF0A55Ah, then the Flash Descriptor is considered
> valid and it will operate in Descriptor Mode, **else it will operate in Non-Descriptor Mode.**
>

ravenise, 2018-03-02:
What I lucky guy I turned out to be. I'm using an Asus P5Q Pro; with a little mod [you can use a quad cord Xeon](http://www.overclock.net/forum/5-intel-cpus/1431723-mod-lga775-support-lga771-xeon-cpus.html); They are faster than the core2 quad core extreme editions; some of which are [still selling for $270 on ebay](https://www.ebay.com/itm/Intel-Core-2-Extreme-QX9770-3-2-GHz-SLAWM-LGA-775-Prozessor-CPU-Quad-Core/302448306726?epid=2233144679&hash=item466b52ea26:g:hukAAOSw~fpZtg36);  I bought my Xeon 5640 for $18 usd, and mod was only $3.  Highly recommend you cover your mofsets with a [proper heatsink](https://www.frozencpu.com/products/7598/vid-111/Enzotech_MST-81_Forged_Copper_Mosfet_Heatsink_-_ASUS_Gigabyte_-_87mm_x_16mm_x_296mm.html), here is a [good source](https://www.ebay.com/sch/i.html?_odkw=quadcore+extreme&_osacat=0&_from=R40&_trksid=m570.l1313&_nkw=Enzotech+MST-81&_sacat=0).

ravenise, 2018-03-02:
I am curious as to why my Qualcomm/Atheros AR8121/AR8113/AR8114 PCI-E Ethernet Controller (L1e) is enabled during post, on occasion; typically after a BSOD; or after running certain linux distros. When the computer reboots, the lancard remains active; the light on my switch turns on; The only way to stop this is to turn off the computer, unplug it from the wall and remove the battery.  otherwise it eventually stops after a few reboots. The lan card then only loads when driver kernels are loaded up by the O/S;

ValoWaking, 2023-02-26:
ICH7 is not safety to use and u asking abot ICH10, amazing, ARM ARC in your nothbridge have hisone flash, that containe OS, ICH7 is AMT ready chipset, and it can passtrow all from your ARM in nothbridge to your virtual MAC (that have ASF, or DASH).

u can open any datasheet like 945GC+ICH7 or G41+ICH7 and read scheme 

At first u need to ask Intel, NSA or maybe FBR how work all this sh*t

and at first i think u need to read what is PHY NIC, at second - put out your backdoored P5Q in window ))

CodeAsm, 2023-02-27:
> I am curious as to why my Qualcomm/Atheros AR8121/AR8113/AR8114 PCI-E Ethernet Controller (L1e) is enabled during post, on occasion; typically after a BSOD; or after running certain linux distros. When the computer reboots, the lancard remains active; the light on my switch turns on; The only way to stop this is to turn off the computer, unplug it from the wall and remove the battery. otherwise it eventually stops after a few reboots. The lan card then otherwise only does this when kernels drivers are loaded up by the O/S; I thought this may have been a feature of Intel ME. Do you know anything about this? There is a feature on this Atheros developed by Asus called "AI net 2 " which if enabled also boots the Lan card up on post. Enables or disables checking of the Atheros LAN cable during the Power-On Self-Test.
> 
> [+E4427_P5Q_Pro.pdf](https://github.com/corna/me_cleaner/files/1773541/E4427_P5Q_Pro.pdf)

Dont know if your particular cards are the case, but most Wireless and wired network cards get their firmware from the driver, so when you run windows, or linux, from coldboot, either no firmware may be running or a very basic one (the one that allows the bios to netboot maybe or load another firmware from EFI modules?)

anyway, even if I get the first part wrong, the drivers may load a newer firmware into the card and when your windows BSOD, or sleeps, linux sleeps or reboots, the firmware may still be running, the OS (or malicuous actor) may have setup the card to wake up from lan, or send periodic packets? im not familiar with the exact default options, but the card may basicly still be active.

If your not a politician or famous person tho, dont asume youd be a direct target and it may just be a firmware doing its "basic" things, like reading packets and seeing if its for him (to wake up). try wireshark and analyse any packets going in and out, match the mac adress. (and yeah, some networkcards may even have another network device, aka MAC on the same port.) 
If Intel ME isnt running, as previous replies seem to indicate, Intel ME isnt your problem, the networkcard running some firmware while the CPU is halted or BSOD, should not be a problem. you could analyse your network port for what is actually being send, but i asume nothing special.

frantic-from-paranoiawf, 2024-12-03:
> ICH7 is not safety to use and u asking abot ICH10, amazing, ARM ARC in your nothbridge have hisone flash, that containe OS, ICH7 is AMT ready chipset, and it can passtrow all from your ARM in nothbridge to your virtual MAC (that have ASF, or DASH).
> 
> u can open any datasheet like 945GC+ICH7 or G41+ICH7 and read scheme
> 
> At first u need to ask Intel, NSA or maybe FBI how work all this sh*t
> 
> and at first i think u need to read what is PHY NIC, at second - put out your backdoored P5Q in window ))

Why are you defending such an abhorrent thing? The ME gives ZERO security, it only opens up a gaping hole in your computer.
If you actually trust security by obscurity, you need to grow the fuck up.