corna/me_cleaner #173: Intel ME Founder: Disable Intel ME by Using Ram Slots not colored "blue"

ravenise, 2018-02-28:
According to the founder of intel me, if you use the ram slots coloured "blue" Intel ME has access to the ram. By using the ram slots coloured "black" Intel ME will not function. https://www.youtube.com/watch?v=dqOlgDhqrh0

corna, 2018-02-28:
This was true for older architectures, unfortunately  in the "Intel 7 Series - SPI programming guide", "Intel ME disable" section, it is stated that:

> Removing the DIMM from channel 0 no longer has any effect on Intel Mangement 
Engine functionality.

ravenise, 2018-02-28:
Thankfully I'm using pre-Intel 7 series!

corna, 2018-02-28:
Then you don't even need me_cleaner: in the "pre-Nehalem" architectures it was possible to **fully** disable Intel ME by providing an image without the ME region (never tested, I don't own such platform).

See the ich9deblob tool from Libreboot for more info

persmule, 2018-02-28:
@corna 
ich9deblob's result has partially been absorbed into coreboot via [blobtool](https://www.coreboot.org/Board:lenovo/x200#Without_ME_firmware_updates.2FAMT). To fully disable ME some straps should be modified.