corna/me_cleaner #174: Using an after market NIC bypasses Intel ME OOB

ravenise, 2018-02-28:
Can Intel ME bridge after market network cards? And just how much safer are you using an after market NIC on an ME enabled device?

**According to Intel staff:  Are separate Intel gigabit NIC cards a solution to AMT vulnerability?** https://communities.intel.com/thread/114211

> Intel AMT  requires build in Intel AMT enabled LAN PHY (SKUs with -LM at the end of their description) (and/or AMT enabled WiFi Controller HW) as it provides HW means for OOB TCP/IP stack. If you add any additional LAN HW (does not matter which vendor or what bus) it will not support Intel AMT OOB.
> Please note that depending on configuration (Host VPN support and Home Domains) Intel AMT when configured may receive messages over other than AMT interfaces when OS is running. So local vulnerability shall be disabled by blocking LMS services - see Mitigation Guide published at Download INTEL-SA-00075 Mitigation Guide https://communities.intel.com/external-link.jspa?url=https%3A%2F%2Fdownloadcenter.intel.com%2Fdownload%2F26754

ravenise, 2018-02-28:
I have ordered an aftermarket card which I bought for its OPT (one time flash memory) qualities, there is no on board flash ROM. I don't want to bypass intel ME with an after market NIC that has a re-programmable / flashable ROM; I later learned that the [UGreen RLT81111G](https://www.ugreen.com/product/UGREEN_Gigabit_Ethernet_PCI_Express_Network_Controller_Card_RJ45_Lan_Adapter_Converter-en.html#ad-image-0) implements ECMA-393, Intel's ProxZzzy; This standard has ME like qualities. It allows the ethernet card to remain connected on a network and send and receive packets while the computer is in "sleep" mode. (Makes me wonder how the device actually runs, does it need an O/S to mount the driver... is this powered by minix?) It has an inbuilt packet sniffer that is triggered by specific bits to perform specific functions; It can wake the computer up from sleep. ECMA admits this is totally insecure by design, can be hijacked and used to generate rogue packets. I have gone into detail into this here:  [Intel ProxZzzy the next Intel ME?](https://www.linuxquestions.org/questions/linux-security-4/intel-proxzzzy-the-next-intel-me-4175624518/ )

skochinsky, 2018-02-28:
As you quoted, normally ME can only use onboard Intel LAN chip which must be directly connected to the chipset (PCH), so any external cards are not supported. In *some* mobile configurations it may be able to use the resident OS drivers to use the WiFi chip (AFAIK it's supported on Windows only). Note that all this applies only to configurations with AMT functionality (5MB ME firmware), if you have the consumer firmware (1.5MB), or used `me_cleaner`, it has no network functionality.