corna/me_cleaner #179: 'Could not get RCBA address'

solinium, 2018-03-13:
I'm using a Dell XPS 15 9550 and intelmetool seems to be giving me
`Could not get RCBA address`
`Error reading RCBA`
no matter what options I pass. It seems that it's working for other XPS 15 owners so I'm not sure why it would be different in my case? I'm using parrotsec (basically debian with security tweaks) so it's also possible some security-related thing is interfering.
system info: https://pastebin.com/raw/F3S6u9vK

zifxify, 2018-03-18:
add the following kernel parameter when booting: iomem=relaxed

solinium, 2018-03-18:
@zifxify No good, same result

skochinsky, 2018-03-20:
please try flashrom support for flashrom issues.

solinium, 2018-03-25:
@skochinsky thanks for the help
I've since tried it on stock debian testing and I've tested flashrom.
I get `WARNING! You seem to be running flashrom on an unsupported laptop.
Laptops, notebooks and netbooks are difficult to support and we
recommend to use the vendor flashing utility. The embedded controller
(EC) in these machines often interacts badly with flashing.` and `Error: Programmer initialization failed.`
my laptop may be officially unsupported however there are many people in the XPS 15 9550 issue that don't have this issue, so I'm not sure what would be different

skochinsky, 2018-04-17:
The error text comes from the `flashrom` tool, not `me_cleaner`. So you should check flashrom support channels (website, manpage, mailing list, IRC channel etc.) for help with this error. This repository should be used only for issues with `me_cleaner` itself.

closed 2019-01-11 by solinium