corna/me_cleaner #180: Debian packaging: Some improvements

ge-fa, 2018-03-13:
Hi all,

- There is a [ITP in Debian](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=885152) to get this packaged.
- Therefore, I would like to ask for some improvements:
    - I would really like to see the releases signed, to ensure the code was not tampered with while importing into Debian. See [this](https://wiki.debian.org/Creating%20signed%20GitHub%20releases) for some details on this.
    - Could we add a changelog to get an overview what changed between one version to the next? This doesn't need to happen, necessarily, for the past, but it would be nice to have for the future.
    - Could a new version be released?

Thanks for your work on this -- highly appreciated!
In my opinion, this is an important tool.

Cheers,
Georg

corna, 2018-03-23:
Hi,

> * I would really like to see the releases signed, to ensure the code was not tampered with while importing into Debian. See this for some details on this.
> * Could we add a changelog to get an [...]

Sure.

> Could a new version be released?

I can release a new one just after closing #181 and rewriting the README.

Are you sure about the name `intel-me-cleaner`? This software is known as `me-cleaner`, adding an `intel-` in front of it can cause confusion in the users.

ge-fa, 2018-03-23:
Hi,

Great! 

Regarding the name: I'll take this to my fellow Debian people, thanks for the hint!

Cheers,
and thanks for your work,
Georg

ge-fa, 2018-03-25:
Also, a (short) man page would be nice, as required by Debian policy. There are tools available to help with this, for example `help2man`. (I could help with this as well.)

corna, 2018-03-25:
> Also, a (short) man page would be nice, as required by Debian policy. There are tools available to help with this, for example help2man. (I could help with this as well.)

I am not used to man pages, if you create a minimal one starting from the help I can fill it with the detailed information and examples.

sakaki-, 2018-03-25:
Writing raw `roff` manpages is a bit of a chore. There are a number of decent markdown to manpage convertors out there, for example [md2man](https://github.com/sunaku/md2man) or [ronn](https://github.com/rtomayko/ronn), which make the process much easier.

corna, 2018-03-25:
manpage added in the `dev` branch, now I just have to rewrite the README.

corna, 2018-03-25:
Btw, thanks @sakaki- for the tip.

corna, 2018-03-27:
@ge-fa README.md rewritten, tell me if you need something else before the 1.2 release.

corna, 2018-03-31:
Done.

closed 2018-03-31 by corna

ge-fa, 2018-10-09:
@corna I took over the ITP in Debian and will start with packaging within the next days, finally. I would like to start with the current state. Therefore: I could pull in the last patches (made since the last release) and ship these in Debian, but wanted to ask first, before doing this, if it would be possible to make another release? Thanks & cheers!

corna, 2018-10-09:
@ge-fa the commits after v1.2 have not been tested enough yet, I'm afraid I have to wait for more tests to release v1.3. The next version will probably be out within a month, it's up to you to decide which commit to pick for the Debian package

ge-fa, 2018-10-10:
@corna Thanks for the fast reply, highly appreciated! :) I'll just start with v1.2 then.