corna/me_cleaner #181: HP Elitebook 8440p OutOfRegionException problem

jjurkus, 2018-03-16:
I've built this external flasher, and am using it with a SOIC clip: https://learn.adafruit.com/programming-spi-flash-prom-with-an-ft232h-breakout/overview

I've dumped 8MB of the flash to a file and tried me_cleaner on it. It fails with an error:
`
# ./me_cleaner.py -S 8440p.bin
Full image detected
The ME/TXE region goes from 0x3000 to 0x500000
Found FPT header at 0x3010
Found 14 partition(s)
Found FTPR header: FTPR partition spans from 0xf2000 to 0x1b2000
ME/TXE firmware version 6.2.60.1066
Public key match: Intel ME, firmware versions 6.x.x.x
Reading partitions list...
 FOVD (0x00000400 - 0x000002000, 0x00001c00 total bytes): removed
 EFFS (0x00002000 - 0x0000e2000, 0x000e0000 total bytes): removed
 NVKR (NVRAM partition, no data, 0x00010000 total bytes): nothing to remove
 BIEL (      no data here      , 0x00003000 total bytes): nothing to remove
 BIIS (      no data here      , 0x00036000 total bytes): nothing to remove
 NVTD (NVRAM partition, no data, 0x00002000 total bytes): nothing to remove
 NVCM (NVRAM partition, no data, 0x00004000 total bytes): nothing to remove
 NVOS (NVRAM partition, no data, 0x00011000 total bytes): nothing to remove
 NVSH (NVRAM partition, no data, 0x00007000 total bytes): nothing to remove
 BIAL (      no data here      , 0x00009000 total bytes): nothing to remove
 NVVE (NVRAM partition, no data, 0x00000c00 total bytes): nothing to remove
 GLUT (0x000e2000 - 0x0000f2000, 0x00010000 total bytes): removed
 FTPR (0x000f2000 - 0x0001b2000, 0x000c0000 total bytes): NOT removed
 NFTP (0x001b2000 - 0x000500000, 0x0034e000 total bytes): Traceback (most recent call last):
  File "./me_cleaner.py", line 739, in <module>
    mef.fill_range(part_start, part_end, b"\xff")
  File "./me_cleaner.py", line 95, in fill_range
    raise OutOfRegionException()
__main__.OutOfRegionException
`

However, the AltMeDisable works fine. 
This laptop is a bit wonky, there is something broken in the bios battery circuit I think. Can the error above be explained by those NVRAM partitions with no data? After disconnecting the main battery one needs to set the bios settings again.

jjurkus, 2018-03-23:
Yes, this fixed it! Thanks!

(BTW, where do I report working motherboards or laptops? The HP elitebook 8560p works great with external flashing)

corna, 2018-03-23:
Great!
#3

closed 2018-03-23 by corna