corna/me_cleaner #182: Supermicro X11SSA-F with SPS not working

jjurkus, 2018-03-23:
I've tried me_cleaner with the X11SSA-F, but unfortunately it's not working. Probably because of the SPS firmware. 

After flashing the computer keeps rebooting. It starts the bios and executes a few things, but somewhere it reboots. With the -s flag it reboots even earlier, and one of the messages is 0x18=mrcColdBootRequired.

The -F means it has IPMI on board, and this might affect it too. I don't see it in the block diagram in the manual but there is something connected between IPMI and SPS, if I remember correctly.

Output looks good though:
```
# ./me_cleaner.py -S X11SSA-nieuw.bin
Full image detected
The ME/TXE region goes from 0x1000 to 0x500000
Found FPT header at 0x1010
Found 15 partition(s)
Found FTPR header: FTPR partition spans from 0xa000 to 0x6a000
Found FTPR manifest at 0xa0e8
ME/TXE firmware version 4.1.4.54
Public key match: Intel SPS, firmware versions 4.x.x.x
Reading partitions list...
 FTPR (0x0000a000 - 0x00006a000, 0x00060000 total bytes): NOT removed
 FTUP (0x000f9000 - 0x0002f9000, 0x00200000 total bytes): removed
 DLMP (      no data here      , 0x00000000 total bytes): nothing to remove
 PSVN (      no data here      , 0x00000000 total bytes): nothing to remove
 IVBP (      no data here      , 0x00000000 total bytes): nothing to remove
 MFS  (0x0008a000 - 0x0000ee000, 0x00064000 total bytes): removed
 NFTP (      no data here      , 0x00000000 total bytes): nothing to remove
 ROMB (      no data here      , 0x00000000 total bytes): nothing to remove
 FPTB (0x00001000 - 0x00000a000, 0x00009000 total bytes): removed
 MFSB (0x0006a000 - 0x00008a000, 0x00020000 total bytes): removed
 IVB1 (0x000ee000 - 0x0000f2000, 0x00004000 total bytes): removed
 IVB2 (0x000f2000 - 0x0000f6000, 0x00004000 total bytes): removed
 FLOG (0x000f6000 - 0x0000f7000, 0x00001000 total bytes): removed
 UTOK (0x000f7000 - 0x0000f9000, 0x00002000 total bytes): removed
 OPR1 (0x000f9000 - 0x0002f9000, 0x00200000 total bytes): removed
Removing partition entries in FPT...
Removing EFFS presence flag...
Correcting checksum (0x51)...
Reading FTPR modules list...
 FTPR.man     (uncompressed, 0x00a0e8 - 0x00a58c): NOT removed, partition manif.
 rbe.met      (uncompressed, 0x00a58c - 0x00a622): NOT removed, module metadata
 kernel.met   (uncompressed, 0x00a622 - 0x00a6b0): NOT removed, module metadata
 syslib.met   (uncompressed, 0x00a6b0 - 0x00a714): NOT removed, module metadata
 bup_rcv.met  (uncompressed, 0x00a714 - 0x00abc0): NOT removed, module metadata
 rbe          (LZMA/uncomp., 0x00abc0 - 0x012bc0): NOT removed, essential
 kernel       (Huffman     , 0x012bc0 - 0x022940): NOT removed, essential
 syslib       (Huffman     , 0x022940 - 0x0382c0): NOT removed, essential
 bup_rcv      (Huffman     , 0x0382c0 - 0x06a000): NOT removed, essential
The ME minimum size should be 454656 bytes (0x6f000 bytes)
The ME region can be reduced up to:
 00001000:0006ffff me
Setting the HAP bit in PCHSTRP0 to disable Intel ME...
Checking the FTPR RSA signature... VALID
Done! Good luck!
```

Also tested with the -s flag:
```
# ./me_cleaner.py -s X11SSA-F.bin
Full image detected
The ME/TXE region goes from 0x1000 to 0x500000
Found FPT header at 0x1010
Found 15 partition(s)
Found FTPR header: FTPR partition spans from 0xa000 to 0x6a000
Found FTPR manifest at 0xa0e8
ME/TXE firmware version 4.1.4.54
Public key match: Intel SPS, firmware versions 4.x.x.x
Setting the HAP bit in PCHSTRP0 to disable Intel ME...
Checking the FTPR RSA signature... VALID
Done! Good luck!
```

Output of ME Analyzer:
```
Firmware Family: CSE SPS
Firmware Version: 04.01.04.054
Firmware Release: Production
Firmware Type: Region
Firmware SKU: 2
Security Version Number: 3
Version Control Number: 0
Production Version: No
OEM/ODM Unlock Token: No
Firmware Date: 2017-09-25
Firmware Size: 0x2F9000
Flash Image Tool Version: 04.01.04.054
Firmware Chipset Support: Unknown
```

jjurkus, 2018-03-23:
I only have one RAM stick.
External flash, remove RAM stick, boot, and after boot fails due to missing RAM replace stick?

corna, 2018-03-24:
I wanted to force a cold boot by invalidating the MRC cache, but if you don't have another memory stick I don't know how. Maybe try changing the slot.

jjurkus, 2018-03-26:
Well, a new bios got released, so I tried that.
It doesn't give the mrcColdBootRequired with the -s firmware anymore, but now it just restarts after the ACPI initialization.

I think I'm going to give up on this one.