corna/me_cleaner #184: X230 bios - repeat application of me_cleaner claims that ME is still present?

rennis250, 2018-03-26:
I've successfully flashed the bios on my X230 with coreboot and it boots successfully, but it is unclear to me if ME was actually disabled. Dmesg, lspci -vnn, and intelmetool -s all give output very similar to what people post in #3 (i.e., ME still shows in PCI, but intelmetool fails at the end), but I'm not fully convinced... For example, if I apply me_cleaner to the 8MB part of the bios that comes from the lower chip, then it finishes successfully and produces a new bios ROM. However, if I apply me_cleaner to this new file, it claims to find ME again and then says that it successfully removed it, eventhough it should already be gone. Posts on the internet lead me to believe that this second application of me_cleaner should result in an error: "ME not found", but I don't have that.

Any tips?

corna, 2018-03-27:
`me_cleaner` will always find the ME firmware, it can't be completely removed (except for ME 6.0, but it's not your case).

This is the output of `me_cleaner -c` on an untouched ME image:

<pre>
Full image detected
The ME/TXE region goes from 0x3000 to 0x500000
Found FPT header at 0x3010
<b>Found 14 partition(s)</b>
Found FTPR header: FTPR partition spans from 0xf2000 to 0x1b2000
ME/TXE firmware version 6.1.10.1052
Public key match: Intel ME, firmware versions 6.x.x.x
Checking the FTPR RSA signature... VALID
</pre>

and this on the same image, after the cleaning process:

<pre>
Full image detected
The ME/TXE region goes from 0x3000 to 0x500000
Found FPT header at 0x3010
<b>Found 1 partition(s)</b>
Found FTPR header: FTPR partition spans from 0xf2000 to 0x1b2000
ME/TXE firmware version 6.1.10.1052
Public key match: Intel ME, firmware versions 6.x.x.x
Checking the FTPR RSA signature... VALID
</pre>

rennis250, 2018-03-27:
Ah, okay. Makes sense. I had a suspicion that it would register the deactivated case also and would give some message that the job had already been done. I understand. Am I to assume then that everything has indeed worked correctly? Even though the intelmetool initially says that I have a QC77 chipset (or something like that... I should double check) with a copy of ME that cannot be deactivated or disabled?

rennis250, 2018-03-27:
Ok, I double checked. It says I have a QM77 Express Chipset LPC Controller and "ME: Error Code:" says "No Error" and "ME: Progress Phase State:" says "Check to see if straps say ME DISABLED", which, according to https://github.com/corna/me_cleaner/wiki/Internal-flashing-with-OEM-firmware, means that it didn't work.

rennis250, 2018-03-27:
And at the end of the intelmetool output, it says:

```
ME: failed to become ready
ME: failed to become ready
ME: GET FW VERSION message failed
ME: failed to become ready
ME: failed to become ready
ME: GET FWCAPS message failed
```

corna, 2018-03-27:
Don't worry about the "cannot control or disable" message.

> And just so that I understand, it is okay that MEI is still on the PCI bus?

Yes, it's not a problem; sometimes it disappears, sometimes it remains.

> The "ME: Error Code:" says "No Error" and "ME: Progress Phase State:" says "Check to see if straps say ME DISABLED"

Whoops, I forgot to update the return codes on the wiki when using the AltMeDisable bit. Don't worry, that's the expected output.

Intel ME seems disabled on your PC.

rennis250, 2018-03-27:
Okay, great. That puts me at ease :) I'll leave this open until the error codes are updated for the AltMEDisable bit.

And naturally, thanks a lot for the hardwork on this tool. It's really nice! :+1:

closed 2018-03-30 by corna