corna/me_cleaner #186: HM70 & i3/i5/i7 30 Minute Restriction Help!

nzg187, 2018-04-02:
Sorry I am new to GitHub don't know if this is the correct place to post this but

Is me_cleaner able to get past/remove the 30 minute shut down restrictions intel has on the HM70? kinda losing my mind on this one.

I have a HP Compaq CQ45-802TU Notebook PC with the HM70 chipset that want to use my i7 3632qm 35w in.

Windows 8.1 64 bit

corna, 2018-04-02:
Yes, read [this](https://github.com/corna/me_cleaner/blob/master/README.md#what-can-be-done).

nzg187, 2018-04-02:
I read this post on https://www.win-raid.com/t2544f39-cpu-upgrade-leads-to-min-shutdown.html by plutomaniac 

"The Platform Controller Hub (PCH or chipset traditionally) is a chip which is soldered to the motherboard and has nothing to do with what BIOS/SPI chip you buy, reflash etc. The HM70 PCH supports only Pentium and Celeron processors and the restriction is hardware coded/enforced at the PCH itself. Nothing you do at the cpu, SPI chip or the latter's firmware will change that fact. So the SPI image regions and/or ME firmware in those have nothing to do with this restriction. The other laptop you mentioned uses HM76 chipset which can work with Core i-series cpus. Intel has a website with all their products and list the specifications, datasheets, what is compatible etc. Here is the page of the HM70 PCH cpu compatibility. To improve the performance of that particular system, pick the strongest of those Celeron or Pentium processors mentioned at the HM70 ark page."

Are you sure the HM70 can have the 30 minute restriction removed cause I'm so keen to get it done allready sold the laptop to a friend and I have spend the money so I need to get it sorted for him asap.

nzg187, 2018-04-02:
I read this post on https://www.win-raid.com/t2544f39-cpu-upgrade-leads-to-min-shutdown.html by plutomaniac

Special: Intel Management Engine » cpu upgrade leads to 30 ...<https://www.win-raid.com/t2544f39-cpu-upgrade-leads-to-min-shutdown.html>
www.win-raid.com
Hello, forgive me firstly for my lack of knowledge in this area, however I will continue on and try my best to describe the issue I am currently experiencing.

"The Platform Controller Hub (PCH or chipset traditionally) is a chip which is soldered to the motherboard and has nothing to do with what BIOS/SPI chip you buy, reflash etc. The HM70 PCH supports only Pentium and Celeron processors and the restriction is hardware coded/enforced at the PCH itself. Nothing you do at the cpu, SPI chip or the latter's firmware will change that fact. So the SPI image regions and/or ME firmware in those have nothing to do with this restriction. The other laptop you mentioned uses HM76 chipset which can work with Core i-series cpus. Intel has a website with all their products and list the specifications, datasheets, what is compatible etc. Here is the page of the HM70 PCH cpu compatibility. To improve the performance of that particular system, pick the strongest of those Celeron or Pentium processors mentioned at the HM70 ark page."

Are you sure the HM70 can have the 30 minute restriction removed cause I'm so keen to get it done allready sold the laptop to a friend and I have spend the money so I need to get it sorted for him asap.

Thanks in Advance

corna, 2018-04-02:
If your objective is to make a i3/i5/i7 work on that PCH I'm afraid it's something out of me_cleaner's scope: it could work, but it's up to you to test it. The 30-mins window I was referring to is the one enforced by Intel ME when its firmware is absent.

nzg187, 2018-04-02:
So I can remove the 30 mins timer with me_cleaner? that is all I am wanting.

nzg187, 2018-04-02:
[image]
All I get is invalid syntax with the commands

corna, 2018-04-02:
> So I can remove the 30 mins timer with me_cleaner? that is all I am wanting.

_me_cleaner_ doesn't trigger the 30 mins timer caused by an invalid Intel ME firmware, I know nothing about your 30 mins timer for an unsupported CPU.

>  All I get is invalid syntax with the commands

You don't have to run me_cleaner within a Python shell, just run `cmd.exe` and run that command (without the $, the dollar sign simply means that the command can be run as standard user).

As usual, be careful! You may have to unbrick your laptop with an external programmer if your firmware can't handle a modified ME image.

BoltOrion, 2018-05-05:
It this Work ...? anybody ? Someone ?
nzg187...?

catzy007, 2018-11-11:
This won't work. I tried using me_cleaner last night, replace pentium with i5 and it seem that 30 min shutdown still exist

topiga, 2020-01-21:
It works, you have to flash the modified bios that me_cleaner gives you. Update your bios with the OEM website. Then, buy a ch341a programmer, save the bios file, run me_cleaner with the bios file you just saved, and flash the modified one. Everything should work properly after.

catzy007, 2020-01-21:
Yes me cleaner works just fine. But if you want to use intel i3/5/7 sandybridge series in HM70 chipset, it will give you 30 minute shutdown https://catzy007.github.io/blog/#2018-11-04-disabling-intel-me

mukika009, 2020-04-17:
Hey guys, sry but im very noob how can I use this python thing? :D
Some body can tell it to me please step bystep how to use this thing?

dasdev20, 2020-07-02:
I checked script ME Cleaner on Leovo B590 (HM70). I dumped bios by SPI Programmer and flashed patched bios...its not worked and after ~30min system halted. I used CPU i3 3xxx series (35W). But, I not use option -S (HAP enable) when I patching. My notebook isn't booting if HAP enabled.

Fusseldieb, 2022-09-23:
Has anyone had a success with this chipset and me_cleaner?

GitMathHub, 2023-01-07:
Satellite C850, I had success.
But first ME and Flash descriptor were locked.
It is quite easy, anyway: there are two flashes on the board, at least on mine. One is BIOS, the other one FD, ME and a bunch of other things.
Dumped this second one with a programmer (EZP2010), initial 4K are the FD.
Unlock the BIOS substituting right bytes to FF (check winraid forum). Six bytes to be clear.
Do not dump or touch the BIOS flash, my flasher was not able to read and program in circuit, I had to recover it desoldering it.
Now FD and ME are unlocked.
I modified BIOS following the procedure https://youtu.be/ZL_NExttRQk (it is enough to change ME with Hex editor, and then reset settings about it, I put the same version 7, instead of the original 8).
I extracted the bios utility from original file, F8 version, and modified it.
Then I modified platform.ini first, allowing to overwrite same version bios file, and especially rewriting ME. The ME also file has been substituted with 7 version before.
Doing this reflash I have solved the problem. No more 30 minutes reset on Toshiba Satellite C850 HM70 version.
Some notes:
1) I didn't use meclean beaches my setup was very similar to the Brazilian man;
2) I didn't reflash bios directly because of problem I told, and one more: the dump was very different to bios file.

Happy modding!

kabal-pl, 2023-12-09:
> Has anyone had a success with this chipset and me_cleaner?
Yes! :)

Hello,
I'll share my experiences here on this topic.
I have laptop ASUS X401A-WX095V (intel HM70 chipset) - originally came with Celeron B820. 2 weeks ago I swapped CPU to i5-2410M (from Toshiba Satellite Pro C660 on HM65 chipset) and like other Users every 30 min laptop suddenly turned off. But...

Yesterday I found a solution (for me and my x401a) which is as follows:
1. Dumped BIOS from EEPROM using external programmer (I using latest BIOS for this laptop ver 212), no need to desolder,
2. Next used Hex editor for downgrade ME Region to 7.0.4.1197. I used stock/clean ME Region 7.0.4.1197 for this downgrade purpose,
3. I flashed back new modded BIOS with new ME Region,
..and after that, laptop was able to turned on, enter to BIOS, but couldn't boot from any connected media drive (ssd, usb)..., so I applied Linux version of "corna / me_cleaner " from github.
So last step was:
4. I runed "me_cleaner.py -S" script on my new modded BIOS with downgraded ME Region 7.0.4. In my case works first suggested option from me_cleaner instruction with capital S.

Thats it!!
After that my laptop is able to boot from any media drive and working without 30 min shutdown!

note:
-my Asus X401A on HM70 had original BIOS v212 with ME Region 8.0.10.1464
-donor Toshiba C660 on HM65 had BIOS v1.50 with ME Region 7.0.4.1197

Maybe this will help those who are still looking for a solution. Hope it does ;)

Best regards from Poland!

leonardputra, 2023-12-14:
Hi @kabal-pl, this is huge if true!
Can you elaborate more on the 2nd step?
How to downgrade ME region using Hex editor?

Thanks a lot!

kabal-pl, 2023-12-14:
> Hi @kabal-pl, this is huge if true! Can you elaborate more on the 2nd step? How to downgrade ME region using Hex editor?
> 
> Thanks a lot!

Hi @ leonardputra, Sure :) 
Take a look at yt videos about cleaning/changing me region using hex editor:
https://www.youtube.com/watch?v=zDMnD_t6n_4
https://www.youtube.com/watch?v=wYlZD-_K0zA

Videos in Polish language, but You can find simmilar videos in English ;)
Good luck ;)

Sanalkabuss, 2024-01-20:
Lenovo G580 type 20157
stock cpu celeron 1000m
upgrade cpu i5 2430m

Sorunsuz Çalışıyor...

Yapılan işlemler:

1. bios chip sökülür...
2. ch341 eprom cihazı ile bios dökümü alınır...
3. me anlyzer programı ile me versiyonuna bakılır..8.0.4.1441  1.5 MB SKU
4. winraid sitesinden Intel ME 7 Firmware Repository r29 indirilir
5 dosya açılır...içindeki dosyalardan 7.0.1.1141_1.5MB_PRD_RGN.bin başka bir yere kopyalanır
6. Sonra HxD editörle orjinal dosya açılır...başka bir pencerede 7.0.1.1141_1.5MB_PRD_RGN.bin açılır
7. adres olarak 00001000 satırına gelinir.öteki penceredeki 7.0.1.1141 tamamı işaretlenip kopyalanır.
8 sonra orjinal dosyada 00001000 satırının en başı işaretlenerek sağ tık üstüne yazarak yapıştırılır.
9 yukarıdaki menulerden chksum sha 256 ile tamamlanır
10. dosya hazır...bundan sonra dosyayı ch341 ile tekrar flaşlanır.
11. 30 dakika olayı bitti ve bilgisayarda sorun yok...herşey normal

Super-Matter, 2024-01-30:
I can confirm that the me swap trick works !!!

Thanks for help!

GOJOAmimir, 2024-02-03:
> I can confirm that the me swap trick works !!!
> 
> Thanks for help! @kabal-pl 

Do you need that exact 7.0.4.1197 version?

Super-Matter, 2024-02-03:
> Do you need that exact 7.0.4.1197 version?

I don't think so, but I used it for the sake of a clean experiment. The thing is to have me regions of the same size (in that case of 1.5mb) and the donor one should be a lower version 7.0.4.1197 is the last 7.0.4 release as far as I get. 

All the files needed were downloaded off GDrive link below the video. The guy does some unnecessary steps when editing the hex file, you can just paste write me region onto selected one. Be sure to calculate the end address properly (it was 17FFF0 for me if I'm not mistaken). 

If you have any other questions, feel free to ask. 

P.S.  I didn't get most of the guy's talking because he speaks Portuguese, and my Brain Operating System supports English and Russian only, so I could miss some crucial information

Super-Matter, 2024-02-03:
The only caveat I ran into was flashing bios back: I've spent an hour until I realised that you need to blank the chip first and only after that the flash goes smoothly and without any verification errors

kabal-pl, 2024-03-06:
> hello i need a little help with the way you do this. @kabal-pl i have an c50-a toshiba with celeron cpu. i changed it to an i5 2450m and did the steps as in the video. downgraded the me in the bios file then flashed it back to the chip. the laptop boot normally in windows. everything works perfect and it doesnt have timer to close at 30mins. i can restart it ok. but when i completely close it then it doesnt post again. it gets in a close-open loop. if i flash the bios again then it opens again as before. any ideas what i am doing wrong?

Hi!

Have You tried with original ME region from Celeron CPU?

kabal-pl, 2024-03-06:
> thank you for your time @kabal-pl yes i tested with original me region and it didnt work. 
Ok.

> the me_cleaner script i cant run. i installed python did the steps as i read but it not generate any file
Too bad. 
Try with usb bootable linux and use me_cleaner - that needs to be done.

aetolofitis, 2024-03-07:
@kabal-pl you were right. i tried with bootable linux and run me cleaner to the bios file with the downgraded 7.0xxxx ME and and after that it worked. it solved the problem i had with post. fantastic. i am running i5 on hm70 chipset now and i am trying to improve a little the cooling system so i can add an i7 4 core...hahaha
have you tested any i7 on hm70?

kabal-pl, 2024-03-07:
> @kabal-pl you were right. i tried with bootable linux and run me cleaner to the bios file with the downgraded 7.0xxxx ME and and after that it worked. it solved the problem i had with post. fantastic. i am running i5 on hm70 chipset now and i will try to improve a little the cooling system so i can add an i7 4 core...hahaha have you tested any i7 on hm70?

Glad to see that its working ;) 
Nope, dont have any i7 on this socket.

QuanTrieuPCYT, 2024-03-14:
Trying to modify the BIOS of a Toshiba Dynabook Satellite B452-H here (HM70 chipset), the thing booted **but** boot time in Windows being increased significantly by around 20 to 30 seconds, and the fan speed went full all the time (not controllable in BIOS). Did some tinkering and found out that: the machine booted fast with factory ME, but slowly with maxed out fan speed all the time.

I could live with the reduced booting speed as I planned to use Linux Mint on the thing, still any ideas to fix the maxed out speed of the built-in thermal fan?

kabal-pl, 2024-03-14:
> > Trying to modify the BIOS of a Toshiba Dynabook Satellite B452-H here (HM70 chipset), the thing booted **but** boot time in Windows being increased significantly by around 20 to 30 seconds, and the fan speed went full all the time (not controllable in BIOS). Did some tinkering and found out that: the machine booted fast with factory ME, but slowly with maxed out fan speed all the time.
> > I could live with the reduced booting speed as I planned to use Linux Mint on the thing, still any ideas to fix the maxed out speed of the built-in thermal fan?
> 
> Update on the matter: The machine booted fine, fan speed normalized with a v7 ME-downgraded bios. Cleaning the ME resulted in the machine being stuck for a while when booting Windows again and the fan also ramped up to its max speed. Our goal right now is just to make Core i-series CPUs work on the chipset, so I decided not to clean the ME again on this thing lol
> 
> Still, greatly thanks @kabal-pl for the method!

I'm glad to see that another user has a positive result :) 
Cheers!

QuanTrieuPCYT, 2024-03-15:
> Update on the matter: The machine booted fine, fan speed normalized with a v7 ME-downgraded bios. Cleaning the ME resulted in the machine being stuck for a while when booting Windows again and the [...]

[image]

Got myself a cheap i3-3120M for 2$ to test out the new BIOS. Seems to be working, fan speed ramps up with high load like normal, everything is fluid!

@aetolofitis If you want I also plan on purchasing an i7-3720QM (4 cores 8 threads) and will definitely test that CPU out on this chipset.

aetolofitis, 2024-03-20:
@QuanTrieuPCYT hi
i bought an i7 2670qm it still works but its 45watt so the temperatures were very high and it couldnt manage it so i removed it and got the i5 back. after that i saw the i7 3632qm which is 4 core and 35watt. so i ordered it and wait to test that too . hopefully because of the tpd i think that the cooling will manage it. 
what version of me you used for the i3 3120m?

QuanTrieuPCYT, 2024-03-20:
> @QuanTrieuPCYT hi
>
> i bought an i7 2670qm it still works but its 45watt so the temperatures were very high and it couldnt manage it so i removed it and got the i5 back. after that i saw the i7 3632qm [...]

Using 7.0.4.1197 as well, the 1.5MB SKU for my BIOS.

gamabaja, 2024-03-29:
> > ¿Alguien ha tenido éxito con este chipset y me_cleaner?
> 
> ¡Sí! :)
> 
> Hola, ha pasado un tiempo, pero compartiré mis experiencias aquí sobre este tema. Tengo una computadora portátil ASUS X401A-WX095V (chipset Intel HM70); originalmente venía con Celeron B820. Hace 2 semanas cambié la CPU a i5-2410M (de Toshiba Satellite Pro C660 en el chipset HM65) y, como otros usuarios, cada 30 minutos, la computadora portátil se apaga repentinamente. Pero...
> 
> Ayer encontré una solución (para mí y mi x401a) que es la siguiente:
> 
> 1. BIOS volcado de EEPROM usando un programador externo (yo uso el BIOS más reciente para esta computadora portátil versión 212), no es necesario desoldar,
> 2. A continuación se utilizó el editor hexadecimal para degradar la región ME a 7.0.4.1197. Utilicé stock/clean ME Region 7.0.4.1197 para este propósito de degradación,
> 3. Actualicé el nuevo BIOS modificado con la nueva región ME... 
>    y después de eso, la computadora portátil pudo encenderse, ingresar al BIOS, pero no pudo arrancar desde ninguna unidad de medios conectada (sata, usb)..., así que apliqué Versión para Linux de "corna/me_cleaner" de github. 
>    Entonces el último paso fue:
> 4. Ejecuté el script "me_cleaner.py -S" en mi nuevo BIOS modificado con ME Region 7.0.4 degradada (en mi caso, funciona la primera opción sugerida con S mayúscula de la instrucción me_cleaner) y nuevamente mostré el segundo BIOS modificado.
> 
> ¡¡Eso es todo!! ¡Después de eso, mi computadora portátil puede arrancar desde cualquier unidad de medios y funcionar sin apagarse durante 30 minutos!
> 
> nota: -mi Asus X401A en HM70 tenía BIOS v212 original con ME Region 8.0.10.1464 -el donante Toshiba C660 en HM65 tenía BIOS v1.50 con ME Region 7.0.4.1197
> 
> Quizás esto ayude a quienes todavía están buscando una solución. Espero que así sea ;)
> 
> Saludos cordiales desde Polonia!

hola, soy muy novato y no se nada sobre modificar bios, alguien tiene un videos exacto de lo que tengo que hacer, aunque este en otro idioma puedo tratar de traducirlo, compre un i7 y se apaga cada 30min, ayuda por favor no quiero perder mi dinero:(

QuanTrieuPCYT, 2024-04-10:
Update: 

We managed to update from an Intel Core i3-3120M to an i7-3720QM, and with some thermal modding, the temps are acceptable.

[image]

aetolofitis, 2024-04-10:
hi again. good news. i upgrade mine to i7 3632qm which is 35w. so totally the laptop from celeron 1005m went to an i7. 
found on ebay a heatsink with double chopper from a similar model and added that too. the temps where at max 81c. after that i put liquid metal too so now it never gets past 72-73c. seriously i never thought that this laptop can be so good.

Naddly, 2024-05-21:
> @kabal-pl you were right. i tried with bootable linux and run me cleaner to the bios file with the downgraded 7.0xxxx ME and and after that it worked. it solved the problem i had with post. [...]

It also worked with my Acer Aspire E1-531-2606 (LA-7912P motherboard) that I had almost the same issue as you, but in my case it just worked one single time after the fist BIOS flash and then never again, even reflashing the same exact file did nothing.
This laptop in specific had some weird things going on with it's BIOS, I was able to solve all of them and am available to help if anyone is having issues with it or similar models.

christen90, 2024-05-23:
> I was able to solve all of them and am available to help if anyone is having issues with it or sim
Hi @Naddly,
I have the same laptop and would also like to upgrade to i7 3632qm. Could you please share detailed step by step instructions?

BR Christian

Naddly, 2024-05-24:
@christen90 I already plan to make a detailed step by step with pictures and all and post to winraid forums, but I'm a bit of a perfeccionist and don't have much time to write it right now.
If you want it done ASAP, I can provide an already modified v2.21 BIOS without MAC, SN etc, with the instructions for you to add yours and flash it.
There are other less intrusive methods I have to test, but I know for sure that a CH341A (or similar) will work 100%, so make sure that you have one in working conditions before doing any BIOS mods.

christen90, 2024-05-24:
@Naddly 
That sounds great. Kindly post a link then.

Processor and programmer are ordered but not delivered yet. No pressure. 🙂

aetolofitis, 2024-05-24:
I am also here to help if anyone needs. its great to give life to these old laptops

Naddly, 2024-05-25:
@christen90 I'll _try_ to explain it quickly. Here's the file: [Acer E1-531 FullUnlock.zip](https://github.com/corna/me_cleaner/files/15431123/Acer.E1-531.FullUnlock.zip)

### DISCLAIMER - Do everything at your own risk! The steps listed here can turn your laptop into a over-engineered paper weight. Read it all BEFORE doing anything to your laptop!
I'll try to help with any problems anyone might have, but I can't guarantee that it will work and I'm not responsible for what you do to your devices. Be sure to have backups and not to risk any device you depend on. :)

Well, first of all I have to inform you that the LA-7912P motherboard has a bit of an unusual BIOS setup, there are actually 2 distinct ICs. The main 4MB one _(W25Q32BVSIG)_ under the Wi-Fi card and a secondary 1MB one _(MX25L8008E)_ directly behind it, on the other side of the motherboard close to the Super I/O chip.
I'll assume you already know how to use the CH341A, SOIC8 clip and the NeoProgrammer app, if tou don't, you can just search some tutorials on YouTube.

Before doing any BIOS reading/flashing on this motherboard using the SOIC8 clip you will have to remove the wifi card (duh), Battery Pack, CMOS Battery, AC Charger and any storage you have (RAM and CPU can stay).
After that, do it in the following order:
- Connect the SOIC8 clip to your programmer _(mind the orientation of the pins)_;
- Securely connect the clip on the desired IC _(again, be cautious with the orientation, it's easy to mess up)_;
- Connect the programmer to the other PC/Laptop;
- Carefully plug in the laptop's charger.

Now you can do anything with the IC. To remove the programmer just follow it backwards.

The next (and most important) step is to backup your current BIOS, you must dump the contents of the 2 ICs (preferably more then 1 time, to be sure) and keep them in a safe place, maybe even have some copies somewhere.

Next you'll need a Hex editor, HxD is free and a good option. Using it you'll have to extract some data from your 4MB backup and put inside the one I uploaded to you.

Open your 4MB backup file with HxD, tap _Ctrl + E_, type: Start-offset: 180000, End-offset: 182FFF and click ok.
[image]
Copy the selection with Ctrl + C and open the 4MB file I uploaded to you (inside the "Split Files" folder) with HxD.
Tap Ctrl + E again and fill with the same offsets as above, click ok and do a Ctrl + V, it will replace my data with yours.
After that we'll have to verify that you did it right, confirm that the size of the file wasn't altered selecting all with Ctrl + A and looking for "Block(h): 0-3FFFFF".
[image]
If yours looks like this, save the edited 4MB file and go to the next step, if not, close the unsaved edited file and try it again.

Now you can flash the files, the 4MB edited file will go to the W25Q32BVSIG IC under the Wi-Fi card and my unedited 1MB file to the MX25L8008E close to the Super I/O.

Remember to check all boxes on the NeoProgramer and to use the "Detect" option.
[image]
Now you can just flash both of them.

I really hope you understood everything, if you didn't, feel free to ask me about anything
I also have something that could be useful, the official Service Guide in PDF so it's easier to disassemble: 
[aspire_e1531 service guide.pdf](https://github.com/corna/me_cleaner/files/15439876/aspire_e1531.service.guide.pdf)

And pictures of the 2 ICs so you know what to look for:

[image]
Left: MX25L8008E (1MB);     Right: W25Q32BVSIG (4MB)

I hope it goes well for you.

Naddly, 2024-05-25:
Another thing is that this i7 **will** overheat, my i5-3380M (35W) can barely maintain 3.3GHz (when it can turbo up to 3.6GHz) at 100% use with a synthetic load, reaching temps up to 90°C. It's the same thermal paste that does just fine with my Desktop i7-3770 @ 4GHz.
My fan speed also seems to be tied to the CPU use, like, if the CPU hit a 100% load spike for 2 seconds the fan will rapidly spin up to 100% too.
My laptop has this cooler:
[image]

But I know there are other models that _**seem**_ to be compatible with the LA-7912P Motherboard, like those:

(AT0HI00B0R0 & AT0N70020R0)
[image]

(AT0HI009DR0)
[image]

(AT0N4005A or AT0N4004AA0, idk)
[image]

(AT0HI007DA0)
[image]

I haven't tested any of those and didn't see no one testing them, but could be a nice future upgrade, even if your laptop, as mine, doesn't have any dedicated GPU. AT0HI00B0R0 and AT0N70020R0 will fit for sure, the others I don't know.

aetolofitis, 2024-05-25:
in your i case most propably i would take this AT0HI00B0R0 & AT0N70020R0 and cut the extra . combined with a good thermal paste i dont think that you will have any heat problems. in my laptop with the bigger cooler i had good temperatures with noctua nt-1h and i became greedy and added liquid metal too.

what thermal paste you use now and you have these temperatures?

Naddly, 2024-05-27:
@aetolofitis I don't think that cutting the extra would be a good idea, because the liquid inside the heatpipe would just evaporate rendering it useless. I plan to just let the extra be there.
Another thing that I noticed is that the PCH stays at 70-80C almost all the time, this can't be good in long term, right? Would be nice having one of those coolers with this little extension for the PCH like the AT0HI007DA0 if someone could confirm they work with this motherboard.

The thermal paste I used has Rise Frost written on it, and as I mentioned it's the same that I had success with my i7-3770, a 77W desktop CPU.

coa9000, 2024-05-31:
@kabal-pl @QuanTrieuPCYT @Naddly I have a Toshiba Satellite C850 with Intel Pentium B970, and I would like to use another CPU like i3 2348m, i have one just laying around... but i don't know how to swap the ME region. In downloaded bios from dynabook, i've got two files with fd file format, one is F8_ME_FW.fd other one is F8680.fd so can someone explain how this is done with this Insyde bios files? I assume that the one with "me" in the file name has to be edited? It is exactly 1.5MB in size. Thank you!

christen90, 2024-06-02:
Hello @Naddly,
my upgrade using your instructions was of success. My old laptop now has AES, Miracast, HyperThreading and VT enabled.  Using it feels a lot more smooth than before. Also your modded bios unlocked a bunch of additional settings like fan-control. Thank's a lot. Feel free to post a paypal me link. I would like to send a small thankyou.

[image]
Thermal behaviour looks ok for me. Yes, it reaches up to 90 °C but as thermal limit of i7 is 105 °C I consider it as enough headroom. Normal usecase for my case is not synthetic load. But I will keep it in mind if I want to run simulations or something heavy calculation causing apps. Interesting was also that if I combine CPUz stress test with GPU stress test (furmark) that the cpu was throttling frequency to 1.2 GHz. Probably in i7 GPU + CPU under full load exceeding 30 W TDP. 
[CT-Log 2024-06-02 10-08-46.csv](https://github.com/user-attachments/files/15524625/CT-Log.2024-06-02.10-08-46.csv)

If someone is not having experience with the BIOS recorder I add some photos of the wiring:
1MB chip
[image]

4MB chip
[image]
My PCB has a marking U36. If someone is searching for it.
[image]

The only thing to do now would be to get matching RAM (1600 MHz). Does someone tested to put more than 8 GB in?

BR Christian

Naddly, 2024-06-02:
I received a notification with a reply in this post that mentioned me, but it's not showing up for me.
@coa9000 What happened to your comment? I would be happy to try and help you. 
You can send me the files or I can make a little tutorial on how you can do it yourself if you want. It's relatively easy to do it using UEFITool and HxD.

And @christen90 I'm happy that it worked out for you, I really am.
You better be cautious with those new BIOS settings, some of them can brick your system and need a reflash, it happened to me when I tried to set the pre-allocated VRAM to 1GB, had to to a reflash. My laptop seems to only work with 512MB VRAM or less. It's also better not to mess with any power settings, as those could do irreversible damage to your motherboard.

About having 16GB RAM, yes it works. I have 2x8GB 1600MHz DDR3L on mine and I actually find them to be a bit overkill lol, but yeah it totally works.

I would like to inform that I did some research and saw that the AT0HI007DA0 cooler will work just fine with the LA-7912P motherboard. Mine has already been shipped from china (AliExpress) and I'll post an update when I have fully tested it.
It seems to be a 45W capable cooler so should be enough for our 35W CPU

Also, I appreciate it but I actually don't know how to create a PayPal me link, I have an account but couldn't find the option.

coa9000, 2024-06-02:
I'm not sure why, but I deleted my first comment. I have a Toshiba Satellite C850, HM70 chipset AU/NZ version. And I downloaded the latest bios from dynabook site. Flashed that with Insyde Flasher from Windows, and flash go as it should, i checked with aida64 and it showed it updated correctly. I then flashed "unlocked" bios from bios-mods site, and that went as it should, but as my cpu don't support VTX Pentium B970, nor any of the options that bios unlocked I reverted to unmodded version... but as you already know about 30 min timer, when I upgraded cpu and changed cooling to better one, with two pipes, I discovered that it will shutdown itself exactly every 30 min.

So, my problem is when I unpack the exe with 7zip, I've got two files in .FD format. One named F8_ME_FW.FD which is 1.5mb in size and the other F8680.FD, 8mb in size, how this can and what should be edited to make it work with i5 sandy/ivy? Thank you!

Naddly, 2024-06-03:
@coa9000 I just had a look at the BIOS you provided in the link and what I can say is that your best chances are using an external programmer, like the one christen90 used. 
You would extract your currently working BIOS and I would do the ME Mod for you or make a little tutorial on how you could do it yourself, your choice.

I have to warn you all that some laptop BIOS just **CAN'T** be read nor flashed while soldered on the motherboard, it's hit and miss, me and christen90 have had the luck that our motherboard only needs the charger plugged in for it to work but that may not be the case to you.
I advise you to: 
- Get a cheap CH341 Programmer _(or anything that can reprogram ICs, a Raspberry Pi can do it too)_ with a SOIC8 clip.
- Try to read and save your BIOS, and proceed to try and mod it **ONLY** if you can reliably read it multiple times in a row getting the same exact file _(check using hash, like SHA-256 or any other really)_. 
- If you can't read it, you're out of luck, just stop right there and don't try nothing BIOS related until you can somehow safely remove the BIOS chip and put it back again, because if you brick your BIOS you WILL have to remove it to reprogram it.

This is also good so you don't risk losing things like your Serial Number and MAC Address.

The basic modification to have i3, i5 and i7 working on the HM70 consists of just replacing your current Intel ME v8.X by a v7.X, in my case I used [v7.0.4.1197](https://github.com/user-attachments/files/15528416/7.0.4.1197_1.5MB_PRD.zip), it's as simple as `ME Region > right click > Replace as is` using [UEFITool 28](https://github.com/LongSoft/UEFITool/releases/tag/0.28.0) _([not NE](https://github.com/LongSoft/UEFITool/issues/67))_, this will probably work on most HM70 motherboards.
The LA-7912P _(Acer Aspire E1-531 and family)_ was the only one I've seen to need also the use of the [ME Cleaner by Corna](https://github.com/corna/me_cleaner) so it can work after the first boot and _(I hope)_ forever, but yours probably won't need that extra step.

It's all a very big risk, it could go wrong even if you do everything right, so make sure you don't heavily depend on this laptop and are ok with beaking it.

christen90, 2024-06-03:
> me and christen90 have had the luck that our motherboard only needs the charger plugged in for it to work

Thanks for the reminder. I forgot to write.
I was able to connect/use the bios recorder without having the charger plugged in.

BR Christian

Naddly, 2024-06-03:
@christen90 so it seems to be a per device thing? Or we have different board revisions? 
Well, I don't really know for sure, all I know is that the data I got from any of the 2 BIOS chip was all scrambled and different everytime I tried reading it without a charger.

And for the record, here is my E1-531-2606 in all it's glory and half-naked:
[image]

coa9000, 2024-06-03:
Thank you @Naddly . I don't have any flasher at the moment, but i will sure get one soon. Can rpi Pico be used to extract the .rom from chip? I was planning to buy that for some game console mods... I don't care if i brick this laptop, it's very old and cheap to get... I also downloaded clean ME 7 pack from win-raid, and I see there are files exactly the same size as my F8_ME_FW.FD but those downloaded are in bin format. I will inspect this a bit more. I am just thinking to find the same one you used and just rename it as "F8_ME_FW.FD" and try to flash that with Insyde Bios Flash. I will post here later if i do this. Anyway thank you once again for your quick and clear reply!

Naddly, 2024-06-03:
@coa9000 a RPi can do the job, but I would not be able to help you with it as I don't know how to use it nor have one myself. 
Anyway a CH341 programmer is so cheap that it's just worth it to buy one along with it's SOIC8 clip to program ICs (hopefully ) without removing them.

I think I can prepare an update file already modded with the v7 Intel ME, here's it: [F8.zip](https://github.com/user-attachments/files/15539257/F8.zip)
But I have to warn you that it will be risky to use without having a backup made directly from your BIOS chip.

The changes I did to this file was: 
- Replace the original `F8_ME_FW.FD` version 8 to a version 7;
- Replace the ME inside your 6MB BIOS usind HxD, they were different sizes but all the extra half megabyte that your BIOS file had was blank, all FFs, so I just replaced the exact size of the ME region ignoring all the extra FFs;

I also modified the `platform.ini` so the settings are as follow: 
[image]
(do a backup before flashing)

[image]
(confirm before flashing)

[image]
(flash the ME region too)

I don't exactly know how this BackupROM works but we might as well test it, it's supposed to backup your current BIOS to the C drive. So in the worst case scenario you'll still have a working backup to recover your laptop.

As you could guess, I don't have any way to test it before you do, so you are assuming all the risks of what can happen.
I still strongly advise you to get a programmer ASAP and try to do a backup before any bios mod.

Fusseldieb, 2024-06-03:
I advise you to:
> * Get a cheap CH341 Programmer _(or anything that can reprogram ICs, a Raspberry Pi can do it too)_ with a SOIC8 clip.
>
>     * Try to read and save your BIOS, and proceed to try and mod it **ONLY** [...]
Something worth noting is that the BIOS chip is usually readable if you remove the CPU from its socket.

Naddly, 2024-06-03:
@Fusseldieb yes, but there are BIOS that can't be read even removing everything you can.
I actually have a laptop motherboard that is like this and I'm waiting for a new soldering station to arrive so I can remove the IC and flash a modified bios that supports 3rd gen Intel, as the original doesn't have the 3rd gen Microcode.

coa9000, 2024-06-04:
> @coa9000 a RPi can do the job, but I would not be able to help you with it as I don't know how to use it nor have one myself. Anyway a CH341 programmer is so cheap that it's just worth it to buy one along with it's SOIC8 clip to program ICs (hopefully ) without removing them.
> 
> I think I can prepare an update file already modded with the v7 Intel ME, here's it: [F8.zip](https://github.com/user-attachments/files/15539257/F8.zip) But I have to warn you that it will be risky to use without having a backup made directly from your BIOS chip.
> 
> The changes I did to this file was:
> 
> * Replace the original `F8_ME_FW.FD` version 8 to a version 7;
> * Replace the ME inside your 6MB BIOS using HxD, they were different sizes but all the extra half megabyte that your BIOS file had was blank, all FFs, so I just replaced the exact size of the ME region ignoring all the extra FFs;
> 
> I also modified the `platform.ini` so the settings are as follow: ![change1](https://private-user-images.githubusercontent.com/69136951/336221728-3b561cb1-5a1b-426c-906d-050462e404bd.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc0OTQ5MzMsIm5iZiI6MTcxNzQ5NDYzMywicGF0aCI6Ii82OTEzNjk1MS8zMzYyMjE3MjgtM2I1NjFjYjEtNWExYi00MjZjLTkwNmQtMDUwNDYyZTQwNGJkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjA0VDA5NTAzM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPThmNjZjYmY0Zjk4YmI4NjBmODQxMzY0M2I5ZDMwYTIyY2IxYWFkYTUzNWE3ZGE2NDRiMGZlMDZjMzIxYjEzOTImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.R-zwrca6sm-p4YOD0as7jHC9MDg0LHDen7vxU-T_5rs) (do a backup before flashing)
> 
> ![change2](https://private-user-images.githubusercontent.com/69136951/336221779-ddfc8c89-8c26-4234-9ee6-f57ce5370442.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc0OTQ5MzMsIm5iZiI6MTcxNzQ5NDYzMywicGF0aCI6Ii82OTEzNjk1MS8zMzYyMjE3NzktZGRmYzhjODktOGMyNi00MjM0LTllZTYtZjU3Y2U1MzcwNDQyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjA0VDA5NTAzM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWUzZmFkNzZlNDg5MjdiZTkyYWZmYjkwMTAzNjZiZjY4YjIwYzkwNTk4OGNiOGU2MzFkNmViNWFlODkwNGRhMDImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.m9005Uy-bCTsYiaW7WWRCtg4QooB6_DYaZbRVlFJrOE) (confirm before flashing)
> 
> ![change3](https://private-user-images.githubusercontent.com/69136951/336221808-93c8a203-22df-4d53-a9fb-b624ae57b929.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc0OTQ5MzMsIm5iZiI6MTcxNzQ5NDYzMywicGF0aCI6Ii82OTEzNjk1MS8zMzYyMjE4MDgtOTNjOGEyMDMtMjJkZi00ZDUzLWE5ZmItYjYyNGFlNTdiOTI5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjA0VDA5NTAzM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTU4NmJhZTIzOGJkZTUwN2M3ZGMxYzc2MzJmMzZiYTczYmFjNTk0MDM3ZmMwMGFlZDFkNWM2M2EzZGJkZmYxOTEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.v1ti__q5iAQ6Ai9OLLPE6ZoYC7RcCpCdvjTl-3Dy5Gg) (flash the ME region too)
> 
> I don't exactly know how this BackupROM works but we might as well test it, it's supposed to backup your current BIOS to the C drive. So in the worst case scenario you'll still have a working backup to recover your laptop.
> 
> As you could guess, I don't have any way to test it before you do, so you are assuming all the risks of what can happen. I still strongly advise you to get a programmer ASAP and try to do a backup before any bios mod.

Thank you mate, I have ordered ch341 online in my country for 8 euro... it will arrive in a couple days, so I will wait just to be sure I can unbrick it if anything goes wrong.

Naddly, 2024-06-05:
@coa9000 I'm glad you took my advise. 8 euro for this programmer is almost free, too little for anyone to risk modding their BIOS without it.

@christen90 I just noticed that sometimes my Ethernet port just stops working, it shows that it's plugged but with no internet, even when I unplug the cable it still shows it's plugged.
It only works again when I go to Network Connections in the Control Panel, disable and activate the Ethernet adapter.

It's kinda random and rare to happen, but I noticed one _very specific_ moment it will always happen. It is when I'm streaming Forza Horizon 4 (steam) from another PC to my E1-531 and I try to connect my Dualshock 4 via Bluetooth, doing that, the controller will work for a couple of seconds before the connection drop and my stream freeze. I have a Gigabit Ethernet link between the 2.

My first thought is that this can be caused by something BIOS related, as I don't remember it happening before I started doing mods on it.
I ask you to keep an eye on that to see if it happens to you at any moment, so we would confirm it's BIOS related and try to fix it.

NihadTermux, 2024-06-18:
Hello, who can help me?
I have a hm70 Intel pentium b960 laptop and I have a problem.

Naddly, 2024-06-18:
@NihadTermux What exactly is the problem you are having? Want to remove the 30 min restriction? It would also be nice to know the exact model of your laptop.

NihadTermux, 2024-06-18:
> @NihadTermux What exactly is the problem you are having? Want to remove the 30 min restriction? It would also be nice to know the exact model of your laptop.

[image]

[image]

There are two bios here. Which one should I install? There is extraced, there is stock, there is extraced bios. After I edited it with Intel Me Easy, this is what happened. The flash tool image text is also gone.

[63 more replies]