corna/me_cleaner #194: Flashing on Dell XPS 9530?

ArchangeGabriel, 2018-04-19:
Hi there,

I’ve been following this project for a while and I’m going to try it for myself soon. I have a Dell XPS 9530, on which I have found the following chip that I suspect to be the BIOS one:
[image]

According to [this documentation](https://github.com/corna/me_cleaner/files/1928498/w25q64fv.pdf), it should be a SOIC-8 chip if I got everything right.

I’ve not located any other similar chip, but I guess the dump will tell us.

What kind of tools would you suggest to flash it? I have a Cubieboard available, but I’m not sure how to use it for SPI. Next, I don’t know what is the safest way to wire the board to the chip (SOIC clip, SMD clip, something else?).

Any help will be welcome.

corna, 2018-04-19:
(note that me_cleaner didn't work in the past, see #34, however the situation may have changed with the HAP bit)

A SOIC clip is cheap and easy to use, go for it.

The cubieboard should have a SPI interface, but I don't know how to activate it or where it is located on the GPIO header. Otherwise, if you want something easy, you can buy the CH341A programmer (some € on eBay).

ArchangeGabriel, 2018-04-19:
#34 is for a newer model (9550, based on Skylake, while mine is 9530, based on Haswell). However in #34 you say you hate SOIC clip and recommend SMD ones, is that correct?

jjurkus, 2018-04-19:
Those SOIC clips work pretty good when it's new. If you keep connecting it daily to other stuff it wears quite fast.

I currently have one of those Pomona SOIC clips, works well, but sometimes it's a pain to get it to connect well. If it breaks I will probably use those seperate SMD clips (and pay more attention to pinout).

corna, 2018-04-20:
> #34 is for a newer model (9550, based on Skylake, while mine is 9530, based on Haswell).

My bad, I didn't realize it was a different model.

> However in #34 you say you hate SOIC clip and recommend SMD ones, is that correct?

I've tried a few good one and I've changed my mind, if you buy a decent quality one they're quite handy.

ArchangeGabriel, 2018-04-23:
I’ve ordered a CH341A and a SOIC clip, they will be at home in some weeks~months (the times it takes to ship from China). I’ll keep you updated.

ArchangeGabriel, 2018-05-26:
So as a matter of fact I received them almost a month ago but did not have time to try them before today.

I’m having a hard time connecting the clip correctly, or maybe it’s the CH341A that is not working well…

After around one hundred attempts, I’ve finally go two dumps… that don’t match. The good news is that it’s the BIOS+ME as expected, and me_cleaner was able to modify it. But I don’t want to flash that as long as I don’t have two identical dumps.

So, let’s describe a bit more the issue. When I plug the CH341A alone and try to dump, I get this:
```
$ sudo flashrom -p ch341a_spi -r original_dump.bin
flashrom v1.0 on Linux 4.16.11-1-ARCH (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
No EEPROM/flash device found.
Note: flashrom can never write if the flash chip isn't found automatically.
```
At this point, the red power led on the CH341A is very bright. Then I plug the clip onto it, and the red led is dimmed quite noticeably. Then, attempting to dump:
```
$ sudo flashrom -p ch341a_spi -r original_dump.bin
flashrom v1.0 on Linux 4.16.11-1-ARCH (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Couldn't open device 1a86:5512.
Error: Programmer initialization failed.
```

Is it an issue with the clip, the CH341A, me or something else? In the first case, should I buy a Pomona 5250 clip instead of the one that came with my CH341A?

Thanks in advance for your help!

jjurkus, 2018-05-27:
I am using a FT232H, which has a setting to make it run slower.  Maybe the CH341A has something like that?
```
flashrom -p ft2232_spi:type=232H,divisor=10
```
Other issue could be that you're powering the entire laptop through the CH341A. Try to measure VCC on the chip. I've had reading issues with a certain laptop and the voltage had gone down to about 0,8 Volts. I've had some luck by reading while having the laptop on, but ymmv and it may break things.

corna, 2018-05-27:
As @jjurkus pointed out, it seems like a power supply issue. Writing the chip with the laptop powered on it very dangerous and you risk to break the laptop, instead connect a voltage power supply (like a laboratory one or, if you don't have it, use the 3.3V line, orange cable IIRC, of the PSU of a standard PC) with the VCC and GND pins of the chip, while the others to the CH341A. Then connect the GND of the CH341A with the GND of the PSU, then try again.

ArchangeGabriel, 2018-05-27:
Hum, power supply issue makes sense. Is there any risk in just plugging its power cable to the laptop without powering it on? Maybe that could solve the issue.

Else I will have to get more stuff, because I don’t have a voltage power supply nor a PC PSU available, neither a way to wire it to the chip (I just have the SOIC clip).

I’m not even sure how to measure VCC currently (maybe directly on the CH341A?), since once the SOIC clip is in position, I cannot access the chip pins anymore…

ArchangeGabriel, 2018-05-27:
OK, measuring on the CH341A (well on the connector above it) between pin 4 and 8, I indeed get 3.3V while not wired to the chip, but 0.8V when wired. So that’s definitively the issue. Waiting for your on input on whether plugging the laptop is a good idea or not before envisaging to go further in acquiring or seeking for the required equipment.

corna, 2018-05-28:
> So that’s definitively the issue.

Yep

> Is there any risk in just plugging its power cable to the laptop without powering it on?

It depends on the hardware, there isn't a general answer.

You can connect the programmer to the chip with the cable inserted or the PC powered on (without connecting the VCC line of the CH341A), however:
 * you have to insert resistors (4.7 kΩ is a common value) between the programmer and the chip (MOSI, SCK and CS lines), otherwise you risk to fry either the programmer (bad) or the laptop (worse)
 * the read/write may fail

ArchangeGabriel, 2018-05-28:
Hum OK, that is pretty as much hassle as powering the chip externally (I still need something else than my SOIC clip, and to get more stuff —resistors, etc.).

So I will go the voltage PSU way. Any recommendation on hardware and means to wire everything together?

corna, 2018-05-28:
You probably need some SMD clips, which are a bit more complicated to wire but they're much more flexible. Any PSU will suffice, but if you could first measure the 3.3V current with the CH341A (just to be sure you're not short-circuiting anything) it would be great.

corna, 2018-05-28:
Alternatively you can also use two brand-new non-rechargeable alkaline batteries in series, which should provide 3.0-3.2 V.

ArchangeGabriel, 2018-05-28:
You mean measuring the current in the present setup (I can’t make sense of this in the PSU case)? If so, I’m not sure how to do that because I would have to place the ammeter somewhere in between, some place that don’t really exist in my current setup. And I don’t have a current clamp obviously.

JFingerle, 2020-01-22:
I got the same error while trying to flash a bios chip (MX25L6406E). The reason was that the bios chip was fried. Check the resistance between the VCC and ground pin of your bios chip. On a working chip it should be a few Ω. On my broken chip I measured 7 mega Ω which is way to high. This causes the voltage drop from 3.3V to 0.8V on the CH341A. The CH341A USB device is not even detected anymore as soon as this happens.

-> Not a flashrom bug. Please **Close**.