corna/me_cleaner #195: Intel ME manufacturing mode?

jjurkus, 2018-04-19:
With the Supermicro X11 and X10 motherboards you have to set a jumper on the motherboard to flash the ME region via the AFU* utilities. This is called the manufacturing mode, sometimes recovery mode.

When this jumper is selected I can flash the ME firmware, but, also in the BIOS the ME firmware says it has version 0.0.0.0, and in windows the device manager shows the Intel Management Engine Interface device as not connected.

What would the implications of this jumper be? Switch to manufacturing mode and forget?

BTW, I previously used the -S flag with these motherboards. If you ever need to enable ME again, it might not be as simple as flashing the BIOS again, because there is of course no FW update module in the ME region.

jjurkus, 2018-04-19:
The X11SSA-F I had an issue with before has two ME jumpers:
JPME1 ME Recovery Pins 1-2 (Normal)
JPME2 Manufacturing Mode Select Pins 1-2 (Normal)

Now I'm not sure what either of those do, and if any of them put the ME in a special mode.
I have replaced this mainboard with a X11SAE, because I had more issues with it, and I'm not going to build it up again.

K1773R, 2018-04-19:
if you just leave the jumper, do you get the 30 min issue?

jjurkus, 2018-04-19:
No, I've turned one on to try, and luckily it stays on. (otherwise I'd have to go back to work)

corna, 2018-04-20:
It's probably the HDA_SDO pin (or something similar for the server platforms), see #135.

If you want a safe way to disable ME you can use this (but of course you're still trusting that ME fulfills your request).

jjurkus, 2018-04-20:
I think I will flash BIOS files that are treated with the -s flag, and set this jumper. I think that will give me the most options.

chewy212, 2021-02-17:
The X11SRM-F exhibits this behavior.

vbooka1, 2022-03-21:
On X11SSH-F with the JPME2 jumper on pins 2-3 BIOS still shows ME and SPS versions, but its "State" is "Recovery". When the JPME2 jumper is on pins 1-2 the "State" is "SPS ME FW Active".

Fun fact: when ME is in "Recovery" state the IPMI could not read the CPU and PCH temperatures (both web panel and `ipmitool` show "n/a"), but simple `sensors` with coretemp module does show CPU temperature.