corna/me_cleaner #196: MSI GL62M-7REX Error 451: The host CPU does not have write access to the target flash area

legends2k, 2018-04-29:
Hi
I've a Core i7-7700hq in a MSI GL62M-7REX laptop.  I got the latest OEM BIOS bin, patched it with `me_cleaner -s`.  I got this

```
Full image detected
The ME/TXE region goes from 0x1000 to 0x200000
Found FPT header at 0x1010
Found 11 partition(s)
Found FTPR header: FTPR partition spans from 0x1000 to 0xa8000
Found FTPR manifest at 0x1448
ME/TXE firmware version 11.8.50.3399
Public key match: Intel ME, firmware versions 11.x.x.x
The HAP bit is NOT SET
Setting the HAP bit in PCHSTRP0 to disable Intel ME...
Checking the FTPR RSA signature... VALID
Done! Good luck!
```

I flashed it with the in-build BIOS UEFI flasher v1.4 (Aptio 5) but after a couple of auto reboot cycles, it entered into Windows and I got

```
$ MEInfoWin64.exe -FWSTS

Intel(R) MEInfo Version: 11.8.50.3460
Copyright(C) 2005 - 2017, Intel Corporation. All rights reserved.

FW Status Register1: 0x90000245
FW Status Register2: 0x0BF60506
FW Status Register3: 0x00000020
FW Status Register4: 0x00084004
FW Status Register5: 0x00000000
FW Status Register6: 0x40000000

  CurrentState:                               Normal
  ManufacturingMode:                          Disabled
  FlashPartition:                             Valid
  OperationalState:                           CM0 with UMA
  InitComplete:                               Complete
  BUPLoadState:                               Success
  ErrorCode:                                  No Error
  ModeOfOperation:                            Normal
  SPI Flash Log:                              Present
  FPF HW Source value:                        FPF HW Not Set
  ME FPF Fusing Patch Status:                 ME FPF Fusing patch NOT supported in this FW Version
  Phase:                                      ROM/Preboot
  ICC:                                        Valid OEM data, ICC programmed
  ME File System Corrupted:                   No
  PhaseStatus:                                AFTER_SRAM_INIT
  FPF and ME Config Status:                   Match
```

I came to know that the BIOS UEFI flasher sometimes avoids flashing the ME region and only updates the BIOS region, so I downloaded `AfuWin64` from AMI.  I tried

```
$ AFUWINx64.EXE "C:\Users\Sundaram\Downloads\E16J9IMS_s.321.bin" /P /B /K /ME
+---------------------------------------------------------------------------+
|                 AMI Firmware Update Utility v5.10.01.1648                 |
|      Copyright (C)2018 American Megatrends Inc. All Rights Reserved.      |
+---------------------------------------------------------------------------+
 Reading flash ............... done
 - ME Data Size checking . ok
 Secure Flash enabled, recalculate ROM size with signature... Enable.
 - FFS checksums ......... ok
 - Check RomLayout ......... Ok.
 Loading capsule to secure memory buffer ... done
 Erasing Boot Block .......... done
 Updating Boot Block ......... done
 Verifying Boot Block ........ done
 Erasing Main Block .......... done
 Updating Main Block ......... done
 Verifying Main Block ........ done
 Erasing NCB Block ........... done
 Updating NCB Block .......... done
 Verifying NCB Block ......... done
 - Error : No Memory Allocated!!
a1 - Error: BIOS does not support ME Entire Firmware update.
```

I read somewhere that using the FPT is more appropriate in such situations, so I tried `FPTW64.exe -D me.bin -ME` and got

    Error 318: The host CPU does not have read access to the target flash area. To enable read access for this operation you must modify the descriptor settings to give host access to this region.

So if it doesn't even let me read, I'm sure it wouldn't let me write to it, so I tried playing with `me_cleaner` and fix the `desc` region:

```

$ python me_cleaner.py -d -s -O desc.bin org_bios.bin
Full image detected
The ME/TXE region goes from 0x1000 to 0x200000
Found FPT header at 0x1010
Found 11 partition(s)
Found FTPR header: FTPR partition spans from 0x1000 to 0xa8000
Found FTPR manifest at 0x1448
ME/TXE firmware version 11.8.50.3399
Public key match: Intel ME, firmware versions 11.x.x.x
The HAP bit is NOT SET
Setting the HAP bit in PCHSTRP0 to disable Intel ME...
Removing ME/TXE R/W access to the other flash regions...
Checking the FTPR RSA signature... VALID
Done! Good luck!
```

Tried writing it back in

```
$ FPTW64.exe -f desc.bin -DESC
Error 451: The host CPU does not have write access to the target flash area.  To enable write access for this operation you must modify the descriptor settings to give host access to this region.
```

Any help would be greatly appreciated.

legends2k, 2018-04-29:
I was able to downgrade my bios to a 6 month older version (E16J9IMS.318).  Then upgraded back to the latest version (E16J9IMS.321) but after patching it with `me_cleaner -s` and the upgrade happened successfully, but still `MEInfoWin64.exe -FWSTS` keeps giving the same output.

I think the ME block is never touched by the BIOS UEFI updater both in its down/upgrade, it only said writing

* Boot block
* NVRAM
* Main Image

legends2k, 2018-04-29:
Used [this resourceful article](http://forum.notebookreview.com/threads/msi-gt73vr-bios-unlock-mod-factory-restore-clear-cmos-and-prevent-common-issues.812372/) and flipped the _Me FW Image Re-Flash_  and _BIOS Lock_ bits.  I'm able to flash new BIOS images but not ME regions.  Trying the patched ROM file still doesn't give a difference since the BIOS UEFI flashing too definitely seems to be oblivious to the ME region. So I continue to get

```
Host Read Access to ME                       Disabled
Host Write Access to ME                      Disabled
```

legends2k, 2018-04-30:
There's a `ME State` in the hidden part of the BIOS which I flipped from 1 to 0 and when I reboot into BIOS, in the _System Information_ page, I see _Intel ME Engine version_ zeroed out!  However, this is short-lived joy since just after on reboot (or just entering into Windows) I see that it's alive again.  Next time I see that it's on and working.

corna, 2018-05-01:
Hi!

The modern SPI flash chips are partitioned, and contain a set of different regions, each with a specific purpose. The first region (called _descriptor_, or _ifd_), always starting at address 0, contains the configuration of all the regions, including the R/W access for each one.

It seems that in your case (which is what Intel recommends) the _descriptor_ region is read-only and the _ME_ region is no-read-or-write, in this way no one (except for the BIOS during the first phases of the boot) can modify the Intel ME firmware or the _descriptor_ region. 

So, through software, your only chance is to "trick" the BIOS to update the ME region, no other software will be able to flash the ME region (or to modify the access permissions, which are saved in the _descriptor_ region, which is read-only).

> Trying the me_cleaner -s patched ROM file still doesn't give a difference since the BIOS UEFI flashing tool definitely seems to be oblivious to the ME region. So I continue to get

Small note: `me_cleaner -s` sets the HAP/AltMeDIsable bit in the _descriptor_ region, not in the _ME_ region. Both are not writable, so the result is the same.

> There's a ME State in the hidden part of the BIOS which I flipped from 1 to 0 and when I reboot into BIOS, in the System Information page, I see Intel ME Engine version zeroed out!

The BIOS can issue a special command to ME to disable it, but it doesn't survive reboots, as you've already discovered.

I'm afraid the easy option here is an hardware programmer.

legends2k, 2018-05-01:
> I'm afraid the easy option here is an hardware programmer.

@corna  Yep, ordered the flasher and the clip! Thanks a lot for _me\_cleaner_. I can see huge amounts of efforts that you've put in. Accept my sincere thanks on this one.

legends2k, 2018-05-23:
No longer have the laptop.

closed 2018-05-23 by legends2k

Zibri, 2019-01-09:
No need. All you have to do is unprotect the flash.
If you need the procedure ask me, but state (and link) you exact bios version.

Zibri, 2019-01-09:
Another note: with the same procedure is also possible to just disable the ME directly from a hidden bios setting.

legends2k, 2019-01-11:
Thanks for your comments @Zibri but as I mentioned earlier, I no longer have the laptop with me, so I can't answer your questions.