corna/me_cleaner #197: x230 and cleaning OEM-firmware

sswamp, 2018-05-10:
I'm sorry that this isn't exactly an issue, more as an question, but I didn't know where else to ask.
I would like to clean my Lenovo X230 OEM-firmware and get rid of ME using SOIC-8 clip and Raspberry.
What is the best (only?) way to power flash chip in this situation?
 I have read that using Wake-On-LAN process is the safest, but this site says, that it doesn't work with ME neutering and you must power the flash chip with Raspberry 3.3v (which may not be safe). https://github.com/mfc/flashing-docs/blob/master/walkthrough%20for%20flashing%20heads%20on%20an%20x230.md
Is there some precautions that I should take to make sure that flash chip is powered safely?
I want to be able to use OPAL/SED SSD encyption, so I don't want to flash coreboot at this moment.

merge, 2018-06-06:
Power is with the RPi's 3.3V. But make sure you specifiy an spi speed for flashrom. `flashrom -p linux_spi:dev=/dev/spidev0.0,spispeed=128` should work.

corna, 2018-06-06:
Peter Stuge is right, powering the motherboard directly from the 3.3V line is not exactly safe, however it's the only method that guarantees that the SPI master is not writing at the same time (thus interfering with our read/write, possibly in a dangerous way). However, most of the me_cleaner/coreboot users flash their chip in this way, and it has been proven quite reliable. Moreover, some manufacturers (like MSI), started to include an SPI connector (with the 3.3V line included) for this exact purpose, which suggest that their motherboards are designed to be powered in such way.

In conclusion, go ahead and follow @merge suggestions.