corna/me_cleaner #200: Can't get status of Intel ME

suoyney, 2018-05-14:
Hey, first of all, thank you for your work, this project is amazing.

I managed to flash my modified .bin (-S, flashed internally) and everything is working great, no slowdowns, no 30 min shutdown etc. But I can't get the status of ME with coreboot's intelmetool as documented on here: https://github.com/corna/me_cleaner/wiki/Get-the-status-of-Intel-ME

I run sudo ./intelmetool -m and the output result is only:
"Can't find ME PCI device"

I have to mention that this is the **same output from before I flashed the modified BIOS**, I'm not sure if I have to set something else to get this info... :(
I'm already running iomem=relaxed as a parameter on the kernel...

Additional info: my MB chipset is H270, my CPU is an i5 7400, entire PC specs here: https://support.hp.com/us-en/document/c05384163

Any help would be greatly appreciated, thank you!!

corna, 2018-05-24:
Unfortunately `intelmetool` doesn't support the newer Intel architectures. You can either try to add support for your system (it should be quite easy), or you can use the proprietary Intel tools (MEInfo) to probe the ME status.

suoyney, 2018-05-24:
Thanks for the confirmation! I thought it was something of the sorts, I found a thread in here of somebody manually adding their Sunrise Point into `intelmetool` a while ago, I was into doing the same for my Union Point but got sidetracked, knowing I was in the right path makes me happy.

I will get back into it now, my coding skills are rusty but I do like the challenge :smiley:

Thank you!!