corna/me_cleaner #203: No me_cleaner nessecary: ASRock Z370 Taichi Firmware with Intel Management Engine Disabled

rugabunda, 2018-05-22:
https://www.techpowerup.com/forums/threads/asrock-z370-taichi-firmware-with-intel-management-engine-disabled.243939/

corna, 2018-05-24:
I don't understand, what's the issue? They've simply applied me_cleaner on the board's firmware and shared the modified (and tested) image.

rugabunda, 2018-06-06:
It is useful information; Many Intel_Me cleaning sources claim it necessary to use a hardware device to hard flash a motherboard in order to remove Intel ME. This proves otherwise; Many people including myself would much rather buy a board where all that is necessary is to download a pre-modified bios, instead of hard flashing or jumping through the same hoops to remove intel_me. If its already done its already done, and saves people a lot of effort.

rugabunda, 2020-01-06:
new threat is here: https://www.techpowerup.com/forums/threads/asrock-z370-z390-taichi-and-some-others-actively-modding-firmware-with-intel-management-engine-disabled-new-method.259319/page-2