corna/me_cleaner #204: Cannot find BIOS chip / model number

vdbroek, 2018-05-27:
Having the 30 minutes problem with my Lenovo Flex 2 Pro.
I want to try to disable / reflash the intel me chip but i cannot find the BIOS chip or model number.

Can someone help me with figure out the vendor and chip model number is?

Details notebook:
Lenovo Flex 2 Pro (80FL0005NX)
Intel Core i7-4510U

Mainboard:
[image]

The maybe chip:
[image]

kitsunyan, 2018-05-27:
> Can someone help me with figure out the vendor and chip model number?

It's written on the chip.  The letters are small so it may be hard to read it.

jjurkus, 2018-06-21:
Well, if you didn't meddle with ME before this will probably not be an issue that can be solved by using me_cleaner.