corna/me_cleaner #205: External flash on a PLCC-32 ?

TheSanchezRick, 2018-05-29:
Hi everybody :)

I was wondering if it was possible to process the external flash on a computer with a PLCC-32 bios chip (I've only found tutorial for SOIC 8 model) and if so, how to do it ?
[image]
The chip seems to be soldered on the motherboard, which will probably complicate things.

Thank you in advance
Have a nice day !

skochinsky, 2018-05-29:
what chipset is it? PLCC is unlikely to have ME since it generally only supports SPI flash.

TheSanchezRick, 2018-05-30:
Really ? :o This computer has a Intel Celeron D inside, which i assume has a pre-skylake ME version (stored on the external chip) or maybe i'm wrong ?

Thanks for your answer !

corna, 2018-05-30:
We need more informations:

 * Remove the BIOS label and read the chip's model
 * Run `lspci`

platomav, 2018-05-30:
There is no Engine on a 2004-2005 platform.

corna, 2018-05-30:
> There is no Engine on a 2004-2005 platform.

I missed the "Intel Celeron D" part ;)

closed 2018-05-30 by corna