corna/me_cleaner #207: BIOS has wrong data from CPU sensor

RedSkyAround, 2018-06-10:
Thanks for this project, that's actial and useful.

I have Nehalem CPU with Lynnfield core i5-760 and motherboard Foxconn H55MXV. I using last OEM BIOS.
Looks like me_cleaner work, PCI simple communication device disappeared from device manager. But seems something necessary was also removed from BIOS.
After application me_cleaner i got incorrect CPU Temperature in BIOS. Now CPU Temperature in BIOS is stuck and not changing - now it's constant. CPU cooler does not want change PRM depending on real CPU temperature. But with stock BIOS it's work properly. Would be something exists here, what i can do to resolve it?

Logs:

me_cleaner.py -S -O Modified_1.bin 1.bin
Full image detected
The ME/TXE region goes from 0x1000 to 0x700000
Found FPT header at 0x1010
Found 7 partition(s)
Found FTPR header: FTPR partition spans from 0x32000 to 0xd2000
ME/TXE firmware version 6.2.0.1022
Public key match: Intel ME, firmware versions 6.x.x.x
The AltMeDisable bit is NOT SET
Reading partitions list...
 FOVD (0x00000400 - 0x000002000, 0x00001c00 total bytes): removed
 EFFS (0x00002000 - 0x000032000, 0x00030000 total bytes): removed
 NVKR (NVRAM partition, no data, 0x00010000 total bytes): nothing to remove
 NVTD (NVRAM partition, no data, 0x00002000 total bytes): nothing to remove
 NVVE (NVRAM partition, no data, 0x00000c00 total bytes): nothing to remove
 FTPR (0x00032000 - 0x0000d2000, 0x000a0000 total bytes): NOT removed
 NFTP (0x000d2000 - 0x0006ff000, 0x0062d000 total bytes): removed
Removing partition entries in FPT...
Removing EFFS presence flag...
Correcting checksum (0xe7)...
Reading FTPR modules list...
 BUP              (Huffman, fragmented data, ~31 KiB  ): NOT removed, essential
 VE_FW_NAND       (Huffman, fragmented data, ~155 KiB ): removed
 KERNEL           (Huffman, fragmented data, ~100 KiB ): removed
 POLICY           (Huffman, fragmented data, ~91 KiB  ): removed
 VENOM            (LZMA   , 0x091c25 - 0x093a85       ): removed
 HOSTCOMM         (LZMA   , 0x093a85 - 0x098644       ): removed
 RSA              (LZMA   , 0x098644 - 0x09e5da       ): removed
 CLS              (LZMA   , 0x09e5da - 0x0a2f5a       ): removed
 TDT              (LZMA   , 0x0a2f5a - 0x0a6136       ): removed
The ME minimum size should be 270336 bytes (0x42000 bytes)
The ME region can be reduced up to:
 00001000:00042fff me
Setting the AltMeDisable bit in PCHSTRP10 to disable Intel ME...
Checking the FTPR RSA signature... VALID
Done! Good luck!

me_cleaner.py -c Modified_1.bin
Full image detected
The ME/TXE region goes from 0x1000 to 0x700000
Found FPT header at 0x1010
Found 1 partition(s)
Found FTPR header: FTPR partition spans from 0x32000 to 0xd2000
ME/TXE firmware version 6.2.0.1022
Public key match: Intel ME, firmware versions 6.x.x.x
The AltMeDisable bit is SET
Checking the FTPR RSA signature... VALID

RedSkyAround, 2018-06-13:
After application me_cleaner, CPU temperature in BIOS depend on the installed video card :)

icceland, 2018-06-19:
Maybe you can try with **-s** options, which should be more "soft" solution?

forkoz, 2018-06-27:
I have corei5 ivybridge and -S option killed the cpu temperature and fan controls. Also, lan would stay on after the computer shut down. After trying -s instead I can't enter CTRL-P menu anymore and lan light shuts off with the computer. So I think me_cleaner has to be more judicious with what modules are removed. Fan running at full speed is kind of a show stopper.

skochinsky, 2018-06-27:
depending on the board/platform, ME may handle clocking, fans, and/or sensors, so removing too many modules may lead to such situations. you could try playing with the whitelist option and try to leave more modules, or just leave the firmware alone if you don't want problems.

forkoz, 2018-06-28:
Does me cleaner unprotect the ME region in the FPT? Because to play with stuff like this I have to desolder the chip every time. This is only the 2nd system I ended up with that had ME, the first one I did manually and I could turn ME on/off since I was the one who edited the FPT.

corna, 2018-06-28:
No, but you can unlock/lock the regions with `ifdtool -u`/`ifdtool -l` (from the [coreboot project](https://github.com/coreboot/coreboot/tree/master/util/ifdtool)).

RedSkyAround, 2018-06-28:
> Maybe you can try with -s option, which should be more "soft" solution?
That's bad news for me. With -s option BIOS has same problem with data from CPU sensor. I not sure on 100%, but i think that this is the result of a violation of some checksum.

forkoz, 2018-06-28:
If you write back that full backup you took does it work?

RedSkyAround, 2018-07-04:
> If you write back that full backup you took does it work?
Yes. It work also with all BIOS files from Foxconn website.

forkoz, 2018-07-05:
What if you flash bios with ME disable bit only and then flash the smaller bios that likely doesn't have ME fw update in it. Like the top ones from here: http://www.foxconnchannel.com/ProductDetail.aspx?T=motherboard&U=en-us0000488

They were also kind enough to provide a layout https://pastebin.com/SYixCSvU

RedSkyAround, 2018-07-08:
> What if you flash bios with ME disable bit only and then flash the smaller bios that likely doesn't have ME fw update in it.

Thanks for care. But that has same result.
I already saw the map of BIOS file in zip archive from manufacturer website. I compared BIOS Region (00700000 - 007FFFFF hex) of BIOS with ME disable bit only (-s) with default BIOS Region (00700000 - 007FFFFF hex) from website. Checksumms are match. If it compare with BIOS with ME disable bit and remove modules (-S), checksumms are not match.
This motherboard has the jumper for disable ME, but it not work. It only disabling display version info in BIOS menu, but after load windows, PCI simple communication device NOT (!) disappears from device manager.