corna/me_cleaner #208: [Success] ASRock Fatal1ty Z370 Gaming-ITX/ac

Grimeton, 2018-06-13:
Hello,

I just used me_cleaner with the latest BIOS image (1.90) on an Fatal1ty Z370 Gaming-ITX/ac from ASRock and it works just fine. ME is now reported as version 0.0.0.0 in the Bios.

Cu

$ ./me_cleaner.py Z37GIX_1.90
Full image detected
The ME/TXE region goes from 0x3000 to 0x200000
Found FPT header at 0x3010
Found 11 partition(s)
Found FTPR header: FTPR partition spans from 0x1000 to 0xa8000
Found FTPR manifest at 0x1448
ME/TXE firmware version 11.8.50.3425
Public key match: Intel ME, firmware versions 11.x.x.x
The HAP bit is NOT SET
Reading partitions list...
 FTPR (0x00001000 - 0x0000a8000, 0x000a7000 total bytes): NOT removed
 FTUP (0x00110000 - 0x0001bc000, 0x000ac000 total bytes): removed
 DLMP (      no data here      , 0x00000000 total bytes): nothing to remove
 PSVN (0x00000e00 - 0x000001000, 0x00000200 total bytes): removed
 IVBP (0x0010c000 - 0x000110000, 0x00004000 total bytes): removed
 MFS  (0x000a8000 - 0x00010c000, 0x00064000 total bytes): removed
 NFTP (0x00110000 - 0x0001bc000, 0x000ac000 total bytes): removed
 ROMB (      no data here      , 0x00000000 total bytes): nothing to remove
 FLOG (0x001bc000 - 0x0001bd000, 0x00001000 total bytes): removed
 UTOK (0x001bd000 - 0x0001bf000, 0x00002000 total bytes): removed
 ISHC (      no data here      , 0x00000000 total bytes): nothing to remove
Removing partition entries in FPT...
Removing EFFS presence flag...
Correcting checksum (0xf2)...
Reading FTPR modules list...
 FTPR.man     (uncompressed, 0x001448 - 0x002018): NOT removed, partition manif.
 rbe.met      (uncompressed, 0x002018 - 0x0020ae): NOT removed, module metadata
 kernel.met   (uncompressed, 0x0020ae - 0x00213c): NOT removed, module metadata
 syslib.met   (uncompressed, 0x00213c - 0x0021a0): NOT removed, module metadata
 bup.met      (uncompressed, 0x0021a0 - 0x00274a): NOT removed, module metadata
 pm.met       (uncompressed, 0x00274a - 0x0027f8): NOT removed, module metadata
 vfs.met      (uncompressed, 0x0027f8 - 0x003158): NOT removed, module metadata
 evtdisp.met  (uncompressed, 0x003158 - 0x0032e6): NOT removed, module metadata
 loadmgr.met  (uncompressed, 0x0032e6 - 0x00340e): NOT removed, module metadata
 busdrv.met   (uncompressed, 0x00340e - 0x003792): NOT removed, module metadata
 gpio.met     (uncompressed, 0x003792 - 0x0038dc): NOT removed, module metadata
 prtc.met     (uncompressed, 0x0038dc - 0x003a8c): NOT removed, module metadata
 policy.met   (uncompressed, 0x003a8c - 0x003c4e): NOT removed, module metadata
 crypto.met   (uncompressed, 0x003c4e - 0x003dd8): NOT removed, module metadata
 heci.met     (uncompressed, 0x003dd8 - 0x003fa4): NOT removed, module metadata
 storage.met  (uncompressed, 0x003fa4 - 0x0042a0): NOT removed, module metadata
 pmdrv.met    (uncompressed, 0x0042a0 - 0x0043c4): NOT removed, module metadata
 maestro.met  (uncompressed, 0x0043c4 - 0x0044ae): NOT removed, module metadata
 fpf.met      (uncompressed, 0x0044ae - 0x0045ba): NOT removed, module metadata
 hci.met      (uncompressed, 0x0045ba - 0x0046bc): NOT removed, module metadata
 fwupdate.met (uncompressed, 0x0046bc - 0x0047c4): NOT removed, module metadata
 ptt.met      (uncompressed, 0x0047c4 - 0x0048d0): NOT removed, module metadata
 touch_fw.met (uncompressed, 0x0048d0 - 0x004a00): NOT removed, module metadata
 rbe          (Huffman     , 0x004a00 - 0x007100): NOT removed, essential
 kernel       (Huffman     , 0x007100 - 0x0170c0): NOT removed, essential
 syslib       (Huffman     , 0x0170c0 - 0x028f80): NOT removed, essential
 bup          (Huffman     , 0x028f80 - 0x053140): NOT removed, essential
 pm           (LZMA/uncomp., 0x053140 - 0x055740): removed
 vfs          (LZMA/uncomp., 0x055740 - 0x05d6c0): removed
 evtdisp      (LZMA/uncomp., 0x05d6c0 - 0x05f0c0): removed
 loadmgr      (LZMA/uncomp., 0x05f0c0 - 0x061ec0): removed
 busdrv       (LZMA/uncomp., 0x061ec0 - 0x063780): removed
 gpio         (LZMA/uncomp., 0x063780 - 0x0648c0): removed
 prtc         (LZMA/uncomp., 0x0648c0 - 0x065480): removed
 policy       (LZMA/uncomp., 0x065480 - 0x06a040): removed
 crypto       (LZMA/uncomp., 0x06a040 - 0x077c40): removed
 heci         (LZMA/uncomp., 0x077c40 - 0x07bb40): removed
 storage      (LZMA/uncomp., 0x07bb40 - 0x080180): removed
 pmdrv        (LZMA/uncomp., 0x080180 - 0x081340): removed
 maestro      (LZMA/uncomp., 0x081340 - 0x083100): removed
 fpf          (LZMA/uncomp., 0x083100 - 0x084b00): removed
 hci          (LZMA/uncomp., 0x084b00 - 0x0853c0): removed
 fwupdate     (LZMA/uncomp., 0x0853c0 - 0x08a100): removed
 ptt          (LZMA/uncomp., 0x08a100 - 0x09f7c0): removed
 touch_fw     (LZMA/uncomp., 0x09f7c0 - 0x0a8000): removed
The ME minimum size should be 360448 bytes (0x58000 bytes)
The ME region can be reduced up to:
 00003000:0005afff me
Checking the FTPR RSA signature... VALID
Done! Good luck!

RedSkyAround, 2018-06-13:
Reports about success works [url=https://github.com/corna/me_cleaner/issues/3]here[/url].

Grimeton, 2018-06-13:
So, turns out. Disabling the ME, kills the network card. The Intel i219-V is there and visible but the drivers cannot be loaded. I was only able to work around this by flashing the main BIOS including the ME.

Grimeton, 2018-06-13:
"-s" worked.

corna, 2018-06-13:
Try with -S

Grimeton, 2018-06-13:
Actually I'm not a 100% convinced....

I had the problem and installed the BIOS including the ME and it worked, afterwards I installed the "-s" version and it still works. I wonder if the ME-enabled BIOS made the driver install happen and now it's not a problem anymore as were a fresh install would create the same problem again :-/

Will test and report.

Cu

Futureknows, 2018-07-29:
I used ME_cleaner on the latest BIOS  (v3.10) for a 370m ITX/AC.
All of the steps works as they should but on reboot BIOS still shows ME version 11.8.50.3470.

`mei-amt-check` returns 
`Error: Management Engine refused connection. This probably means you don't have AMT`

`intelmetool'  -m r`eturns
`Can't find ME PCI device`

I download the Intel me detection tool intel_sa00086 which returns the following data:

```
INTEL-SA-00086 Detection Tool
Copyright(C) 2017-2018, Intel Corporation, All rights reserved.

Application Version: 1.1.169.0
Scan date: 2018-07-29 06:03:53 GMT

*** Host Computer Information ***
Name: localhost.localdomain
Manufacturer: To Be Filled By O.E.M.
Model: To Be Filled By O.E.M.
Processor Name: Intel(R) Core(TM) i7-8700K CPU @ 3.70GHz
OS Version: Fedora 28 Twenty Eight (4.17.9-200.fc28.x86_64)

*** Intel(R) ME Information ***
Engine: Intel(R) Management Engine
Version: 11.8.50.3470
SVN: 3

*** Risk Assessment ***
Based on the analysis performed by this tool: This system is not vulnerable. It has already been patched.
```

It seems like the newer firmwares prevent the routine detection of ME. 
I might try the v1.9 BIOS you had success with unless anyone has anyone insight here.

xinterix, 2018-08-29:
i failed with an asrock z370 myself,it was a dual bios,  and i have no idea where things went wrong but somehow after flashing to the 1st bios chip, the one used by default, and restarting, it immediately entered into restore mode and begin copying data from the other chip, which should have contained a good unaltered bios, but after restore completes, and restarts pc, it immediately does the same thing again, over and over. Now interestingly, i had never touched the other chip since it was remove box, never even touched the jumper either. So I move a jumper to the header that activate the other chip and it do the same thing, boot up copying stuff from the first chip...basically endless restore loop

Futureknows, 2018-08-29:
Mine wasn't bricked it works fine, I'm just not certain it cleaned the ME. After reading more I think it did, but my BIOS always reports the ME number no 0.0.0.0. 

Is this the same for everyone else?