corna/me_cleaner #209: Does me_cleaner disables the 30 minutes watchdog?

icceland, 2018-06-17:
Hi All,
I had already an me_cleaner  action withput result. The whole operation is a case for other topic, which will probably come. The fact is I had now i3-3120m CPU left and a Sony Vaio - **Intel hm70** based notebook.
Suprisingly the processor works fine with the "unsupported" chipset. Although with 30 minutes shutdown. 

My question is, if **me_cleaner** is able to disable this shutdown or it just stops the ambiguous ME functionality? In other words, is it worth to dettaching the BIOS chip from the mainboard. It's fragile action and I don't want to risk in hopeles case.

If it's worth, I'llshare my experiences.

Kind regards,
Gregor

platomav, 2018-06-17:
HM70 worked only with Pentium and Celeron and the restriction was hardcoded in the PCH so afaik no, mc_cleaner won't help you.

icceland, 2018-06-18:
But after inserting of the CPU 30 minutes works everything perfectly. This reminds me the ME watchdog.

platomav, 2018-06-18:
It was an artificial limitation to separate the entry level HM70 systems from the mainstream market. Bricking the ME firmware won't help.

icceland, 2018-06-19:
O.K. So the issue could be clam for closed. Thanks platomav!

closed 2018-06-20 by corna

qnxdev, 2020-12-07:
@icceland I have a SONY VAIO with Intel® HM76 Express Chipset

I haven't tried it yet but I wish to.