corna/me_cleaner #210: Medion Akoya p6640 shutdown after 30 inutes

icceland, 2018-06-18:
I recently became problems with notebook Medion Akoya p6640. It started to shut down after 30 minutes. I found out, the reason could be some ME issue, so I decided to disable it using **me_cleaner**.

The result was dicouraging. Me_cleaner ends succesfull, but there is no change after applying **-S** option.
The **intelmetool** yields:
`[[31mBad news, you have a `HM76 Express Chipset LPC Controller` so you have ME hardware on board and you can't control or disable it, c

^[[0mMEI found: [8086:1e3a] 7 Series/C216 Chipset Family MEI Controller #1

ME Status   : 0x3006
ME Status 2 : 0x80040000

ME: FW Partition Table      : OK
ME: Bringup Loader Failure  : NO
ME: Firmware Init Complete  : NO
ME: Manufacturing Mode      : NO
ME: Boot Options Present    : NO
ME: Update In Progress      : NO
ME: Current Working State   : Platform Disable Wait
ME: Current Operation State : Preboot
ME: Current Operation Mode  : Normal
ME: Error Code              : Image Failure
ME: Progress Phase          : Clean Moff->Mx wake
ME: Power Management Event  : Clean Moff->Mx wake
ME: Progress Phase State    : Unknown 0x04

ME: Extend Register not valid

ME: has a broken implementation on your board withthis firmware
ME: failed to become ready
ME: failed to become ready
ME: GET FW VERSION message failed
^[[31mBad news, you have a `HM76 Express Chipset LPC Controller` so you have ME hardware on board and you can't control or disable it, c

^[[0mDid you run 'modprobe msr'?
ME Capability: BootGuard Mode                             : ^[[36mUnknown
^[[0m`

With CPU (i3-3120m - Ivy Bridge) is everything right - was tested on other device.
Any Idea? Suggestion?  Ray of hope?

corna, 2018-06-20:
>  ME: Current Working State : Platform Disable Wait

Yep, that's definitely ME, however it seems that this is not related to _me\_cleaner_.

If _me\_cleaner_ didn't complain about the signature of the FTPR partition, it might be related to Intel Anti-Theft, @platomav is probably more skilled than me in this field.

platomav, 2018-06-20:
Nothing to do with me_cleaner. Follow the [Engine CleanUp Guide](https://www.win-raid.com/t1658f39-Guide-Clean-Dumped-Intel-Engine-CS-ME-CS-TXE-Regions-with-Data-Initialization.html#msg24423).

icceland, 2018-06-21:
Thank you **platomav**. I have the feeling, your answer clears a lot. But it sounds very complicated. I'll contact some possible sources for full BIOS backup of my notebook before I'll try your soulution. As soon there's some progress, I'll inform about my results.

icceland, 2018-06-22:
I overwrited the ME region, like it is described in the link from you and the notebook works already few hours. The problem seems to be solved. Because I'm highly happy to can use the Laptop again, I decided not to apply **me_cleaner**. I spend already enough time to rescoue it.
You have my big respect for your knowledge and huge Thank You!

Is there some way to support your work? You deserved a free beer at least for excellent job you do.

platomav, 2018-06-22:
The CleanUp Guide is there to re-configure, re-flash & repair the Engine firmware. It covers 99.99% of cases (including your own). If ones wants to brick the Engine firmware because they are afraid/skeptical of it then me_cleaner is their tool.

Thank you for your offer, you can find the relevant link at my ME Analyzer repository, but I shall not discuss that further here in order to respect Nicola's repo. The issue can now be closed.

dammysholove, 2019-01-10:
> 
> 
> Thank you **platomav**. I have the feeling, your answer clears a lot. But it sounds very complicated. I'll contact some possible sources for full BIOS backup of my notebook before I'll try your solution. As soon there's some progress, I'll inform about my results.

please do you have the bios dump?? i need it pls  Medion P6640 48.4UY05.011

icceland, 2019-01-26:
https://drive.google.com/open?id=1sL1kBcjcuwkKAd8pi82Duhr1wBXTq_Q5

That's latest BIOS file I created during the solving of problem. Unfortunately I'm not if it's already solved the problem. It's already a bunch of time ago. You should be aware. You can update it only by unsoldering the chip and reprogramming it externally, because official tools allow only access the second 4MB of BIOS. I did it on Orange Pi Zero with Armbian on, but Orange/Banana Pi should be as good. The software I used is described on this forum. The good thing is. It still works like a dream!

dammysholove, 2019-01-26:
Thanks so much sir, i have many programmers that i can use,,, i will get back to you if it work.. am grateful