corna/me_cleaner #211: Unrelated - Information On GitHub Website - VERY IMPORTANT.

GarmentXEX, 2018-06-19:
Hi guys, this very important but almost unrelated to the "Issues" in the conventional sense of the word.

Github is now purchased by Microsoft - enough said. Full acquisition is by the end of this year. 

For those of you who do not know, Microsoft is that notorious company who created your original spy network OS called Windows 10.

Obviously the intent is quite apparent, with a strong strangle hold of repositories, repo like me_cleaner may or may not "disappear" from this site. 

I got this from a secondary source, he's quite a trustworthy guy judging from the many videos I've seen so far:

https://www.youtube.com/watch?v=7c4ZMlXRQF0

corna, 2018-06-23:
I personally don't have any particular reason to migrate this repo to another service at the moment. Yes, Microsoft is acquiring GitHub, but I don't think that it doesn't impact any repository of this kind.

Let's suppose for a moment that Microsoft and Intel have an agreement (quite unlikely, Microsoft probably has no interest on this, while Intel seems to not care enough) and they want to:

 * take down this repo: the code is mirrored (at each major change) to [coreboot's self-hosted repo](https://review.coreboot.org/cgit/coreboot.git/tree/util/me_cleaner), moreover I have multiple clones of this repository on my personal PCs so I could easily push it to another service. In this case Microsoft earns nothing but a bad reputation.
 * modify the code and insert something malicious: thanks to `git`, at my first push I would notice that (plus, me_cleaner is a 900 LoC Python script, it's pretty hard to hide something there). Moreover, I started signing the tags (pubkey [here](https://pgp.mit.edu/pks/lookup?search=nicola%40corna.info&op=index) and [here](https://keybase.io/corna)), so you can easily verify them. Same result as above.
 * modify the wiki: also the wiki is managed through `git`

Moreover many projects depends on this repository (like [this one](https://source.puri.sm/coreboot/coreboot-files/blob/master/build_coreboot.sh#L654)), so moving this repo to another service would only create troubles for them.

I could just mirror this repo to another service but, at the moment, I don't see any particular reason to duplicate it. The only valid point is the non-free nature of the GitHub code, but:

 * this repo can be cloned directly through `git`
 * I can (and do) receive issues and questions through [email](mailto:nicola@corna.info) (PGP-encrypted if needed, key above) if there are valid reasons

This is all, unless I missed something, in that case feel free to correct me.

corna, 2018-06-28:
Feel free to reopen if you have objections, I'm available to discuss this.

closed 2018-06-28 by corna