corna/me_cleaner #213: Support for 12.x?

tinylabs, 2018-06-28:
Wondering if support will be added for 12.x?  Doesn’t seem to recognize the bios dump at all. Happy to share my bios binary.

Intel Core i7-8750H
ME version: 12.0.0.1069
Clevo P95X/ER

platomav, 2018-06-28:
Integrated Firmware Image (IFWI) support is needed for CSME 12+ and CSTXE 3+

krystian-wojtas, 2018-10-09:
Hi platomav, could you please explain what does it mean? What is needed to reduce Menagment Engine version 12?

maenpaa24, 2019-02-14:
Hi!
I am not sure about which processors have intel ME 12. I thought it was something like:
1st generation => 6
2nd=>7
3rd=>8
4th=>9
5th=>10
6th=>11
7th=>12
... and so on

However in the issue #3 people have reported success even with a core i9-9900k (https://github.com/corna/me_cleaner/issues/3#issuecomment-456526845). So I wonder how I could find out which version of intel ME has the processor I am planing to buy? Because as far as I know intel does not offer such info or I couldn't find it.

Thanks.

Yannik, 2020-12-27:
PR #282 adds support for setting the HAP bit on ME 12 and newer.