corna/me_cleaner #214: New BIOS option: "ME Operation Mode" (Temporary Disabled setting)

ruany, 2018-06-29:
Board: ASUS PRIME Z270-P (Kaby Lake)
BIOS: 1205 (2018/05/25)

After updating BIOS I've found a new option under PCH-FW Configuration.

[image]

Is this effectively the same as `Alt Disable Mode` (HAP/AltMeDisable)?
My system does not shut down after 30 minutes while using this option.

Intel ME PCI device is missing:
```
~ » lspci|grep ME
~ » 
```

```
~/workspace/coreboot/util/intelmetool » sudo ./intelmetool -m
Can't find ME PCI device
```

I've searched around on "ME Operation Mode" and got no useful results at all.

[BIOS changelog](https://www.asus.com/us/Motherboards/PRIME-Z270-P/HelpDesk_BIOS/) states "Intel New ME Update, Improve Stability."

ruany, 2018-06-30:
Interesting. It feels like I'm the first person to notice this option, and I haven't seen any mention of such a BIOS option in the wikis or elsewhere, only a hardware switch (#195, #135)

The similarity with the jumper disable method is that it apparently allows for the ME to be flashed using FPT. And since this option can be toggled from BIOS, theoretically ME can be fully neutered through internal flashing without ever touching hardware or flashing motherboard firmware (correct me if I am wrong).

I've noticed no side effects whatsoever from disabling ME using this BIOS option. Network is fine, all hardware works as expected, fully stable.

bol-van, 2018-06-30:
You can do a test.
Do not enable "temporary disable" mode in bios.
Run "fpt -disableme"
or same thing in linux by : https://github.com/bol-van/me-disable
and test if ME region becomes writable

ValoWaking, 2018-09-20:
"ME will go back to normal mode once power off..."

OneB1t, 2019-01-10:
interesting is that for example my motherboard does not have that option O.o
Prime Z270-A with bios 1302

0100001001000010, 2020-12-27:
Not sure if it's meaningful, but my new Ice Lake laptop (Acer Aspire 5 A515-55-56VK, InsydeH2O rev 5) has that option too. On mine, it has the siide effect of making suspend/resume work exactly once after a reboot, and then the system will freeze after Linux tries to suspend the second time. (After a reboot, the kernel's last words in the log show no problems and the system having successfully prepared and about to enter the low power state.) It cannot be woken up except by holding the power button, the keyboard backlight (which is not exposed to the OS so is presumably controlled by the EC) turns off and won't come back on, and the display is turned off, but the fan still runs and the power light remains steady blue. (Sleep = flashing orange.) systemctl poweroff shuts down the OS but does not power down the computer if I've used up the first suspend. That's the only side effect I've noticed, though it's quite a bad one so I had to re-enable it.

the-j0k3r, 2021-01-30:
@ruany 

that was made so that when you update your Bios it wont send the ME into recovery mode or display messages like 

`A7 ME FW Downgrade Request MESpilock failed`

Machine will work fine without shutdowns (as Ive observed) or other side effects but it will take your startup time to 70seconds or 90seconds depending on settings.

But sadly not all z270's have that option and none of the methods of flashing BIOS Asus provides for bios recovery etc will fix this.

So instead they added that option, so that bios update or ME updates dont cause this issue, and its widespread.

Only way to fix then is by HW programmers to bypass FD software locks etc.

the-j0k3r, 2021-01-30:
@ruany and apologies to everyone for this (late) reply, its a topic of some interest for myself.

That was made so that when you update your Bios it wont send the ME into recovery mode that will cause messages like the following to display 

`A7 ME FW Downgrade Request MESpilock failed`

Machine will work fine without shutdowns (as Ive observed) or other side effects but it will take your startup time to 70seconds or 90seconds depending on Bios settings. (as opposed to 10 seconds or there abouts)

But sadly not all z270's have that option and none of the methods of flashing BIOS Asus provides for bios recovery etc will fix this and more often than not after flashing Bios or ME will cause this on motherboards that dont have a service mode of sorts except by shorting pins on audio chip or HW programmers.. At least Im not aware of any other solution to this issue.

So instead they added that option, so that bios update or ME updates dont cause this issue, and its a more widespread issue than Asus admits and also provides no support for because Asus are the best at customer support (not).

Only way to fix then is by HW programmers to bypass FD software locks etc or shorting pins on audio chip.

As a side note, when doing repairs I would rather disable the ME via HAP mode or me_cleaner rather than use the ME images floating around the web, cause most wont be able to discern if the ME bins are clean or contain some payload, and not everyone has skills to view the contents in hex editor and know whats going on.

Personally I dont understand why the Updated bios version dont contain an updated ME region and instead all will be older version than the manufacturer provides, this may also cause issues upon flashing, I may be wrong here but Im not an expert on ME or Bioses in any case.

Asus will not provide any more official ME updates for z270's in any case, only 1 update was done and no matter how many CVE's ME has, your only choice as a regular user is to use ME bins that may or may not do more harm than good.

ruany, 2021-01-30:
@the-j0k3r 
Startup time is unaffected on my side, with or without ME Temporary Disabled it takes under 10 seconds to boot (from POST to desktop)

No issues with suspend/resume either. After all these years I've still noticed no negative effects from this option, besides the mild inconvenience of using it.

0100001001000010, 2021-01-30:
> That was added so that when you update your Bios it wont send the ME into recovery mode because its Tuesday, Such issues include messages like the following to display after P.O.S.T.

Wait, is the Tuesday thing literal or figurative? I would hope it's figurative, but knowing how shitty OEM firmware tends to be, I wouldn't be surprised by bugs that occur only on Tuesday in things that shouldn't be using the rtc at all.

If the option is meant to be a workaround to an issue triggered when updating, why doesn't the firmware just turn it set to temporary disabled automatically before update and back on after update, not exposing it to the end user? That sounds like it would be better for UX than consumers potentially bricking their mobo because they didn't change a setting before updating because they didn't know the update routine didn't actually do the entire process for them. Am I missing something (I'm not that familiar with firmware) or is this just an instance of OEM firmware crappiness?

the-j0k3r, 2021-01-30:
> @the-j0k3r
> Startup time is unaffected on my side, with or without ME Temporary Disabled it takes under 10 seconds to boot (from POST to desktop)

I never said it was, I explained why the feature you described is implemented to prevent such issues.

toiletflusher, 2022-06-13:
It seems my BIOS (Acer Aspire 3 A315-53-34FE) when advanced options are enabled has this option too. When disabled, it says it is temporarily disabled however it keeps saying ME Failed when going into the ME settings. Only side effects i noticed is that the bootup became very slow (probably because it is expecting ME to start). 

[image]

the-j0k3r, 2022-06-16:
@toiletflusher That is to be expected, since ME is disabled you cant really get into the ME settings, you should be able to use tools lime MeManuf to verify state and other important ME status information.

You cant really use intelmetool for that. MeManuf will likely work.

Also boot time is impacted on some systems when ME is disabled or in recovery mode, not all in my experience..

VorpalQ, 2023-01-26:
[image]
Interestingly I seem to have this feature on my LG Gram 2021 16" laptop as well, in the hidden advanced BIOS settings (you have to do some shenanigans with ctrl+alt+shift+f7 to open it, and it has TONS of options for everything from power management to PCI features, etc). I was having trouble finding out anything about it on Google. Would it be safe to disable the ME this way, do you think?

bruceblore, 2023-01-26:
Interesting. I have that giant menu with that insane amount of options on my Acer. On mine, you reveal it by pressing Fn+Tab+Power to boot the machine.

In one of my earlier replies, I mention that trying it seems to have no side effects, other than only allowing the laptop to enter a low power state once per boot:

> Not sure if it's meaningful, but my new Ice Lake laptop (Acer Aspire 5 A515-55-56VK, InsydeH2O rev 5) has that option too. On mine, it has the siide effect of making suspend/resume work exactly once [...]

So *probably* fine to try it, but make sure to fully test for broken functionality. (And, if you can, that the ME is actually disabled.)