corna/me_cleaner #215: Finally a way to firmware-kill ME discovered by some accident!

WaseemAlkurdi, 2018-07-08:
Dear @coma and all interested,
I have been tinkering on my laptop, an HP EliteBook Revolve 810 G2 (i5-4300u w/ vPro), with BIOS version L86 01.40.
I keep a macOS / Linux dual-boot on it.
Yesterday, after rebooting from macOS, a strange red error message "ME State Recovery Mode" appeared in red letters on the top-left corner of the screen. The system boots normally though.
In the ESC startup menu, the options to enter Intel ME setup (F5) and Initiate Intel CIRA (F6) vanished. However, the Intel ME options in the EFI firmware (BIOS) setup menu remained present (but have no effect).
I can't enter the MEBx panel.
Doing `# ./mei-amt-check` results in:
```
Error: Management Engine refused connection. This probably means you don't have AMT
```
instead of the usual output.
Also, checking my ```/sys/firmware/efi/efivars``` under Linux, I found this new variable:
```
VPROState-a82471a8-e491-42c1-9240-4fde3073c0fd
```
Attempting to 'cat' it results in a weird character, and opening it w/ ```nano``` shows a ^A.

This problem is in itself known to HP and the solution is to reflash. But we don't want to re-enable ME. We want to preserve then reverse engineer the change that happened so we can do it anytime.
Isn't that a possible kill-switch for ME?
Anything I can help with?

WaseemAlkurdi, 2018-07-14:
@forkoz As in "dump the BIOS chip with a hardware programmer", right?

xCuri0, 2018-08-06:
@WaseemAlkurdi yes

FrostKnight, 2019-01-23:
@EaseemAlkirdi, please document it especially for processors that intel has abandoned, especially gen 3 and gen 4