corna/me_cleaner #216: What doesn't me_cleaner remove?

jdjjkfjkfkjda, 2018-07-19:
What doesn't me_cleaner remove? What can the proprietary software still do once me_cleaner has ran?

szymonmatuszczak, 2018-08-03:
In a brief, it doesn't remove only a hardware initialization part of the Intel ME. It is needed to disable a watchdog. If you want to remove every proprietary part possible from your computer's FW, flash coreboot with me_cleaner **stripped and relocated** ~90kB Intel ME blob (relocation is important, because it physically removes unused partitions from the flash chip, not only disables it by altering the partition table) and without a VGA blob - Linux doesn't need that.

corna, 2018-08-03:
@szymonmatuszczak is correct, however the relocation merely (in words, in reality it is quite difficult) moves the partitions to a lower address to recover the empty space. Both with and without the relocation the partitions are gone, both in the partition table and their code.

On generation 1 (before Nehalem, 1st generation core i3-i5-i7) it removes everything, ME is completely disabled (like in ich9deblob)

In generation 3 (after Skylake, 6th generation core i3-i5-i7) the kernel launches the bup (hardware initialization), so we have to keep more stuff (kernel, system libraries, hardware initialization), about 300 KB of compressed code.

szymonmatuszczak, 2018-08-03:
@corna Thanks for explaining the relocation thing!

Before Nehalem you don't need the me_cleaner, libreboot, the coreboot's fork incapable of loading blobs is all you need to be libre! :)

lwoody7110, 2018-12-04:
I'm hear after struggling with an eBay HP Z200 workstation.  MEBx appears in the lower right hand corner during post but Ctrl-P doesn't bring up the AMT config screens.  I've run me-cleaner on a HP z200 motherboard which has the watchdog turning the 'board off after 30 minutes.  me-cleaner removed many ME components but it didn't resolve the 30 minute issue for me.  

Subsequently, all attempts to downgrade the BIOS fail using HP flash but I think this is by design of the latest BIOS.  All attempts to install AMT drivers in Windows fail.  All attempts to flash the ME bios in either DOS or Windows fail.  Whilst perhaps not me-cleaner related, is there a method to restore ME/AMT to factory state in the hope of getting this 'board back to life - or another way to run me-cleaner that might disable the watchdog?

fredericoschardong, 2020-05-15:
@lwoody7110 any luck with the watchdog? I have the same issue.

lwoody7110, 2020-05-15:
I got it working in the end.

The (long) thread moved here and was resolved.

https://www.win-raid.com/t4165f39-Z-switches-off-after-minutes-unable-to-Ctrl-P-or-flash-ME-firmware.html

In my case, I used a Raspberry Pi and a clip-on programmer to reflash the motherboard.  I was a mixture of ‘really easy’ and frustration.  My recollection of detail has faded.

Take time to read the thread and see how you get on.  The guys at Win-Raid are highly knowledgeable and can help you like they did for me.

Good luck