corna/me_cleaner #217: UNKNOW VERSION by ASUS

SeutinJP, 2018-07-21:
Hello
My rom is not uefi
the version is P5G41M.ROM from asus.
me_cleaner message : unknow version.
What can i do ?

Manufacturer: 	ASUSTeK Computer INC.
Model(s): 	P5G41-M
Chipset: 	Intel® G41 /ICH7
Release Date: 	2010-04-27
BIOS ID: 	64-0308-000001-00101111-070709-Eaglelake
Chip Size: 	1024k

Thank-You for your help.

skochinsky, 2018-07-21:
it probably does not contain the ME firmware so you don't need me_cleaner. To be 100% sure you need to dump the flash chip contents from the board.

SEUTIN, 2018-07-24:
dump the flash chip contents from the board :
crypted ??

SEUTIN, 2018-07-27:
ROM is crypted i think, what can i do ?

skochinsky, 2018-07-27:
what makes you think it's "crypted"? did you dump the flash chip?

platomav, 2018-07-28:
There is no ME on that platform.

SeutinJP, 2018-07-29:
> thread [2].
> without ME intel can modify the boot rom ?
> Links:

SeutinJP, 2018-07-29:
Helllo , without ME , intel can modify the boot rom ?

SEUTIN J-P

SeutinJP, 2018-07-29:
"crypted"
I will say all is hexa only "intel" is alphanumerique.
Sorry
JPS