corna/me_cleaner #222: I was able to clean-up the ME.bin

xinterix, 2018-08-06:
So I download the ME vulnerability patch from Asus website for my Z270 Mark 1, Extracted the ME.bin file from FW directory and proceeded to try ME_cleaner on it. The only way I was able to get it working is without any options specified (clean Me only). I then replaced the original ME.bin in FW folder with the cleaned one. I then executed the **MEUpdateTool.exe** and it flashed the new ME.bin. All is ok I guess but am not able to flash anything to ME.bin anymore at least through internal methods. AFU4win will appear to flash new bios CAP file (original or modified one with or without  HAP set) or Bin successfully but after restarting the ME still appears as same version as before even after flash older version??? I'm wondering if something the cleaner did when I originally flashed the version from Asus patch, that locked the ME from further flashing?

Also I was wondering someone knows whether the following would indicate a cleaned ME or not:

**ME Firmware Capabilities**        
                                                
Full Network Manageability:                                                     Not Capable
Standard Network Manageability:                                                 Not Capable
Manageability (AMT):                                                            Not Capable
Small Business Advantage:                                                       Not Capable
Intel Integrated Touch:                                                         Not Capable
Intel Anti-Theft:                                                               Not Capable
Capability Licensing Service:                                                   Capable
Virtualization Engine:                                                          Not Capable
Intel Sensor Hub (ISH):                                                         Capable
ICC Over Clocking:                                                              Not Capable
Protected Audio Video Path (PAVP):                                              Capable
Network Frame Forwarder (NFF):                                                  Not Capable
Remote PC Assist (RPAT):                                                        Not Capable
IPV6:                                                                           Not Capable
KVM Remote Control:                                                             Not Capable
Outbreak Containment Heuristic (OCH):                                           Not Capable
Dynamic Application Loader (DAL):                                               Capable
Cipher Transport Layer (TLS):                                                   Not Capable
Wireless LAN (WLAN):                                                            Not Capable
Platform Trust Technology (PTT):                                                Capable
Near Field Communication (NFC):                                                 Not Capable

**ME Firmware Feature State**                                                       

Full Network Manageability:                                                     Disabled
Standard Network Manageability:                                                 Disabled
Manageability (AMT):                                                            Disabled
Small Business Advantage:                                                       Not Capable
MEI3:                                                                           Not Capable
Intel Anti-Theft:                                                               Disabled
Capability Licensing Service:                                                   Enabled
Virtualization Engine:                                                          Disabled
Intel Sensor Hub (ISH):                                                         Disabled
ICC Over Clocking:                                                              Disabled
Protected Audio Video Path (PAVP):                                              Enabled
Network Frame Forwarder (NFF):                                                  Not Capable
Remote PC Assist (RPAT):                                                        Disabled
IPV6:                                                                           Disabled
KVM Remote Control:                                                             Disabled
Outbreak Containment Heuristic (OCH):                                           Disabled
Dynamic Application Loader (DAL):                                               Capable
Cipher Transport Layer (TLS):                                                   Disabled
Wireless LAN (WLAN):                                                            Disabled
Platform Trust Technology (PTT):                                                Disabled
Near Field Communication (NFC):                                                 Disabled

dartraiden, 2018-08-28:
Some ASUS mobo (starting from 2xx-series) have seconde copy of ME. So, if motherboard detect ME older than backup copy, it just autoupdate ME firmware.

xinterix, 2018-08-29:
yeah i finally got it working but your right, it only works going up to higher versions or equal with this method equal, can't downgrade using this method. you also loose

closed 2018-08-29 by xinterix