corna/me_cleaner #227: me_cleaner says "unknown image" for Intel Desktop Board DQ45EK

bobafetthotmail, 2018-09-08:
It's a nice mini-itx board from Intel, socket 775.

It appears to be able to do UEFI boot so it's probably UEFI even if the board itself is from 2007 or something, the firmware is the latest available, from 2011 I think.

It has ME in full annoyance mode with AMT for remote management and "remote assistence" available if you press a key on first boot, with a dedicated red LED that shows if ME is active or booting, and when I connect the power plug it powers up briefly and then shuts down while the LED shows the ME becoming active

I desoldered the SPI chip, dumped the contents with a hardware flasher and ran ifdtool on it

````
alby@openSUSE-xeon:~/me_cleaner/coreboot/util/ifdtool> ./ifdtool -d /home/alby/me_cleaner/bios_schedaIntel_backup1.BIN       
File /home/alby/me_cleaner/bios_schedaIntel_backup1.BIN is 4194304 bytes
ICH Revision: ICH10
FLMAP0:    0x04040001
  NR:      4
  FRBA:    0x40
  NC:      1
  FCBA:    0x10
FLMAP1:    0x03100206
  ISL:     0x03
  FPSBA:   0x100
  NM:      2
  FMBA:    0x60
FLMAP2:    0x00000120
  PSL:     0x0001
  FMSBA:   0x200
FLUMAP1:   0x00000aed
  Intel ME VSCC Table Length (VTL):        10
  Intel ME VSCC Table Base Address (VTBA): 0x000ed0

ME VSCC table:
  JID0:  0x001630ef
    SPI Componend Vendor ID:            0xef
    SPI Componend Device ID 0:          0x30
    SPI Componend Device ID 1:          0x16
  VSCC0: 0x20052005
    Lower Erase Opcode:                 0x20
    Lower Write Enable on Write Status: 0x50
    Lower Write Status Required:        No
    Lower Write Granularity:            64 bytes
    Lower Block / Sector Erase Size:    4KB
    Upper Erase Opcode:                 0x20
    Upper Write Enable on Write Status: 0x50
    Upper Write Status Required:        No
    Upper Write Granularity:            64 bytes
    Upper Block / Sector Erase Size:    4KB
  JID1:  0x004125bf
    SPI Componend Vendor ID:            0xbf
    SPI Componend Device ID 0:          0x25
    SPI Componend Device ID 1:          0x41
  VSCC1: 0x20092009
    Lower Erase Opcode:                 0x20
    Lower Write Enable on Write Status: 0x50
    Lower Write Status Required:        Yes
    Lower Write Granularity:            1 bytes
    Lower Block / Sector Erase Size:    4KB
    Upper Erase Opcode:                 0x20
    Upper Write Enable on Write Status: 0x50
    Upper Write Status Required:        Yes
    Upper Write Granularity:            1 bytes
    Upper Block / Sector Erase Size:    4KB
  JID2:  0x001620c2
    SPI Componend Vendor ID:            0xc2
    SPI Componend Device ID 0:          0x20
    SPI Componend Device ID 1:          0x16
  VSCC2: 0x20052005
    Lower Erase Opcode:                 0x20
    Lower Write Enable on Write Status: 0x50
    Lower Write Status Required:        No
    Lower Write Granularity:            64 bytes
    Lower Block / Sector Erase Size:    4KB
    Upper Erase Opcode:                 0x20
    Upper Write Enable on Write Status: 0x50
    Upper Write Status Required:        No
    Upper Write Granularity:            64 bytes
    Upper Block / Sector Erase Size:    4KB
  JID3:  0x0000471f
    SPI Componend Vendor ID:            0x1f
    SPI Componend Device ID 0:          0x47
    SPI Componend Device ID 1:          0x00
  VSCC3: 0x20152015
    Lower Erase Opcode:                 0x20
    Lower Write Enable on Write Status: 0x06
    Lower Write Status Required:        No
    Lower Write Granularity:            64 bytes
    Lower Block / Sector Erase Size:    4KB
    Upper Erase Opcode:                 0x20
    Upper Write Enable on Write Status: 0x06
    Upper Write Status Required:        No
    Upper Write Granularity:            64 bytes
    Upper Block / Sector Erase Size:    4KB
  JID4:  0x00128989
    SPI Componend Vendor ID:            0x89
    SPI Componend Device ID 0:          0x89
    SPI Componend Device ID 1:          0x12
  VSCC4: 0x401ed81f
    Lower Erase Opcode:                 0x40
    Lower Write Enable on Write Status: 0x06
    Lower Write Status Required:        Yes
    Lower Write Granularity:            64 bytes
    Lower Block / Sector Erase Size:    8KB
    Upper Erase Opcode:                 0xd8
    Upper Write Enable on Write Status: 0x06
    Upper Write Status Required:        Yes
    Upper Write Granularity:            64 bytes
    Upper Block / Sector Erase Size:    64KB

OEM Section:
00: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
10: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
20: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
30: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff

Found Region Section
FLREG0:    0x00000000
  Flash Region 0 (Flash Descriptor): 00000000 - 00000fff 
FLREG1:    0x03ff0280
  Flash Region 1 (BIOS): 00280000 - 003fffff 
FLREG2:    0x02770003
  Flash Region 2 (Intel ME): 00003000 - 00277fff 
FLREG3:    0x00020001
  Flash Region 3 (GbE): 00001000 - 00002fff 
FLREG4:    0x027f0278
  Flash Region 4 (Platform Data): 00278000 - 0027ffff 

Found Component Section
FLCOMP     0x00300013
  Dual Output Fast Read Support:       not supported
  Read ID/Read Status Clock Frequency: 20MHz
  Write/Erase Clock Frequency:         20MHz
  Fast Read Clock Frequency:           33MHz
  Fast Read Support:                   supported
  Read Clock Frequency:                20MHz
  Component 2 Density:                 2MB
  Component 1 Density:                 4MB
FLILL      0x00000000
  Invalid Instruction 3: 0x00
  Invalid Instruction 2: 0x00
  Invalid Instruction 1: 0x00
  Invalid Instruction 0: 0x00
FLPB       0x00000000
  Flash Partition Boundary Address: 0x000000

Found PCH Strap Section
PCHSTRP0:  0x10195e08
PCHSTRP1:  0x0002010f
PCHSTRP2:  0x6165696d
PCHSTRP3:  0xffffffff
PCHSTRP4:  0xffffffff
PCHSTRP5:  0xffffffff
PCHSTRP6:  0xffffffff
PCHSTRP7:  0xffffffff
PCHSTRP8:  0xffffffff
PCHSTRP9:  0xffffffff
PCHSTRP10: 0xffffffff
PCHSTRP11: 0xffffffff
PCHSTRP12: 0xffffffff
PCHSTRP13: 0xffffffff
PCHSTRP14: 0xffffffff
PCHSTRP15: 0xffffffff
PCHSTRP16: 0xffffffff
PCHSTRP17: 0xffffffff
ICH_MeDisable bit is not set

Found Master Section
FLMSTR1:   0x1a1b0000 (Host CPU/BIOS)
  Platform Data Region Write Access: enabled
  GbE Region Write Access:           enabled
  Intel ME Region Write Access:      disabled
  Host CPU/BIOS Region Write Access: enabled
  Flash Descriptor Write Access:     disabled
  Platform Data Region Read Access:  enabled
  GbE Region Read Access:            enabled
  Intel ME Region Read Access:       disabled
  Host CPU/BIOS Region Read Access:  enabled
  Flash Descriptor Read Access:      enabled
  Requester ID:                      0x0000

FLMSTR2:   0x0c0d0000 (Intel ME)
  Platform Data Region Write Access: disabled
  GbE Region Write Access:           enabled
  Intel ME Region Write Access:      enabled
  Host CPU/BIOS Region Write Access: disabled
  Flash Descriptor Write Access:     disabled
  Platform Data Region Read Access:  disabled
  GbE Region Read Access:            enabled
  Intel ME Region Read Access:       enabled
  Host CPU/BIOS Region Read Access:  disabled
  Flash Descriptor Read Access:      enabled
  Requester ID:                      0x0000

FLMSTR3:   0x08080218 (GbE)
  Platform Data Region Write Access: disabled
  GbE Region Write Access:           enabled
  Intel ME Region Write Access:      disabled
  Host CPU/BIOS Region Write Access: disabled
  Flash Descriptor Write Access:     disabled
  Platform Data Region Read Access:  disabled
  GbE Region Read Access:            enabled
  Intel ME Region Read Access:       disabled
  Host CPU/BIOS Region Read Access:  disabled
  Flash Descriptor Read Access:      disabled
  Requester ID:                      0x0218

Found Processor Strap Section
????:      0x00000040
????:      0xffffffff
????:      0xffffffff
????:      0xffffffff
????:      0xffffffff
????:      0xffffffff
????:      0xffffffff
????:      0xffffffff
MCH_MeDisable bit is not set
MCH_AltMeDisable bit is not set
```

So far so good

but when I ran the me_cleaner on it it fails 
```

alby@openSUSE-xeon:~/me_cleaner> ./me_cleaner.py -c bios_schedaIntel_backup.BIN 
Unknown image
```

I uploaded the dump here https://www.dropbox.com/s/fpcnyp5184jsrhc/bios_schedaIntel_backup1.BIN?dl=0 if you want to have a look.

As said above I have an external flasher so I can test things without issues.

I would really like to get rid of the ME on this board, any ideas?

persmule, 2018-09-09:
Like PM45/ICH9M system mentioned in https://github.com/corna/me_cleaner/issues/25 , the ME on ICH10 can theoretically be completely obliterated. To do so, you only need to modify the ifd as how libreboot's [ich9deblob](https://libreboot.org/docs/hcl/gm45_remove_me.html) and [newer ifdtool's option -M](https://review.coreboot.org/c/coreboot/+/21437) does (set ICH_MeDisable, MCH_MeDisable, and MCH_AltMeDisable), and (optionally) fill ME region with 0xff.

I do not know whether me_cleaner has the capability to do the same thing on chipsets from ICH8 to Ich10.

bobafetthotmail, 2018-09-09:
Thanks, :+1:  I used ifdtool and now the board's ME appears to be dead. I didn't delete the firmware though.

After I reflashed and resoldered on the chip, the BIOS options for Intel ME disappeared, the system does not do anymore a power-on-then-power-off cycle when connected to power, and the intelmetool reports that it "can't find ME PCI device".

I'd say it's good.

> I do not know whether me_cleaner has the capability to do the same thing on chipsets from ICH8 to Ich10.

It seems it does not, as it was printing "unknown image", see above (I fixed formatting of the OP).

@corna could you add a wiki page to describe how to deal with older firmware that the me_cleaner does not work on (and is probably not worth adding support as it can be done with other tools already)?

corna, 2018-09-09:
@bobafetthotmail I have added support for pre-Nehalem platforms in 57b3fc765efeb0a66d121d9f1f4986d0b22a245a (dev branch) but I forgot to push it to master. Can you run _me\_cleaner_ on your image and check whether the output one is identical to the one from ifdtool?

corna, 2018-09-09:
From [ich9deblob](https://notabug.org/libreboot/libreboot/src/master/projects/ich9gen/sources/src/descriptor/descriptor.h#L251), "ME Alternate Disable: Setting this bit allows ME to perform critical chipset functions but prevents loading of any ME FW applications." which seems identical to the current gen2/gen3 "AltMeDisable". On gen1 it is possible to fully disable it, so setting this bit to 1 when `meDisable` is 1 is probably not necessary.

me_cleaner also overwrites the ME region with 0xff and disables it in the descriptor (`Flash Region 2 (Intel ME): 00fff000 - 00000fff (unused)`)

If you have an easy way to recover the board from a brick it would be helpful (for me and the community) if you flash the me_cleaner's firmware, if you can't just stick with the current one

corna, 2018-09-18:
Great!

closed 2018-09-18 by corna

STPKITT, 2018-11-27:
I own a Intel DQ45EK mainboard whose Intel ME is driving me nuts since I got it years ago. Could somebody provide be with an ME-free BIOS image that I can just flash without an external programmer.

STPKITT, 2018-11-27:
I own a Intel DQ45EK mainboard whose Intel ME is driving me nuts since I got the board years ago. I now tried to dump the BIOS to use me_cleaner on the dump, but I failed at that. Intel's iflash2.exe has no option to dump, the BIOS's own flash function also does not. I tried using the Linux tool flashrom, but that failed to read. So is there anyway to remove ME from my mainboard without desoldering the SPI chip?

bobafetthotmail, 2018-11-27:
I can share info on how to desolder a SPI chip. Place a blob of solder on both sides of the chip so that all pins are bridged, then take two soldering irons and heat both sides together, when you see that it's free you can pull it like a tweezer.

Then use solder wick to remove the solder blob and place the chip in an external programmer. 

SOldering it down is much easier if you have solder wick, you just do it so the pins are soldered, then remove any solder bridges between pins with the wick.

I used a TL866 PLUS  external programmer http://autoelectric.cn/EN/TL866_main.html with a SOIC8 adapter, as its software does support this board's SPI chip (although it's always better to check for yourself, maybe they used a different brand/model in your batch)

The SPI chip is in the middle of the under side of the board.

STPKITT, 2019-01-23:
Thanks for your answer! So if there's no way around soldering I'll abandon using that board since I alck both the knowledge and the tools to solder.

STPKITT, 2019-01-23:
Thanks for your answer! So if there's no way around soldering I'll abandon using that board since I lack both the knowledge and the tools to solder.

skochinsky, 2019-01-24:
You can try Intel's FPT (flash programming tool) to dump the firmware but you likely won't be able to write modified one if region protections are enabled.

bobafetthotmail, 2019-01-24:
> Thanks for your answer! So if there's no way around soldering I'll abandon using that board since I lack both the knowledge and the tools to solder.

If you are in in EU you can buy a Pegatron IPX41-R3 on ebay for like 20 euro. It is very similar to this board (it has no UEFI boot, the DVI port does not work in Windows 10 after you install the graphics driver while it works fine with the generic driver or with Linux, it has only 2 Sata ports), but it does not seem to have Intel ME/AMT, and it has the SPI chip on a socket.