corna/me_cleaner #229: MSI Z97 MPower (lock BCLK, BUS 99.76Mhz)

ValoWaking, 2018-09-20:
Hi. I have some problem with my tweaks:

When I set disable me bit in FIT (Flash Image Tool) - I have lock bclk and 100Mhz BUS
When I use me_cleaner with "-s" - I have lock bclk and 100Mhz BUS too
But when I use me_cleaner with "-S" - I have lock bclk and 99.76Mhz BUS

To use me_cleaner with "-S" i do - in MMTool I deleted bootguardDXE. Without this, when I use "-S" - i have error "boot guard verified dxe that is fail"

At now I want to fix my BUS on 100Mhz with me_cleaner "-S"
Any ideas?

ValoWaking, 2018-09-26:
How can i disable delete FTPR module? Like FPF, TDT etc

tiredtiger, 2018-10-07:
> 
> 
> How can i disable delete FTPR module? Like FPF, TDT etc

by modifying the "unremovable_modules" list in the source (me_cleaner.py). You can edit the source with a text editor like Windows "notepad"

JuneCN, 2019-05-23:
Can you OC the BCLK to more than 100MHz ?